lumapu/ahoy #1057: Langsame WebGUI und MQTT-Verbindungsabbrüche

knickohr, 2023-08-01:
Sind zwar jetzt zweierlei unterschiedliche Probleme, aber ich denke sie hängen direkt oder indirekt miteinander zusammen.

Zuerst mal, dieser Issue wird nur User tangieren die viele Wechselrichter an der DTU haben. Denn mir ist aufgefallen das diese Probleme mir steigender Anzahl der WRs immer schlimmer werden. Das würde auch meiner Meinung nach erklären warum viele 8266-User bereits ab 2 WRs Probleme mit der WebGUI bekommen.

Ich selbst nutze einen ESP32 (kein S3 momentan, damit gibt es ein anderes Problem mit dem ePaper Display ab der Version 0.7.x) und jetzt schon 13 HM-300 WRs. Das Problem mit dem WebGUT trat erstmals ab der 0.7.17 auf, Seiten werden teilweise überhaupt nicht mehr dargestellt, bzw. Es kommen völlig unsinnige Ausgaben. Vor allem die Live-Seite ist mittlerweile für mich unbrauchbar geworden 😲

[image]

Es werden mehrere Total-Boxen angezeigt, hatte schon 7 Stück 😱 und was noch viel schlimmer ist, in keiner Box waren die Werte korrekt !

Andere Seiten sind sehr langsam, so dauert es manchmal 10s und länger bis sie aufgebaut werden. Die „Schwuppdizität“ ist unter aller Sau 😪

Seit der Version 0.7.21 kommt ein weiteres Problem hinzu. Es gibt, nachdem die Inverter alle offline sind, fast im Minutentakt Verbindungsabbrüche vom MQTT.

[image]

Hier kann man es sehr schön erkennen, das Problem ist nur Nachts vorhanden 🤔

[image]

Ob es jetzt ausschließlich an der großen Zahl der WRs liegt, wage ich zu bezweifeln. Da würde das Problem eher unter Tags auftreten. Viel eher denke ich das hier die LWT-Message zuschlägt, weil der Keep-Alive-Timeout kommt. Denn wenige Sekunden später ist MQTT wieder connected.

Was ich bereits getestet habe :
- Factory Reset und komplette Neueinrichtung
- Komplett frisch geflashter ESP mit vorherigem Erase
- Die Verbindung zum Broker steht stabil, sonst würde mir die Raspi-Konsole auch wegfliegen

Ich werde heute Abend mal versuchen das MQTT-Intervall zu verkürzen. Vielleicht wird es damit besser. Momentan nutze ich die Einstellung 0, also nur wenn sich Werte ändern wird gesendet.

Riker18, 2023-08-01:
Hab das Problem auch schon bei einem WR 8266 - hab gestern mal die aktuelle Dev geflasht und dann dauernd Abbrüche über den ganzen Tag gehabt und reboots. Webinterface hängt öfters und dann rebootet die Kiste.   Im Moment bin ich wieder auf 0.7.11 zuück da ich nicht weis ab wann das Problem gekommen ist - dazwischen hatte ich keine Versionen getestet.

knickohr, 2023-08-01:
Das ist frustrierend 😫 Es ist Nacht und die MQTT-Abbrüche fangen wieder an. Eben sind mir beim Setzten des Intervalls sogar die kompletten Invertereinstellungrn flöten gegangen 😡 Das Ahoy-Spielen ist somit für heute Abend beendet ☹️

knickohr, 2023-08-02:
Irgendwelche seltsamen Dinge passieren hier !  Ahoy spukt 😱

Aber mal der Reihe nach. Nachdem ich gestern Abend durch rumspielen an den Settings meine Invertereinstellungen verloren habe, machte ich einen neuen Import der Settings. Somit waren ja erst mal alle gespeicherten Werte von den Invertern nach dem Reboot weg. Egal, aber was auffällt, das es nachts wieder fast im Minutentakt diese MQTT-Abbrüche gab. Heute Morgen habe ich es dann testweise mit einem festen MQTT-Intervall von 35s laufen lassen. Leider gibt es damit dann auch unter Tags diese Abbrüche, wenn auch wesentlich weniger. Aber was positiv auffällt, heute Abend kamen keine Abbrüche im Minutentakt, also schon mal prima 👍

Vor einer halben Stunde habe ich dann das Fusion-Board mit dem S3 raus gekramt und aktiviert.

Hossa die Waldfee ! 😲 Die Swuppdizität der WebGUI ist 1a, besser kann’s gar nicht sein. Und auch die minütlichen MQTT-Abbrüche sind weg. Jetzt erklär mir das mal einer !

[image]

Ich werde das Morgen mal „unterLast“ beobachte. Mir deicht langsam der Verdacht, das der native ESP32 mit 13 Invertern auch am Ende ist ! 😢

Leider funktioniert das ePaper mit dem Fusion und Ahoy nicht mehr, da muß seit der ersten Version 0.7.x was an den Pins seitens Ahoy gedreht worden sein. Ich vermute mal wegen dem CMT 😞

knickohr, 2023-08-03:
Bilder sagen mehr als 1000 Worte :

[image]

Seit gestern Abend gab es nur mehr 2 MQTT-Abbrüche und die Schuppdizität ist perfekt, auch bei der Live-Seite. Und vor allem, die Werte stimmen jetzt wieder 😎

Einen Test muß ich nachher noch machen um das auszuschließen. Ich hoffe nicht, das es das ist, denn das wäre dann ein kompletter Rückschritt seit der Version 0.7.x 🫣

knickohr, 2023-08-03:
Sooo, wir sind einen weiteren Schritt weiter, aber leider kurz vor dem Abgrund ☹️

Mein erster Test hat sich glücklicherweise nicht bewahrheitet. Es ist nicht das ePaper 😎das den MQTT und die Schwuppdizität blockiert.

Und jetzt leider die schlechte Nachricht 😫 Es ist das MQTT selbst. Denn deaktiviere ich es, dann klappt es auch wieder mit der Schwuppdizität 😲

Ein weiterer Test zurück auf die Version 0.7.11 und es funktioniert noch alles 😱

Nun stehe ich da :
- entweder verwende ich den nativen ESP32 ohne MQTT aber mit funktionierendem ePaper
- oder ich nehme den ESP32-S3 (Fusion-Board), dann klappt es mit MQTT aber nicht mit dem ePaper
-  oder ich gehe zurück auf die 0.7.11. da funktionieren ePaper und MQTT am ESP32 noch

Alles Sch…!

knickohr, 2023-08-04:
Faszinierend was so ein neuer Prozessor bewirkt 😲

[image]

Die 3 Abbrüche kommen von manuellen Reboots, ich habe gestern Abend noch etwas gespielt.

Nun stellt sich für mich die Frage ob wir hier in ein Ressourcenproblem rein gelaufen sind, oder ob einfach nur unglücklich programmiert wurde ? 🤔

lumapu, 2023-08-04:
aktuell ist ein workaround für MqTT aktiv, das kann sich bei der hohen Anzahl an Wechselrichtern natürlich bemerkbar machen.
Das heißt für mich, ich muss den workaround rückgängig machen und es direkt in der MqTT Lib beheben.
Hintergrund: die MqTT Lib reserviert aktuell für jedes Paket neuen Speicher und gibt den zyklisch beim tatsächlichen senden wieder frei. Wenn man bedenkt, dass wir sehr viele Pakete auf einmal schicken ist das eine beträchtliche Menge an Speicher. In erster Linie ist das vor allem für den ESP8266 schädlich, da hierdurch sein heap komplett zerstört (durchlöchert) wird.
Als workaround habe ich eine Methode eingebaut, die jedes Paket einzeln sendet und somit den heap nicht belastet - genau dies ist bei 13 WR nicht unerheblich und wohl auch die falsche Lösung.
Ich habe verschiedene Ideen und hoffe, dass wir dieses Problem damit aus der Welt schaffen können.

Was die Abbrüche nachts angeht, kann ich noch nicht nachvollziehen

MartinBachmannHD, 2023-08-04:
Nachdem es auf dem ESP8266 (ein alter, gebrauchter und ein komplett neuer) bei mir mit MQTT (aber nur ein HM-600 angeschlossen) nicht so richtig flüssig im UI war habe ich einen neuen ESP32 probiert - immer noch das selbe Verhalten, unresponsive... 
Mein Verdacht ging noch in eine weitere Richtung die das MQTT-Heap Thema evtl noch verstärken könnte - es kommt immer zu TCP-Verbindungs-Abbrüchen (Energiesparen FritzBox oder ESP? Schlechte Verbindung? Wacklige Verdrahtung? ...). Evtl vertärkt das das MQTT-Heap Thema, da dann die Messsages noch gespeichert werden bis sie gesendet werden -> mehrfache Belastung?

knickohr, 2023-08-04:
OK, der S3 hat einen etwas größeren Heap, aber wirklich mehr Ressourcen hat er nicht. Was aber beim S3 neu ist, neben mehr GPIO, Flash, USB und Bluetooth, er hat 2 Coprozessoren implementiert bekommen die Rechenaufgaben (vor allem Multiplikation und Division) übernehmen, das könnte schon erklären warum es mit dem S3 wesentlich besser klappt. Wäre da nicht das ePaper-Problem 😪

lumapu, 2023-08-04:
was QoS angeht hast du recht, allerdings sammelt die MqTT-Library intern noch die Daten, bevor sie dran kommt die angefallenen Daten rauszuschicken.

lumapu, 2023-08-04:
@knickohr, wie steht eigentlich dein Intervall? Wenn du auf der Live-Seite ganz nach unten scrollst solltest du es auch sehen können. 
*Every X seconds the values are updated*

Evtl. ist das bei der extrem kurz, damit du in angemessener Zeit alle Inverter abgefragt bekommst - ich baue hier mal ein Minimum von 5s ein.

knickohr, 2023-08-05:
Es steht auf 5s, weniger ging schon damals nicht. Dann kam er mit den Retries und den Abfragen nicht mehr hinterher und hat sich vergewohlwurstelt 😉

knickohr, 2023-08-05:
Yeahhh !  Die 23 fluppt wieder 😎😘

Ich werde jetzt aber trotzdem eine zweite DTU aufstellen da ich Empfangsprobleme mit meinen weiter entfernten WRs habe (Gartenhaus). Bin mal gespannt, wie das dann zusammen werkelt 🫣

knickohr, 2023-08-05:
Sieht gut aus, keine Verbindungsabbrüche mehr, auch wenn die DTU offline ist. Wenn’s bis Morgen Abend so bleibt, kann man hier zu machen 👍

„Dual Core“ funktioniert auch, jetzt klappt’s mit den weiter entfernten WRs gut. War einfacher als gedacht, das in Node-Red zu implementieren 😎

[image]

closed 2023-08-05 by lumapu