lumapu/ahoy #1065: Add support for SML-reader

technics42, 2023-08-02:
It would be very nice, if you could directly connect some optical device (photo-transistor/-diode/...) and support SML in order to get values from energy meter.
(e.g. https://github.com/mruettgers/SMLReader)

And by doing so, calculate saved costs/power consumption and display it in web-browser.

rmayergfx, 2023-08-02:
Sorry, this is not the goal of the project. we need a stable solution to get data from (multiple) inverters. Additional hardware and SML will increase the code and also you need hardware with enough performance to support this.

kiu77, 2023-08-03:
@rmayergfx 
good point.

Ollipop030, 2023-08-03:
> the optical sensor would stop at 0, without any possibility to see overproduction, and has a very high latency

Thats not correct. My Tasmota SML shows a negative value when overproducing, the latency is only a few seconds. I have a second (non-Tasmota) SML, that reads the exact values for each phase when cosuming (1.7.0. - 1.7.3.) and overproducing (2.7.0. - 2.7.3.). But that depends on the powermeter itself. But most powermeters should deliver 1.7.0. and 2.7.0.

MetaChuh, 2023-08-03:
@Ollipop030 
danke für das update und sorry, dachte das gilt für alle smart meter, nicht nur die zähler hier bei uns in wien.
hier ist halt alles slower, außer das stromrechnung abbuchen vom konto 😂
hast du dich auch schon mit dem 3em oder ähnlichem im sicherungskasten rumgespielt ?

nexulm, 2023-08-03:
I'm using a Tasmota based SM reader for my EZB as well. Doesn't make sense to me to integrate it into the Ahoy code as well.
BTW: My Ahoy HW is located in the garden and my EZB SM reader 25m away. :-)
[image]

Ollipop030, 2023-08-03:
> hast du dich auch schon mit dem 3em oder ähnlichem im sicherungskasten rumgespielt ?

Nein, bisher nicht. Ich habe ein paar 1-Phasige Shellys, aber da noch nie wirklich auf die Zeit geachtet, wann welche Daten kommen. Zumindest was die Cloud angeht werden aber demnächst wohl alle Geräte minütlich ihre Werte übertragen:

https://stadt-bremerhaven.de/shelly-mit-abo-smart-control-premium-vorgestellt/

MetaChuh, 2023-08-03:
> Nein, bisher nicht. Ich habe ein paar 1-Phasige Shellys, aber da noch nie wirklich auf die Zeit geachtet, wann welche Daten kommen. Zumindest was die Cloud angeht werden aber demnächst wohl alle Geräte minütlich ihre Werte übertragen:

danke für die info.

cloud verwende ich derzeit nicht, nur die apis der gen1 und gen2 geräte.

falls du code sniplets (bash, php, etc.) für gen1 oder gen2 geräte brauchst, sag bescheid.
gen2 mit rpc api hat gravierende änderungen, hab‘ aber schon fast alles durchprobiert (ist noch nicht alles in der gen2 shelly api dokumentiert)

greetings,
metachuh

closed 2023-08-05 by lumapu