lumapu/ahoy #1066: [Bug] Warning in Home Assistant with multiple inverters because of MQTT entity naming guideline

CantConAnHonestJohn, 2023-08-02:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.9

### Github Hash

I don't know

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

- MQTT is enabled

### Error description

Is anyone else with multiple inverters in ahoy seeing the corresponding warning in Home Assistant to this article:
https://developers.home-assistant.io/blog/2023-057-21-change-naming-mqtt-entities/

princuser, 2023-08-02:
I got the same message. I've ESP8266 Board with version 0.6.9. From Home Assistant version 2024.2.0 it is no longer supported (see https://developers.home-assistant.io/blog/2023-057-21-change-naming-mqtt-entities/). 
[image]

rmayergfx, 2023-08-02:
it´s also on the single inverter setup with 0.7.21.
[image]

Neelixx, 2023-08-03:
Hier noch ein Beispiel für einen HM-600:
Liste der betroffenen Entitäten:

    sensor.hm600_p_ac
    sensor.hm600_temp
    sensor.hm600_ch2_p_dc
    sensor.hm600_ch1_u_dc
    sensor.hm600_efficiency
    sensor.hm600_ch1_p_dc
    sensor.hm600_yieldday
    sensor.hm600_ch2_u_dc
    sensor.hm600_q_ac
    sensor.hm600_ch2_yieldtotal
    sensor.hm600_pf_ac
    sensor.hm600_alarm_mes_id
    sensor.hm600_ch1_i_dc
    sensor.hm600_ch1_yieldday
    sensor.hm600_ch1_yieldtotal
    sensor.hm600_ch2_i_dc
    sensor.hm600_i_ac
    sensor.hm600_ch2_irradiation
    sensor.hm600_u_ac
    sensor.hm600_yieldtotal
    sensor.hm600_ch1_irradiation
    sensor.hm600_ch2_yieldday
    sensor.hm600_p_dc
    sensor.hm600_f_ac

Mehr erfahren:
https://developers.home-assistant.io/blog/2023-057-21-change-naming-mqtt-entities/

nexulm, 2023-08-03:
> ich kann das ganze noch nicht ganz verstehen, skizziere aber mal was ich meine Verstanden zu haben: Homeassistant hätte gerne, dass die Entitäten generisch benannt sind und nicht nach dem Gerät. Das würde im Screenshot Beispiel bedeuten, dass es nicht `sensor.tsun_m800_ch2_yieldtotal` heißt, sondern `sensor.inverter_ch2_yieldtotal`. Was ich nicht verstehe, wie Homeassistant dann auf einen spezifischen Wechselrichter zugreifen will, da ja die Entitäten nicht mehr eindeutig sind.
> 
> Das hier habe ich noch gefunden, allerdings ist auch das Beispiel hier sehr dürftig. Hat einer von euch ein Beispiel eines anderen Geräts für mich - zB. Sonoff / Shelly oä.? https://developers.home-assistant.io/blog/2022/07/10/entity_naming/

Beim Shelly 3EM sieht es bei mir so aus:
[image]
---
Beim Tasmota SM dann so:
[image]

rmayergfx, 2023-08-03:
Hier gibt es ein Topic dazu:
https://community.home-assistant.io/t/psa-mqtt-name-changes-in-2023-8/598099

nexulm, 2023-08-03:
...und hier noch hier eine Quelle wie der Bugfix im EMS-ESP32 Projekt erfolgreich gelöst wurde.
https://github.com/emsesp/EMS-ESP32/issues/1035

Hiermit habe ich nun eine Entität weniger, die auffällt. Bin zuversichtlich, dass mit dem nächsten Ahoy-Update dies auch für alle Entitäten aus der DTU der Fall sein wird. ;-)

lumapu, 2023-08-04:
bitte mit neuer Version prüfen, ob ich es richtig verstanden habe

nexulm, 2023-08-04:
Danke dir für diesen mega schnellen Bugfix. Leider funktioniert er bei mir leider nicht wie gehofft. Nach DTU Update v0.7.22 und HomeAssistant Neustart erhalte ich noch immer zahlreiche fehlerhaften Entitäten als Fehler. :-(
Die Änderung hat noch nicht gegriffen...
[image]

lumapu, 2023-08-04:
hast du vorher die alten Daten gelöscht? Sieht irgendwie noch nicht geändert bei dir aus, so sollte es jetzt sein:

```json
{
    "name": "CH1_YieldDay",
    "stat_t": "esp32/HMS2000.1/ch1/YieldDay",
    "unit_of_meas": "Wh",
    "uniq_id": "123456789012_ch1_YieldDay",
    "dev": {
        "name": "HMS2000.1",
        "ids": "123456789012",
        "mdl": "HMS2000.1",
        "cu": "http://10.20.3.40",
        "mf": "Hoymiles"
    },
    "dev_cla": "energy",
    "stat_cla": "total_increasing"
}
```

nexulm, 2023-08-04:
Du hast natürlich Recht. :-(
Der Bugfix funktioniert. :-)
Ich vermisse allerdings nach löschen und neu hinzufügen die Gesamt "nicht" ChX Werte aus der Übersicht:
[image]

lumapu, 2023-08-04:
die sollten sich da sein, allerdings aus eigenes Gerät mit deinem Ahoy-DTU Namen

nexulm, 2023-08-04:
Genau dieses eigene Geräte hatte ich auch gelöscht, ist jetzt aber im ersten Schritt nicht neuangelegt worden. Reboot der DTU oder HomeAssistant hat auch noch nicht gefruchtet. 
Woran könnte das dann noch liegen?
Starten den MQTT Broker nochmal neu, wobei es sich dabei eher "dummes" rumprobieren handelt.

lumapu, 2023-08-04:
hast du in Ahoy im den Einstellungen über MqTT auf `send` gedrückt?

rmayergfx, 2023-08-04:
Der Workaround funktioniert, aber damit verliert man leider auch bei allen Sensoren die Historie!

lumapu, 2023-08-04:
hmm. Gibt es bessere Vorschläge? Ich denke HA hat hier inkompatible Änderungen gemacht.

nexulm, 2023-08-04:
> hast du in Ahoy im den Einstellungen unter MqTT auf `send` gedrückt?

Ja, mehrfach. Es wurde weder ein weiteres Device Ahoy-DTU wie ich es früher (vor dem löschen) hatte noch die fehlenden Werte aufgenommen.
ABER
Verrückt ist, dass ich ein Downgrade auf v0.7.21 gemacht habe, wieder auf "send" gedrückt habe und siehe da, es wurde zwar weiterhin kein weiteres Device Ahoy-DTU angelegt, aber die übergeordneten Werte Yield Day, ... wurden wieder (diesmal auf dem vorhanden Device) mitgesendet. 
Nach dem Update auf v0.7.22 habe ich nun alle wieder alle Werte ohne die Namens-Fehlermeldungen. Perfekt. :-)

nexulm, 2023-08-04:
> Der Workaround funktioniert, aber damit verliert man leider auch bei allen Sensoren die Historie!

Sieht bei mir aber plausibel aus, nur kurze Unterbrechung (kaum wahrnehmbar) zwischen 19:15 und 19:45 Uhr beim Update der DTU und der Reboots.
[image]

[image]

Beim letzten Screenshot sieht man auch schön die fehlenden Werte von Yield Day und Power P_DC ab ca. 19:10 Uhr. Die anderen Werte werden brav auch mit v0.7.22 erfasst und visualisiert.

nexulm, 2023-08-05:
Der Bug kann aus meiner Sicht geschlossen werden. Mit v0.7.23 habe ich wieder alle Werte verfügbar. MQTT AhoyDTU löschen und einmalig "send" in der Weboberfläche unter MQTT hat bei mir wieder alle Entitäten hergestellt. Nach einem HA Reboot sind dann auch die Warnungen bezgl. MQTT entity naming guideline Geschichte. :-)
Vielen Dank für den schnellen Bugfix vom Bugfix. ;-)
Auch ist die History OK.
Als Beispiel der Hauptkanal, der mit v0.7.22 gestern Abend nach Update nicht ging und mit v0.7.23 mit Update von heute morgen wieder unauffällig funktioniert. :-) 
[image]

Demgegenüber noch ein exemplarische Screenshot aus Ch1 wo die Werte immer (nur durch Updates/Resets kurzzeitig unterbrochen) mit v0.7.22 und v0.7.23 weiter erfasst werden.
[image]

closed 2023-08-05 by lumapu