lumapu/ahoy #1070: Feature Request -  WebGUI Symbole andere Farbe

knickohr, 2023-08-05:
Kann man das orange Häkchen in einer anderen Farbe machen ?  Meinetwegen gelb ?

Beim schnellen drüber schauen geht es mit den anderen orangen Symbolen leicht unter 😉

[image]

lumapu, 2023-08-05:
die Inverter mit dem Status `is available but no data was received until now` bekommen jetzt den grünen Haken, den zuvor dir produzierenden Inverter bekommen haben (in grün).
Letztere bekommen jetzt ein gefüllten Haken auch in grün. 

[image]

closed 2023-08-05 by lumapu

knickohr, 2023-08-06:
Ich hab gesagt nicht dringend 😂 Aber gut, wenn’s denn sofort sein muß.

(Das schreibe ich jetzt immer „nicht dringend“, dann isses Morgen gemacht 😇🫣)