lumapu/ahoy #1071: flashen nicht möglich

flyer213, 2023-08-05:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

nicht möglich

### Github Hash

nicht möglich

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

nicht möglich

### Debug Serial Log output

```shell
nicht möglich
```

### Error description

flashen nicht möglich W11 errorcode 43.
2 verschiedene Boards gleicher Fehler, 3 verschiedene Rechner 2x W10, 1x W11
immer der gleiche errorcode.
Connect wird angezeigt, Preparing dann erfolgt die Endlosschleife.

lumapu, 2023-08-05:
das habe ich zum ersten mal gehört, welcher Browser wurde verwendet? Habe gestern zufällig den Webinstaller geteset und da ging er mit Chrome an einem ESP8266.
Kann zu error 43 nichts im Netz finden. Jetzt verstehe ich erst was mit W10 und W11 gemeint ist (Windows 10 / 11 😅)

flyer213, 2023-08-06:
Browser ist Chrome gewesen, USB ist Micro.Ich konnte den USB Anschluss abziehen, keine Fehlermeldung. Habe nun Board mit USB C bestellt.Windows sagt unbekanntes Gerät, ich kann auch keine Hardware ID auslesen.
Melde mich wenn ich Erfolg habe.
Danke.

rmayergfx, 2023-08-06:
Prüfen ob der passende Treiber für das Board installiert wurde, bzw. WIN10/11 überhaupt merkt das was am USB Board angeschlossen ist.
beliebte Fehler:
- USB Kabel nicht voll belegt (nur Ladekabel)
- Treiber nicht installiert
- keine Admin
- falscher Browser (Chrome oder Edge nutzen)
- 
Ich empfehle immer das Board zuerst einmal mit den ESP Web Tools zu flashen, da man hier recht schnell und einfach prüfen kann ob das Board schrott ist und auch recht einfach den WLAN Connect einstellen und testen kann!

DonKracho, 2023-08-07:
Hatte das gleiche Problem mit ahoydtu_dev.zip vom 06.08.2023 index.html und einem Wemos D1 Mini. Konnte den Chip angeblich nicht initialisieren. Ich hatte erst eine Homeassistant Asguard Instanz in Verdacht, aber auch mit Asguard ausgeschaltet ging es nicht. Online über den Coonect Button von https://ahoydtu.de/getting_started/ funktionierte das Flaschen dann.

flyer213, 2023-08-07:
Ich hatte den Usbport getauscht, ich konnte dann 2St. flaschen, nun ist es aber wieder nicht möglich. Treiber war von Ch341, Boards wurde als Com 3 dargestellt.Der Fehlercode wieder 43, ggf. Timingproblem.

closed 2023-08-17 by lumapu