lumapu/ahoy #1073: [Bug] 0.7.24 - 0.7.25 - Last success Topic wird pro Sekunde gefeuert

Huschtegutsel, 2023-08-06:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.25

### Github Hash

-

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Standard

### Error description

MQTT Publish steht auf default 0 (publish bei Änderung)

Last success Topic wird pro Sekunde gefeuert... geprüft am Broker/Visu und verbundenem MQTT-Explorer....
MQTT-Client-ID eingetragen, meldet sich damit aber nicht an

[image]

knickohr, 2023-08-06:
Ja, da war wohl einer schneller 😉

Aber das wird nicht im Sekundentakt rausgekotzt, sondern bei mehreren Invertern gefühlt im Millisekundentakt 😱

Macht mir den Broker dicht, die Emofangs-LED geht gar nicht mehr aus 😵

Und die Schwuppdizität im WebGUI läßt, vermutlich deswegen, auch wieder zu wünschen übrig 😢

Ganz schnell fixen (nein, ich schreibe besser, hat keine Eile 😂), sonst ist es mit der neuen Release nicht weit her !

[image]

lumapu, 2023-08-06:
😅

lumapu, 2023-08-06:
das erklärt natürlich auch warum sich mein ESP32 sich so träge anfühlt 😀 Fix ist schon in Arbeit

lumapu, 2023-08-06:
Build ist in Arbeit - in 10-15min könnt ihr die `0.7.26` installieren. Entschuldigung für diesen blöden Fehler und vielen Dank für das schnelle reporten!

keinprofi, 2023-08-06:
0.7.26: Fehler ist bei mir noch da:
[image]

sieht dann immer wieder so aus:
[image]

lumapu, 2023-08-06:
ist dein Wechselrichter mit dem Netz verbunden? Sieht für mich so aus, als würde da die Verbindung fehlen

keinprofi, 2023-08-06:
nach Downgrade auf 0.6.9 läuft es wieder einwandfrei (bis auf "total").

knickohr, 2023-08-06:
Was zeigt das braune/orange Feld in der WebGUI ?  Oder haste nur einen Inverter dran ?

knickohr, 2023-08-06:
> das erklärt natürlich auch warum sich mein ESP32 sich so träge anfühlt 😀 Fix ist schon in Arbeit

Ja, die Schwuppdizität ist ein sehr gutes Indiz für Programmierfehler 😂

knickohr, 2023-08-06:
[image]

lumapu, 2023-08-06:
> Ja, die Schwuppdizität ist ein sehr gutes Indiz für Programmierfehler 😂

ich finde das Wort "Schuppdizität" so gut, es passt einfach genau zum Gefühl

knickohr, 2023-08-06:
https://www.it.niedersachsen.de/startseite/it_news/aktuelles/schwuppdizitat-180256.html#:~:text=Der%20Begriff%20stammt%20urspr%C3%BCnglich%20aus,Computers%20beziehungsweise%20eines%20elektronischen%20Ger%C3%A4tes.

Ach übrigens, bei den Releases fehlt wieder mal der S3-Ordner 😉

lumapu, 2023-08-06:
komisch, habe das S3 schon ewig nicht mehr getestet - da muss ich die Tage mal die Staubschicht runterkratzen 😉

knickohr, 2023-08-06:
Mit der 26 ist das ursprüngliche Problem bei mir jetzt weg 👍

keinprofi, 2023-08-06:
Habs nochmal getestet mit der 0.7.26:
[image]
[image]
[image]

Verbindung ist immer wieder weg.
Nach Update keine Einstellungen geändert. Nach Downgrade ebensowenig und funktioniert (außer eben die Daten von "total")

Willy541, 2023-08-06:
Leider auch mit Version 0.7.26 noch das gleiche Problem, wieder downgrade auf 0.6.29.

knickohr, 2023-08-06:
Jetzt mal langsam. Der Issue beschreibt jetzt doch schon mindestens 3 Probleme !

@Willy541  :  Um was geht es ?  Gibt’s dazu einen eigenen Issue ?

@keinprofi  :  Wir machen in #1072 weiter

Ansonsten ist das Problem mit den „Last Success“ in der 26 gefixt, zu das Ding wenn @Huschtegutsel damit einverstanden ist.

Willy541, 2023-08-06:
Nach dem Update auf 0.7.25 und 0.7.26  konnte der Inverter  "Inverter #0: HoyMiles (v0) is not yet available" keine Verbindung aufbauen!
System Information
mac
7C:87:CE:2C:F3:90
sdk
v4.4.4
cpu_freq
240
heap_free
169524
sketch_used
1152
wifi_rssi
-35
ts_uptime
62
version
0.7.26
esp_type
ESP32
heap_total
292224
chip_revision
3
chip_model
ESP32-D0WDQ5
chip_cores
2
Radio
nrf24l01
is not connected
TX count
0
RX success
0
RX fail
0
RX no answer
0
RX fragments
0
TX retransmits
0

knickohr, 2023-08-06:
Also auch ein Problrm für #1072 😉

Willy541, 2023-08-06:
Nach downgrade auf 0.6.29:

System Information
mac
7C:87:CE:2C:F3:90
sdk
v4.4.4
cpu_freq
240
heap_free
189348
sketch_used
1101
wifi_rssi
-35
ts_uptime
27
esp_type
ESP32
heap_total
308424
chip_revision
3
chip_model
ESP32-D0WDQ5
chip_cores
2
Radio
nrf24l01+
is connected
Datarate
250 kbps
Power Level
HIGH
TX count
1
RX success
1
RX fail
0
RX no answer
0
RX fragments
1
TX retransmits
0

knickohr, 2023-08-06:
Überprüft bei der 0.7.x ?

[image]

Willy541, 2023-08-06:
@knickohr 

Beim Punkt "Radio (NRF24L01+)" gibt es kein Feld für einen Haken!

knickohr, 2023-08-06:
Bei den neuen Versionen schon 😉

Willy541, 2023-08-06:
OK ich teste!

Willy541, 2023-08-06:
Abgeschossen, nicht mehr erreichbar!

lumapu, 2023-08-06:
@Willy541 was heißt das jetzt?

knickohr, 2023-08-07:
Er hat sich irgendwo verklickert, das Ding bootet nicht mehr 😵

MiniOh, 2023-08-07:
Mit einem festen MQTT Intervall hat man das "Problem" nicht, richtig?

knickohr, 2023-08-07:
Wenn das so ist, dann würde ich sogar behaupten das hier tatsächlich ein Kommunikationsproblem mit dem einen oder anderen Inverter vorliegt. Das kann ich bestätigen bei meinen 13 Invertern.  Nicht jeder antwortet immer, es kommt schon mal vor das sich der eine oder andere verschluckt. Denn mit der Einstellung 0 im MQTT-Intervall werden nur Messages gesendet wenn auch was von den WRs empfangen wurde.

Willy541, 2023-08-07:
Ich konnte den AhoyDtu nicht mehr erreichen. Nicht mehr per WLAN oder mit der Notfall IP. Im Terminal stand, das ein File corrupt ist . Ich mußte flashen und habe jetzt erstmal 0.6.9 wieder drauf. Ich warte ..... (hatte natürlich kein Backup und mußte alles neu eintragen). War mir mal wieder eine Lehre!

knickohr, 2023-08-10:
Kommt das Last Success nicht noch ein bißchen zu oft ? 😲

Das kommt ja alle Intervallwert über alle Inverter.

knickohr, 2023-08-12:
Ja, ee werden Werte aus der Vergangenheit gesendet 😱

Fixed with Node-Red 😅

[image]

closed 2023-08-17 by lumapu