lumapu/ahoy #1074: ESP32 - Nach Update auf 0.7.26 keine Daten mehr vom Wechselrichter

joedezibel, 2023-08-06:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.26

### Github Hash

Ich weiß nicht was das ist

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Nichts geändert

### Error description

Neuster Release installiert. ESP32 empfängt keine Daten mehr (hat vorher einwandfrei funktioniert) 
ESP8266 im gleichen Netzwerk mit neuster Version funktioniert einwandfrei

lumapu, 2023-08-06:
schaue bitte mal in den Einstellungen nach dem Haken "NRF enable" und prüfe, ob die LEDs aus sind (unter "System Config")

joedezibel, 2023-08-06:
Danke Lukas, genau das war das Problem. "NRF Enable" war deaktiviert. Warum auch immer. Ich habe den Haken gesetzt und jetzt funktioniert es wieder einwandfrei. Vielen Dank.

closed 2023-08-06 by joedezibel