lumapu/ahoy #1075: [Bug] nach Speicher löschen und Neuem Einrichten der DTU funktioniert das ePaper nicht mehr

Loetnase, 2023-08-06:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus, square dot

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.25 und 0.7.26

### Github Hash

 cbdb150 :: 0.7.26

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Display Config Neueinrichtung nach Speicher Löschen 

### Error description

nach Speicher löschen und Neuem Einrichten der DTU funktioniert das ePaper nicht mehr.
Es fehlt in der Display Config die letzte Zeile  BUSY für den Eintrag GPIO12.
Beim Export der Settings steht im JSON file auch "busy":255, drin. 
Beim Eintragen von 12 und Importieren der JSON Datei funktioniert das ePaper wieder. 
Ist es auch möglich diese 6 Zeilen mit dem Standart vorzubelegen?

Iststand 0.7.25 und 0.7.26
[image]

alter Sollstand
[image]

knickohr, 2023-08-06:
Korrekt, der Versionssprung war zu groß 🫣

Da hat sich einiges an den Einstellungen geändert und es gab Glaubensgemeinschaft auch mal eine inkonsistente JSON. Aber Du hast das Problem ja gefixt 👍

{"type":10,"pwrSafe":true,"pxShift":false,"rotation":2,"contrast":100,"data":13,"clock":14,"cs":15,"reset":26,"busy":12,"dc":27}},"inst":

lumapu, 2023-08-06:
das wäre doch @knickohr sofort aufgefallen, wenn das epaper nicht mehr geht.
Leider ist mein epaper wohl defekt, ich kann kein Display Update mehr erzwingen - auch nicht mit den simpelsten Beispielen.

Loetnase, 2023-08-06:
Ich denke das passiert auch nur bei einem neuen ESP32 mit leeren Speicher oder beim kompletten Speicher löschen, ansonsten bleibt ja der Setup im Speicher erhalten. Siehst du und @knickohr im Display Config die 6. Zeile BUSY in der der GPIO 12 einzutragen ist? Die fehlt bei mir. Ich habe dann die JSON Datei exportiert, dort stand bei BUSY 255 drin, die habe ich auf 12 geändert und wieder importiert und ab da hat mein ePaper funktioniert. Vielleicht ist dein ePaper gar nicht kaputt sondern der GPIO 12 wird nicht mehr beschrieben

knickohr, 2023-08-07:
> das wäre doch @knickohr sofort aufgefallen, wenn das epaper nicht mehr geht.
> Leider ist mein epaper wohl defekt, ich kann kein Display Update mehr erzwingen - auch nicht mit den simpelsten Beispielen.

Nee, isses leider nicht. Nachdem die Config bei mir ja schon korrekt drin ist, hat er ja die passenden Werte. Die JSON ist auch korrekt und wird auch korrekt gesichert. Vermutlich ist nach einem Erase aber das nicht so.

Und ja, es fehlt seit der 25 die 6. Auswahl für den Busy im WebGUI 🤔

edw00d4711, 2023-08-07:
Fehlt bei mir auch.
[image]

knickohr, 2023-08-08:
Die Auswahl ist wieder da. Ich bin aber direkt zur 28 👍

Warum ich aber hier schreibe :  Die Schwuppdizität ist wieder schlechter 😱 Nicht so schlecht wie zuvor, aber die Live?Seite kann schon mal 10s auf sich warten lassen 🤔

Loetnase, 2023-08-08:
> Leider ist mein epaper wohl defekt, ich kann kein Display Update mehr erzwingen - auch nicht mit den simpelsten Beispielen.

@lumapu funktioniert dein ePaper jetzt auch wieder? 
War das der Grund, oder doch kaputt 😅

Loetnase, 2023-08-08:
> Ist es auch möglich diese 6 Zeilen mit dem Standart vorzubelegen?
@lumapu 
Ist es möglich auch die Standartbelegung vom Pinout ePaper vorzubelegen
[image]

lumapu, 2023-08-08:
@Loetnase mein epaper geht weiterhin nicht mehr, keine Ahnung was los ist, habe nichts gefunden wie man prüfen kann wo der Fehler liegt. Am gleichen ESP und Kabel geht ein 4.3" epaper problemlos. (natürlich jeweils angepasste Firmware)

Wann soll der Default geladen werden? sobald man epaper auswählt? oder einen extra Button

knickohr, 2023-08-08:
Sosooo, das 4,3“ 😅

Magst nicht als Auswahl mit einbauen ? 😇

Ja, die Vorbelegungen bitte bei der Selektion gleich machen. So isses am komfortabelsten.

Loetnase, 2023-08-08:
Ja da muss ich Knickohr zustimmen. 
Mit Auswahl des ePaper sollte die Vorbelegung der 6 GPIOs stattfinden. 
Dann kann alles manuell verändert werden.
Erst mit Abwahl und erneuter Auswahl des ePaper wird wieder neu vorbelegt. 
Danke 😊😁

lumapu, 2023-08-08:
vorbelegung baue ich ein.

Das 4.3" wird nicht (von mir) kommen, es war kein Ahoy beim Test beteiligt 😊

Das 1.54" Display hatte ich jetzt sehr lange mit dem Ahoy Logo am Schreibtisch liegen - jetzt mag es leider nichts mehr anderes anzeigen 😢

knickohr, 2023-08-09:
Das glaube ich so noch nicht. Mein zweites lag bis vor einigen Tagen auch mit dem Ahoy-Logo nur so rum. Es ließ sich problemlos wieder aktivieren. Allerdings war es komplett als DTU aufgebaut.

Und wenn : Ahoy forever ! 😇

Loetnase, 2023-08-14:
Danke für alles 👍
Ich schließe hiermit das Issue

closed 2023-08-14 by Loetnase