lumapu/ahoy #1076: Led0 & Led1

Peertje62, 2023-08-07:
Hardware;   ESP32; ESP32-D0WDQ5; nRF24L01; external antenna (SMA)
Hallo, 
I am not sure if this site is the right one to ask my question:
Have both Open Dtu and Ahoy DTU.
Flashed de Open Dtu (Blinkyparts.com)  hardware with the Ahoy software, so far so good & working

Now I want to activate the leds again, 
What values should I use ?

regards

lumapu, 2023-08-07:
I don't know the pinout for the blinkyparts.com hardware, but you can ask them for a schematics.

In Ahoy the LED0 is meant to show if one of the inverters is producing. The LED1 is used as MqTT connected indicator:

https://github.com/lumapu/ahoy/blob/cbdb15050f8d0ff619af2ce16918dffc26aa76d8/src/app.cpp#L493-L514

MetaChuh, 2023-08-09:
@Peertje62 

little led hack, that may be of interest:

**how to use a led to indicate inverter communication to ahoy dtu**:
- connect a low current led to tx on the esp
(with a resistor, about 150Ω, depending on the led)
- no further ahoy dtu settings are necessary

**how to use a led as an ahoy dtu heart beat indicator [optional]**:
- connect a low current led to tx on the esp
(with a resistor, about 150Ω, depending on the led) .
- additionally activate "print inverter data" in the ahoy dtu settings.
the heartbeat interval is the interval given at the ahoy dtu inverter interval settting.

note:
tested on esp8266, should work on esp32 using tx0 (yet unconfirmed)

greetings
metachuh

JoergGiencke, 2023-08-10:
The control of the LEDs does not work for me. I have connected LED1 (MQTT) to GPIO 34 and set it in the settings (high active). And of course I checked on the lab power supply whether the LED plus resistor work.

The DTU permanently sends data via MQTT, but the LED remains dark.

Regards,
Jörg

lumapu, 2023-08-10:
can you check if GPIO34 is a valid output for ESP? Maybe the issue is related to hardware (ESP)

JoergGiencke, 2023-08-10:
@lumapu 

Good hint, thanks! In fact, GPIO 34 and 35 cannot be configured as output. With 32 and 33 it works now.

JoergGiencke, 2023-08-11:
@Peertje62 

Works on esp32 using tx0.

closed 2023-08-16 by Peertje62

MetaChuh, 2023-08-16:
@Peertje62 
thanks for closing this issue as completed.
much appreciated gesture at all open source projects 👍

greetings,
metachuh