lumapu/ahoy #1079: HM-1500 für Nulleinspeisung nutzen - ist das Thema noch interessant?

GeThi255, 2023-08-08:
## Hardware
  * [x] ESP32
  * [x] Raspberry Pi

### Version / Git SHA: 
Version: 0.7.8

Hallo,

ich habe eine Regelung programmiert, die bei Verbrauchsänderung (gemessen mit Shelly ProEM3) das Power Limit entsprechend setzt. Bei stark schwankendem Verbrauch (oder schlechter Regelung) kann das alle paar Sekunden sein. 

Das Programm ist in PHP geschrieben und läuft auf einem Raspi 4. Eigentlich könnte man es auch gut in die Ahoy-DTU einbauen oder auf einem weiteren ESP, aber zum Testen ist es so einfacher. Die Verbrauchswerte könnten natürlich auch von einem anderen Messgerät kommen (z.B. TASMOTA SmartMeterReader/IR Lesekopf ).

Der  Shelly ProEM3 senden spätesten alle 3 Sekunden einen neuen Verbrauchswert per MQTT. Wenn ein neuer Wert vom Shelly ProEM3 kommst, frage ich die aktuelle Leistung vom Wechselrichter ab. Das habe ich es zuerst über die REST-API gemacht, läuft jetzt aber auch über MQTT. 

Dann kommt mein Regelalgorithmus, der auch auf schnelle Wechsel reagiert bzw. Überschwinger weitgehend vermeidet. Hier besteht noch Verbesserungspotential, aber funktioniert schon ganz gut. Das Ergebnis der Regelung ist die derzeit benötigte Leistung vom Wechselrichter. Diesen Wert sende ich dann über MQTT an den Wechselrichter. Wie gesagt, alle paar Sekunden. Eine Grafik ist weiter unten angehängt (Regelung auf +60W). 

Jetzt kommt das Problem: 

Mir ist aufgefallen, dass der Wechselrichter immer mal wieder abschaltet und nach ca. 1-2 Minuten wieder online ist. 
Bei meinen bisherigen Beobachtungen ist es mit meinem Wechselrichter reproduzierbar, häufig wenn z.B. eine Wolke den Ertrag auf unter 50 W reduziert und dann vorbeizieht. Wenn die Wechselrichter-Leistung dann wieder ansteigt und ungefähr die Regelgröße erreicht, schaltet der Wechselrichter ab.

Kann das jemand bestätigen?

Viele Grüße
Gerd

PS: Falls jemand an der Regelung interessiert ist oder weitere Informationen haben möchte, gebe ich sie gern weiter.
[image]

dtuuser, 2023-08-08:
Bei meinem HM 700 darf ich nie weniger als 2,8% (ca.20W) als Limit setzen, da der WR ansonsten nichts mehr produziert bis man den WR vom Stromnetz nimmt. 
Heute hat der WR bis jetzt 178 mal das Limit geändert. 10 Minuten vor Sonnenuntergang dann nochmal zum Tagesabschluss auf 100%.

MetaChuh, 2023-08-09:
@dtuuser 
das unter 5% kann ich auch für hm-800 und hm-300 (nachteinspeisung) bestätigen, drum ist bei mir auch 5% als minimum in der regelung drin.
vorher hatte ich den inverter nach 0% mit stakkato senden von inverter befehlen restart, turn off und turn on wieder „zum leben“ gebracht, ohne die dc leitung zu trennen, aber das war sehr hacky und hat meist mehrere sequenzen gebraucht, bis der inverter wieder eingespeist hat.

@GeThi255 
ja, wenn inverter weg ist oder die dtu grad rebootet, passierte das auch bei mir (ebenfalls php)
hab dann eingebaut dass ts_last_success max 120 sekunden von time() abweichen darf.
für die zeithistorien tabelle hab ich dann yieldday, der dann auch kurz (dtu reboot) oder langfristig (inverter re-power) auf null ist mit dem vorherigen aufgezeichneten tageswert verglichen und dann den vorherigen als offset gesetzt, und die neuen werte addiert.
ts_last_success verhindert dann auch, dass ein kurzfristiges yieldday null als offset genommen wird, dass dann schlagartig raufgeht und ohne ts check dann fälschlich hinzugefügt wird.

greetings
metachuh

Ollipop030, 2023-08-09:
Ich mache das hiermit:
https://github.com/reserve85/HoymilesZeroExport/tree/main

Kann jeden erdenklichen Zähler abfragen, unterstützt mehrer WR, Akkus...

Geschrieben in Python. Bei mir läuft das mit Ahoy auf dem ESP32, die Nulleinspeisung auf einem Pi3, 2 Wechselrichter. Alle 11 Sekunden wird der Stromzähler abgefragt und ggf. das Limit geändert. Ich komme so auf mehrere Tausend Änderungen pro Tag, das ganze ist absolut stabil, rennt hier 24/7 seit März. Auch Neustarts von den Wechselrichtern habe ich nie gehabt. Auch Ahoy hält das aus. Die DTU schafft locker mehrere Tage ohne Neustart, habe schon einmal 21 Tage online gehabt, und der Neustart war von mir forciert worden wg. Update.

Aber ich kann meinen Vorrednern nur zustimmen, das Limit sollte nicht zu klein gewählt werden. Ich habe bei mir 10% der WR Leistung als Minimum angesetzt.

GeThi255, 2023-08-09:
Vielen Dank für die Infos. Das untere Limit habe ich auf 100 W (6,6%) gestellt und wurde auch nie unterschritten. Die Ahoy-DTU läuft in der aktuellen Beta-Version auch sehr stabil. In vorhereigen 0.7.x Versionen hatte ich auch mehrmals täglich reboots der DTU, jetzt mit der 0.7.28 nicht.

Meine Regelung schickt zurzeit noch deutlich mehr Änderungen an der WR, was aber eigentlich immer noch langsam für „Elektronik“ sein sollte. (Ca. alle 3 Sekunden)

In den nächsten Tagen soll ja mal wieder etwas mehr Sonne scheinen, dann werde ich es noch mal testen. Wie oben in der Grafik zu sehen ist, passiert es häufig, wenn der Verbrauch höher ist als die Produktion und dann die Produktion sich dem Regelwert annähert.

Natürlich werde ich es auch mit längeren Intervallen testen. Vielleicht ist es wirklich zu schnell.

Bin mal gespannt, was ich heraus finde.

Vielen Dank auf für den Tipp zu „HoymilesZeroExport“. Hört sich nach einer guten Lösung an.

VG 
Gerd

ernstluckner, 2023-08-10:
Spannendes Thema...
Momentan versuche ich die Regelung über Shelly3EM und Home Assistent vom Synology NAS/Docker aus zu machen. YAML ist ein wenig gewöhnungsbedürftig, aber einen Stufenregler mit 100W Steps hab ich mal realisiert. Läuft momentan noch leer im Demo-Mode. Gehört noch über MQTT oder RestAPI an den Inverter rückgeführt.
LG
Ernst

DejanBukovec, 2023-08-10:
Hi everyone,
I will write my experience but English because I don't speak German...
I have 3x HM-800 Inverters, HomeAssistant with MQTT AddOn and Z-Wave power meters for House Grid, Solar and all other devices(Including all lights switches and wall outlets) :) I also have one Shelly Pro 1PM on Solar...

I need fast and responsive zero feed because if I return in grid more than 600W(Last minute average) house meter shut down and house is without power...

First I'm use OpenDTU with MQTT and HomeAssistant Automation it work pretty ok, but OpenDTU GUI stop responding because I set limit every 20 seconds...  I'm using 20 sec trigger in HA, 3 sec DTU pool and 5 sec MQTT pool. If I increase 3sec DTU pool to higher value then DTU start lagging and every time need more and more time to get inverter statistics, because queue in DTU for setlimit commands have priority and in queue has been more and more requests... Sometime also data over MQTT has been stale... My MIN value for power limit has been 10W and MAX 800W.  I didn't have any issues with that limits...

Then I'm switch to AhoyDTU and because I use some MQTT topics which Ahoy DTU don't have I'm start using HoymilesZeroExport.py script written by reserve85. Im use next settings:
```
LOOP_INTERVAL_IN_SECONDS = 5
SET_LIMIT_DELAY_IN_SECONDS = 5
SET_LIMIT_DELAY_IN_SECONDS_MULTIPLE_INVERTER = 2
POLL_INTERVAL_IN_SECONDS = 1
ON_GRID_USAGE_JUMP_TO_LIMIT_PERCENT = 0
SET_LIMIT_RETRY = 2

POWERMETER_TARGET_POINT = -75
POWERMETER_TOLERANCE = 50
POWERMETER_MAX_POINT = 0
```

[image]

And must say it work very well. BUT I run it from my PC which is not ok. I'm not sure if it can be run from HA on HAOS so I'm decide to make javascript copy of HoymilesZeroExport and run it on Shelly Pro 1PM... I need to strip a lot of things from script because scripting on Shelly is very limited... Right now script size is 24kB so it do not run on new 1.0.0 fw so I use 0.14.1 fw which do not have script size restriction...
Main of problem is that there is no way to run HTTP request in blocking mode so all request are async and need to wait for result and process it later which is not big deal if you are not limited to 5 RPC calls :) This mean if I request 3 inverter info HTTP request and in some function before I get data back call 3 power limit request script will crash with error and stop executing. Also there is no Delay or Wait function with which you can wait until some response is returned back... Also some big for loop is also not possible and shelly will reboot and disable script if it will detect usage of that... So all this make me big challenge and yesterday I'm finally make one script which do not crash but is to soon to be sure...

Why I want to run it on Shelly? Because I have old friend which have 2 Shelly devices, Shelly Pro 3EM to measure grid usage and Shelly Pro 1PM to measure Solar production without HA or some other device. So plan is that Pro1PM device will get data from Pro3EM and by script control zero feed to AhoyDTU :)

GeThi255, 2023-08-11:
Hello, 

I think that the Shelly Devices are not fast enough. Probably it is more purposeful to use a Raspi or an ESP32 for the control.

Raspi with MQTT Broker
- Listens to the two Shellys
- Can also listen to the values of the Ahoy-DTU
- Determine the new limit with the values
- Limit via MQTT to the inverter

There should be performant enough.

In my control with the Raspi I get every 3 seconds values from the Shelly Pro 3EM and via the Ahoy-DTU from the HM-1500. If from my calculation then the limit changes, I send the new limit via MQTT to the Ahoy-DTU. I have many power changes, so I send about 400-800 new limit values per hour to the inverter.

BR
Gerd

DejanBukovec, 2023-08-11:
Shelly devices are ESP32 devices :) 

Using RPI or ESP32 do not provide you Fail Safe option if your MQTT, WIFI/LAN, HA(Or some other home automation SW) stop working... In that case your solar system will in case that limiter are set higher than your power usage return to grid to much power and after 1 minute house will be without electricity(If you power provider enable limiter in meter)... In that case you need on meter hold button for few seconds to get electricity back...
If Shelly set limits then it know every time states of meters and DTU and if there is some problem it can turn off relay for few seconds and turn it back on and then inverter/s will start with persistent limit in my case is 200W so 3x200W = 600W which is under limiter in meter... So if DTU/WIFI/LAN FAIL it will produce 600W and return more in grid as usual and it will prevent turning off all house electricity...

That low pool time as you use is not possible in my case no matter on which system I run script... Why? Because communication between DTU and all 3 inverters can't be done that fast... When I’m test with 1 inverter everything work great with very short pool time then I’m add control for another 2 inverters and problem has start... DTU by my experience need around 1-2 seconds that send command to one inverter and get back reply that command is send(OK by GUI). So 3 inverters will need in best case 3-6 seconds but usually they need more because distance between DTU and inverters is few meters and metal roof add some troubles... 

Another problem is also some fluctuating devices like "induction cooking panels" which every second change power for one cooking panel from few 100W to 2000W ... And it produce spikes in power usage...

If AhoyDTU will someday have "zero feed in" option as original DTU it will be great... I think that it can be implemented regulation by MQTT or API we send current GRID power usage. In inverter settings we need add MAX power option for each inverter(Or it read it from inverter if possible). Then it calculate max power of all inverters and when grid power is changed it distribute power change to inverters... 

Right now script run ok. I don't know how can I count how many changes has been made, because solar power meter do not change only from regulation but also sun produce changes... Because of script size or some bug HTTP editor on Shelly stop working(Script code is lose and name is changed to script_1) but script work normally...

This is my today script test. My setting is that it try keep power between -25W and -125W:
[image]

3gfisch, 2023-08-19:
I have a similar setup AhoyDTU, HM1500, Shelly Pro 3EM, HAOS on raspberryPi. I do the controlling part in a small node red script. Works well so far but im never going over 600W to be compliant with the „Balkonkraftwerk“ rules..
Are there really house meters which can completely shut off the house oO i thought they can only send commands to heat pumps to stop but not that they can include 64 or 128A switches in the meter.

beebee, 2023-08-24:
Maybe #964 can help. I do a Zeroexport / Nulleinspeisung with 20s reaction time (timed trigger) with HomeAssistant via MQTT to Ahoy-DTU which aside from Panels also has a Hoymiles HM-400 for unloading the battery attached to Ghetto ESS Battery (a Powerstation for which I trigger the charging on Overproduction separately). So Panel run always full power and I start charging at fixed rate when above a limit currently (also supporting with concurrent unloading). So its a pure AC based system with the components I had available.
Also using Shelly 3EM for Household power measuring and Shelly Plug S for the (multiple) Inverters for the PV panels on East/West Balconys.

GeThi255, 2023-09-06:
Hallo,
vielen Dank für die Hilfe und die vielen Informatinen. Zu dem ursprünglichen Problemen (Wecheslrichter hängt für 1-2 Minuten) kann ich jetzt sagen, dass die Probleme mit dem Wechselrichter mit den aktuellen Ahoy-Dev Versionen nicht mehr da sind. Scheinbar hat sich irgendetwas zum besseren geändert.
Viele Grüße 
Gerd

GeThi255, 2023-09-06:
@DejanBukovec

Hello DejanBukovec,

_... Using RPI or ESP32 do not provide you Fail Safe option if your MQTT, WIFI/LAN, HA(Or some other home automation SW) stop working..._ 

This ist really a good point. I've never used Shelly scripts but I'm very interesting on your solution. Would you please share the code or some basics of the code?

BR
Gerd

DejanBukovec, 2023-09-07:
@GeThi255 I will upload script to github. Right now is very big(over 20kB) and is not supported by latest 1.0.0 firmware but support multiple main power usage sensors(MQTT, Shelly Pro 3EM, Shelly Pro 1PM), multiple inverters and multiple DTU's(For example assign different DTU's to different inverters)...
Right now Im on vacations until middle of month and after that I need resolve some new discovered issues with script...

Burton1224, 2024-03-04:
Ich weis nicht ob das jemand gesagt hat aber bei dem Wechselrichter solltest du immer nonpressistant (nennt sich so bei Hoymiles) senden damit der Speicher des WR nicht jedes mal vollgeschrieben und überschrieben wird sonst ist der Speicher irgendwann hinüber.

bytesinmotion, 2024-07-24:
> @GeThi255 I will upload script to github. 

Hi Dejan. Could you provide us with your Shellyscript? I want to make an automation with my Shelly EMPro too thanks!

bytesinmotion, 2024-07-24:
@DejanBukovec 
I will upload script to github.

Hi Dejan. Could you provide us with your Shellyscript? I want to make an automation with my Shelly EMPro too thanks!

DejanBukovec, 2024-08-02:
Hi @bytesinmotion 
Sorry for late reply...

Here it is:
https://github.com/DejanBukovec/Shelly_Script_Hoymiles_ZeroExport

But around one year I do not use it anymore so maybe is not compatible with latest AhoyDTU FW and need adjustments. Im make zero-export system based on ESP32 which collect data from sensors and control zero export on 3 hoymiles HM-800 inverters...

michaels-github, 2024-10-10:
In case you want shelly to send more often a power value you need to configure it. I use 1 sec.

300 = 5 minutes
1 = every sec
http://x.x.x.x/settings?mqtt_update_period=300