lumapu/ahoy #1081: [Bug] Yield Day reset after AhoyDTU restart

robson80, 2023-08-10:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.26

### Github Hash

cbdb150

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

- HM-1500 microinverter installed
- Power Limit set to 100%
- MQTT server setup
- Interval set to 30s
- SSD1306 installed

### Error description

Since I have updated from v0.6.9 to v0.7.26 the "Yield Day" values are reset after every AhoyDTU restart. The values also reset even I did not change anything in the configuration.

This behaviour has not been observed with prior (stable) releases.

lumapu, 2023-08-16:
where are your values reset? On your MqTT broker?

robson80, 2023-08-17:
In the Web UI and thus also resetted values are tranaferred via Mqtt.
However, I just noticed that my System is now up for approx. 6 days and the values are still low. Like it is resetting frequently without rebooting the system. 
[image]

robson80, 2023-09-04:
I have downgraded to 0.6.9 again and Yield Day values are again working fine.

lumapu, 2023-09-04:
I would prefer to test the lastest relase `0.7.36`.

robson80, 2023-09-04:
> I would prefer to test the lastest relase `0.7.36`.

Sorry, I missed to mention that. I also installed 0.7.36 after it has been released. But without any effect. V0.6.9 is currently the last version where my YieldDay values will not be resetted at random times.

lumapu, 2024-01-02:
can you check and report again with latest release `0.8.36`?

robson80, 2024-01-06:
Just updated to 0.8.36. Will provide an update within the next days.

Thanks @lumapu for all your efforts!

robson80, 2024-01-07:
Unfortunately, my results show that this bug is still existing. Please see screenshots.

However, I am not able to replicate with every reboot. Sometimes the "Yield Day" value shows up after reboot and has not been resetted. Sometimes it resets to 0.

Also I did not experience anymore that the values will be resetted after random times during the day. Only after reboots.

Before reboot:
[image]

After reboot:
[image]

Not sure if it is worth to mention, but each "Yield Day" value for all of my modules has also been resetted to 0 (see screenshots).

lumapu, 2024-01-17:
do you ever triggered a inverter restart from MqTT? Maybe on reboot it subscribes some "old" topics which say to reboot the inverter. Normally only a inverter reboot resets such values.

robson80, 2024-01-18:
Not yet I think. How do I do that?
Adding some values to the restart objects in ioBroker does not force the inverter to restart.
[image]

lumapu, 2024-01-18:
I must test it on my end first, I'm not using this feature 😉
But anyway: The "folder" "restart" was empty before your tests? What is the value of TX counter in `/system` in section `MqTT`?

lumapu, 2024-02-29:
I think it should be solved by our newest development release. There was an issue with "reset values once inverter isn't available"

closed 2024-05-30 by lumapu