lumapu/ahoy #1082: HM / HMS Firmware Update

MiniOh, 2023-08-11:
Hallo, 

es kann sein, dass dies bereits diskutiert wurde, oder auch, dass es eine echt blöde Anfrage ist.
Aber ist es denkbar, dass ein Firmware Update der Inverter über eine Ahoy DTU umsetzbar wäre?

Danke schon mal vorab.

knickohr, 2023-08-11:
Wurde schon des Öfteren diskutiert.

Aber denke mal nach :
- erstens ist es nicht möglichst ohne Account beimHeymeiles
- zweitens, willst du das wirklich ?  Wenn Heumeiles das spitz kriegt 😲
- drittens, wird es dann nicht lange dauern das sie uns die Suppe versalzen und komplett unsere DTUs stilllegen indem sie eine inkompatible Firmware in die WRs rein schieben

Aldo Ententaktik :  Schön über Wasser ruhig halten und unter Wasser kräftig rudern 😅

rmayergfx, 2023-08-12:
Volle Zustimmung. Nicht alles was machbar erscheint sollte man auch wirklich tun. Firmware Update, wofür? Wenn es eine Sicherheitslücke gibt, dann ja, ansonsten wenn das System stabil läuft - Finger weg. TSUN hat auf seinen neuen Geräten ja auch schon die Talent-Plattform integriert. Siehe auch: https://github.com/lumapu/ahoy/issues/939

MiniOh, 2023-08-12:
> Wurde schon des Öfteren diskutiert.
> 
> Aber denke mal nach :
> 
>     * erstens ist es nicht möglich ohne Account bei Heumeiles
> 
>     * zweitens, willst du das wirklich ?  Wenn Heumeiles das spitz kriegt 😲
> 
>     * drittens, wird es dann nicht lange dauern das sie uns die Suppe versalzen und komplett unsere DTUs stilllegen indem sie eine inkompatible Firmware in die WRs rein schieben
> 
> 
> Also Ententaktik : Schön über Wasser ruhig halten und unter Wasser kräftig rudern 😅

OK, klingt logisch und sinnvoll!

MiniOh, 2023-08-14:
Wie siehts es aus mit Grid Profilen zuweisen? Ist das denkbar/sinnvoll?

knickohr, 2023-08-15:
Ich weiß nicht, was ihr immer mit den Gridprofilen habt 🤪

Die sind nun mal so und feddich, wird schon seine Gründe dafür geben. Und wenn jetzt jemand einen Solar-WR für Windkraft oder Batterieeinpeisung verwenden will, ist er meiner Meinung nach im falschen Zug 😇

Aber angucken wäre schon mal schön 😅

MiniOh, 2023-08-15:
> Ich weiß nicht, was ihr immer mit den Gridprofilen habt 🤪
>
> Die sind nun mal so und feddich, wird schon seine Gründe dafür geben. Und wenn jetzt jemand einen Solar-WR für Windkraft oder [...]

Wenn ich es richtig verstanden habe, ist es bei einem neuen Inverter so, dass erst mal kein Profil zugewiesen ist, richtig?
Wenn ich also das aktuelle VDE 4105 2018 implementieren möchte, geht das aktuell nur mit einer Hoymiles DTU.

lumapu, 2023-08-15:
ich sehe es so: Die Daten sind schon im inverter, aber sind nicht explizit einem Profil zugewiesen. Ich habe einen inverter mit Profile und einen ohne, mal sehen was tatsächlich der Unterschied ist.

MiniOh, 2023-08-15:
Ich muss zugeben, ich habe eine etwas spezielle Situtation, die eigentlich der Netzbetreiber lösen muss.
Die Netzspannung ist hier oft an der oberen Grenze (~249-251V).
Die Inverter erreichen somit hin und wieder in kleinen Peaks, die NA Schutz Grenze von 253V und schalten ab.

"Ohne" aktives Profil, findet die Abschlatung sofort statt, sobald die 253V Grenze für einen kleinem Moment erreicht wird.
Mit VDE 4105 2018 als aktives Profil, greift der Wert "10mins Average High Voltage (AHV) = 253V"
Somit würde eine Abschaltug erst dann stattfinden, wenn 10min eine Durchschnittsspannung von 253V herscht. 
Das kommt aber nie zustande. Von daher ist bei mir ein großer Unterschied ob "mit" oder "ohne" Profil.

Grundsätzlich muss hier aber der Netzbetreiber tätig werden.

knickohr, 2023-08-15:
Hast Du Daten über den Verlauf der Spannung ?  Dann schreib den Betreiber an und liefere ihm das. Er muß es korrigieren oder die Ursache beseitigen. Denn irgendwann fliegen deinen Nachbarn die Geräte müßte die Ohren wenn sie immer Überspannung haben. Spätestens wenn sich die Versicherungsfälle häufen wird’s für ihn wortwörtlich brenzlig.

Ich hatte mal einen ähnlichen Fall, war aber eine Frequenz im Netz. Neben den üblichen 50Hz und deren mehr oder minder harmonischen Oberwellen, gab es noch eine bei 3kHz mit sogar nicht sinusförmigen Verlauf. Das störte meinen Lighning-Detector weil dieser auf solche Frequenzen reagiert. Klar, spezielle Fall, interessiert eigentlich niemanden. Aber wenn solche Frequenzen mit hoher Spannung durchs Netz jagen, ist das wie bei einer Kurzwellenantenne. Die ganze Gegend wurde damit verseucht und strahlte. Nun mag man sagen, was sind schon 3kHz ?  Trotzdem ist da Energie drin, die dem Netzbetreiber geklaut wird. Ende vom Lied, eine Firma hatte in der Nähe einen  Umrichter für irgendwelche hochstromigen  Drehstrommaschinen in Betrieb, der defekt war.

Soviel zum Thema Elektrosmog, zog sich durch fast einen ganzen Stadtkreis. Mich wunderts halt das es niemand bemerkt hat, nicht mal die Telekom oder der Netzbetreiber, noch den Verursacher. Aber Tage später war es dann plötzlich weg und diese Firma hatte kurioserweise mehrere Wochen wegen Technischen Defekten geschlossen.

Offiziell gabs nie ein Statement dazu, aber der freundliche Mann vom Netzbetreiber der sich die Situation vor Ort angeschaut hat, bedankte sich Tags später bei mir 😉

MiniOh, 2023-08-15:
> Hast Du Daten über den Verlauf der Spannung ? Dann schreib den Betreiber an und liefere ihm das. Er muß es korrigieren oder die Ursache beseitigen. Denn irgendwann fliegen deinen Nachbarn die Geräte [...]

Danke für die Hinweise.
Ja, das ist alles schon seit gut 8 Monaten in der Mache.
Es haben schon Kurz- und Langzeitmessungen des Netzbetreibers stattgefunden.
Unzählige Mails und Termine.
Um Summe hat sich aber noch nichts geändert.
Aber ich bin noch dran.

closed 2023-08-17 by lumapu