lumapu/ahoy #1083: ESP und NRF24 stabilisieren

Karlo49, 2023-08-11:
Ich habe mir Gedanken zu den Abstürzen bei mir mit 2 Modulen "ESP8266 pro" mit Software 0.7.26 gemacht.
Einmal nach ca 2 Stunden liefert der MQTT keine Daten mehr.
Das Andere mal auch nach ca 2 Stunden hängt sich die Web-Oberfläche auf.

Wenn die Software in Ordnung ist kann es nur an der Hardware liegen.
Die 3,3Volt wird auf meinen Modulen vom Regler TX6211B (Code DE=A1D) erzeugt.
Laut Datenblatt hat der Regler einen typ. Ausgangsstrom von 400mA bei 1 Volt Differenz.
Bei 1,7Volt Differenz (5V - 3,3V) ergibt das einen Strom von 235 mA.
Das reicht für ESP8266 und dem nRF24L01+ selbst bei voller Sendeleistung.
Wenn es nicht der max. Strom ist, können es nur Ein- und Ausschaltspitzen des Senders sein, die den Prozessor in den Wald schicken.
Ich habe einfach einen 2 Ohm Widerstand in die 3,3 Volt Leitung zwischen ESP und Funkmodul gelötet und jeweils ein 100µF Elko an ESP und Funkmodul befestigt.
Die Spannung fällt von 3,31 auf 3,28 Volt ab. Das ist unproblematisch.
Mit dieser Beschaltung läuft das erste Modul mit Software 0.7.26 und MQTT seit 24 Stunden ohne Probleme.
Der heap_frag ist weiterhin auf 25.
Das 2. Modul hat auch den 24 Stunden _Test bestanden mit MQTT und Power Level LOW.
Als 3. Modul ein ESP8266 mini mit MQTT und Power Level HIGH über 24 Stunden ohne Probleme.
Das 3. Modul läuft jetzt mit Power Level MAX seit über 4 Stunden.

Also durch die Beschaltung der 3,3 Volt Verbindung mit 100µF Elko 2 Ohm Widerstand und wieder 100µF Elko zwischen ESP und NRF24 ist AhoyDTU stabiler geworden und man kann alle Power Level nutzen

Karlo49, 2023-08-17:
Der Test mit dem 3. Modul auf Power Level MAX lief über 24 Stunden ohne Probleme.

Weitere Versuche.
Danach habe ich die Elkos durch 10µF Tantalkondensatoren ersetzt.
Der nachfolgende Test auf Power Level HIGH stürzt nach ca. 15 Stunden ab.
Nun habe ich den Widerstand durch eine Drossel 68µH und die Kondensatoren auf 47µF Tantal ersetzt.
Mit dieser Beschaltung liefen alle Test von Power Level LOW bis MAX ohne Probleme.

Eine Drossel und Tantalkondensatoren filtern Schaltpulse besser als ein Widerstand und Elkos.

Drossel 68µH, 410mA, 1,35 Ohm  0,32€
https://www.pollin.de/p/drosselspule-smcc-axial-68-h-410-ma-250259

Tantalkondensatoren 47µF 16V  a 0,59€  1,18€
https://www.pollin.de/p/suntan-tantal-elko-ts19001c101msb0b0r-47uf-16v-rm-2-54-radial-20-211112

Mit dieser Beschaltung sind alle Hardwarevarianten sicher durch Abstürze durch Schaltpulse auf der 3,3 Volt Leitung

lumapu, 2023-08-17:
voll genial was du hier untersuchst. Leider werden deine Tipps die wenigsten beherzigen.

Welche Art von Absturz attestiert der ESP in diesem Fall eigentlich? Ich meine die reboot reason unter "System"

kiu77, 2023-08-18:
Wenn Du das Problem wirklich erschlagen willst, @Karlo49 , dann würde ich einen anderen Pfad einschlagen:
Einen eigenen 3,3V-Regler, der direkt von 5V versorgt wird. Dann kann es keine Rückwirkungen auf den ESP geben.
Eingangsseitig am Regler reicht da ein C mit 1µF, ausgangsseitig wäre ein Keramik-SMD-C mit 10µF völlig ausreichend.
Es eignet sich quasi jeder Regler. Der von Dir angegebene Regler hat tatsächlich eine etwas schlechte ripple rejection bei hohen Frequenzen, welche durch das Sendemodul entstehen können.

Karlo49, 2023-08-18:
@kiu77 
Danke für dein Vorschlag. Aber das Problem ist mit 3 bis 4 facher Sicherheit behoben.
Ein aktiver Regler kann auch wieder Probleme machen bei nicht genügend Masse fangen die leicht an zu schwingen!
Mein Aufbau ist eine Lochrasterplatte mit fester Verdrahtung.
Mit Lochrasterplatten arbeite ich immer bei kleinen Stückzahlen und die Schaltungen leben über mehr als 10 Jahre.

kiu77, 2023-08-18:
Feste Verdrahtung ist gut. 
Aber die "Schwinggefahr" ist quasi gleich Null, auch auf ner Lochrasterplatine. Hauptsache Du platzierst die beiden Cs nahe gut dran.

Karlo49, 2023-08-18:
@kiu77 
Ich habe meine Lösung gefunden und an 3 Aufbauten mit unterschiedlichen Modulen getestet.
Jetzt läuft noch ein Dauertest mit Power Level Max über Wochen. 
Jeder Darf und kann andere Lösungen  Machen und Testen

Gerd-Watzmann, 2023-08-23:
Der Hinweis auf die Antennen ist wichtig! Möglichst externe Antenne für den NRF24, und möglichst viel Abstand zur Printantenne des ESP - wenn man WLAN verwendet. Eine Alternative ist die Anbindung des ESP über Ethernet, dazu gibt es hier ja eine eigene Seite:

https://github.com/lumapu/ahoy/issues/886

Karlo49, 2023-08-23:
Zwischenstand Dauertest
Modul ESP8266 Mini mit NRF24L01+ beide mit Antenne auf Platine, Antennenabstand ca 4,5 cm.
3,3 Volt gefiltert mit 47µ Tantal, Drossel 68µH und wieder 47µF Tantal
Power Level MAX. MqTT ist an.
Nach 7 Tagen keine Auffällichkeiten. Läuft sehr geschmeidig.
Power Level MAX ist nicht nötig, nur für den Test.
[image]

oechsler, 2023-08-23:
@Karlo49 - Vielen Dank für deine Mühen!  Da ich selbst eher von der Software komme und wenig mit Elektrotechnik am Hut habe, ist dieses Issue wie eine Wohltat für mich! Klage über genau das selbe Problem. MQTT stirbt, Webserver gibt auf 1:1 gleich.  Trotz 100µF Kondensator am NRF Modul. Habe nun die Schaltung wie beschrieben nachgebaut. Testen geht leider erst bei Sonnenaufgang.

- Mal so fürs Protokoll: Bei mir fingen die Probleme an, seit ich zwei Wechselrichter über eine DTU auslesen wollte.

oechsler, 2023-08-25:
> @oechsler Und hat alles geklappt? Es wäre schön, wenn ich eine Rückmeldung bekomme nach Einbau meines Vorschlags!

@Karlo49  Läuft bis jetzt seit gestern ohne Probleme.

DanielR92, 2023-08-25:
Kurze Frage, ist das so gemeint?

[image]

Karlo49, 2023-08-25:
@DanielR92 
Ja genau so. Bei dem Widerstand besser 2,2 Ohm (ist der Normwert)
oder noch besser 68µH Drossel anstelle von dem Widerstand.