lumapu/ahoy #1084: [Bug] total adds old power values from inverter not producing anymore

LeSpocky, 2023-08-11:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.26

### Github Hash

cbdb150

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

- Inverter 0: HM-400
- Inverter 1: HM-600

### Error description

Currently the panels are positioned in a way so those connected to the HM-600 stop producing an hour earlier than the panel connected to the HM-400.  However on the ahoy DTU webpage the last value the DTU got from the HM-600 is still added to the P_DC total (orange box on top).  This makes no sense.  That value should only be added if the inverter is available.  See the attached screenshot and especially the different times of last successful transmission.  On MQTT this seems to work better, round about 10 minutes after the last successful transmission I see "available" going to 0, while on the web page even hours later the last values are shown _and_ considered for the calculation of the total power values.

[image]

knickohr, 2023-08-12:
Es dauert 15 Minuten bis die DTU merkt das ein Inverter offline ist. Solange stehen alte Werte noch an. Es kann aber auch sein das der Inverter nicht mehr ganz runter gefahren ist, dann bleiben noch alte Werte hängen, sprich der Inverter hat keine 0 gesendet. Habe ich bei meinen 13 Invertern häufig. Da kann aber die DTU nichts dafür, es kam halt keine Null.

Lösung :
In den Invertersettings den Haken „ Reset values when inverter status is 'not available'“ setzen. Dann ist spätestens nach den 15 Minuten nachdem der Offline (available = 0) erzwungen wird, der Wert 0. Aber prüfen, ich glaube mit diesem Setting ist noch ein anderer Bug vorhanden.

LeSpocky, 2023-08-12:
> Lösung : In den Invertersettings den Haken „ Reset values when inverter status is 'not available'“ setzen. Dann ist spätestens nach den 15 Minuten nachdem der Offline (available = 0) erzwungen wird, der Wert 0. Aber prüfen, ich glaube mit diesem Setting ist noch ein anderer Bug vorhanden. Siehe Issue #1077

Das habe ich versucht, hat aber nicht funktioniert.  Danach sehe ich praktisch ständig 0 W in der Live View, vielleicht wirklich wegen #1077, ggf. aber auch, weil immer mal wieder keine Antwort von einem der Inverter kommt?  Ich hab zwar kaum *RX fail*, aber *RX no answer* ist der Wert immer größer null.  Ich schaue mal, wie es ist, wenn ich das Power Level erhöhe.

Radio
nrf24l01+
is connected
Datarate
250 kbps
Power Level
LOW
TX count
32
RX success
29
RX fail
0
RX no answer
3
RX fragments
65
TX retransmits
4

knickohr, 2023-08-12:
Das no answer darf kommen, ist kein Verbrechen. So bis 10% ist normal.

LeSpocky, 2023-08-15:
Das ganze setzt natürlich voraus, dass bspw. die Werte für YieldDay auch Mitternacht zurückgesetzt werden. Bei mir heute dad "Reboot at midnight" nicht geklappt und jetzt steht bei dem Inverter, der noch nicht online ist, noch der YieldDay Wert von gestern. Den zu addieren wäre jetzt natürlich auch Quatsch.

knickohr, 2023-08-15:
Reboot ist eigentlich der falsche Weg um die Yields zu resetten, besser über Inverter „Reset Yields at Midnight“. Ich weiß aber nicht ob das auch in der WebGUI passiert, die Einstellung ist für MQTT, oder @lumapu  ?

(Bei mir macht er das jedenfalls auch im GUI)

LeSpocky, 2023-08-15:
Reboot mache ich, damit er mir nicht zu zufälligen Zeiten in den Watchdog rennt.  Sonst wäre "Reset Yields at Midnight" natürlich sinnvoller.  Wobei eigentlich nur YieldDay zurückgesetzt werden müsste, YieldTotal könnte ja eigentlich bleiben, oder?

knickohr, 2023-08-15:
YieldTotal bleibt auch 😉

LeSpocky, 2023-08-16:
> Das ganze setzt natürlich voraus, dass bspw. die Werte für YieldDay auch Mitternacht zurückgesetzt werden. Bei mir heute das "Reboot at midnight" nicht geklappt und jetzt steht bei dem Inverter, der noch nicht online ist, noch der YieldDay Wert von gestern. Den zu addieren wäre jetzt natürlich auch Quatsch.

Da "Reboot at Midnight" wiederholt nicht mehr funktioniert, hab ich dafür ein neues Issue aufgemacht: #1097

closed 2023-08-17 by lumapu