lumapu/ahoy #1085: MH-300 Send Power Limit: Error: Deserialization failed - Restliche Kommunikation funktioniert

alauchner, 2023-08-12:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [X] I will attach/upload an Image of my wiring

### Version

0.7.26

### Github Hash

cbdb150

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Ich habe keine Default-Einstellungen verändert. Die Entfernung zwischen ESP und Inverter liegt bei 1 Meter, das ist noch ein Testaufbau.

### Debug Serial Log output

```shell
14:16:54 I: (#0) enqueued cmd failed/timeout
14:16:54 I: (#0) resetPayload
14:16:54 I: (#0) Requesting Inv SN 112184710868
14:16:54 I: (#0) enqueCommand: 0x0b
14:16:54 I: (#0) prepareDevInformCmd 0x0b
14:16:54 15 pid: 80
14:16:54 I: TX 27B Ch23 | 15 84 71 08 68 84 98 08 50 80 0b 00 64 d7 78 36 00 00 00 01 00 00 00 00 92 a3 82 
14:16:54 I: RX 27B Ch23 | 95 84 71 08 68 84 71 08 68 01 00 01 01 31 01 06 03 1d 00 00 00 14 00 00 09 1b ba 
14:16:54 I: RX 27B Ch23 | 95 84 71 08 68 84 71 08 68 82 13 88 02 f9 00 00 00 21 03 e8 01 20 00 02 d3 2b 66 
14:16:54 I: procPyld: cmd:  0x0b
14:16:54 I: procPyld: txid: 0x95
14:16:54 I: Payload (30): 00 01 01 31 01 06 03 1d 00 00 00 14 00 00 09 1b 13 88 02 f9 00 00 00 21 03 e8 01 20 00 02 
14:16:54 I: alarm ID incremented to 2
14:16:54 I: (#0) enqueCommand: 0x11
14:16:59 I: Iv: 0
14:16:59 I: MH 300/ch1/U_DC: 30.500 V
14:16:59 I: MH 300/ch1/I_DC: 2.620 A
14:16:59 I: MH 300/ch1/P_DC: 79.700 W
14:16:59 I: MH 300/ch1/YieldTotal: 0.020 kWh
14:16:59 I: MH 300/ch1/Irradiation: 199.250 %
14:16:59 I: MH 300/ch0/U_AC: 233.100 V
14:16:59 I: MH 300/ch0/I_AC: 0.330 A
14:16:59 I: MH 300/ch0/P_AC: 76.100 W
14:16:59 I: MH 300/ch0/F_AC: 50.000 Hz
14:16:59 I: MH 300/ch0/PF_AC: 1.000 
14:16:59 I: MH 300/ch0/Temp: 28.800 °C
14:16:59 I: MH 300/ch0/ALARM_MES_ID: 2.000 
14:16:59 I: MH 300/ch0/YieldTotal: 0.020 kWh
14:16:59 I: MH 300/ch0/P_DC: 79.700 W
14:16:59 I: MH 300/ch0/Efficiency: 95.483 %
14:16:59 I: 
14:17:04 I: Iv: 0
14:17:04 I: MH 300/ch1/U_DC: 30.500 V
14:17:04 I: MH 300/ch1/I_DC: 2.620 A
14:17:04 I: MH 300/ch1/P_DC: 79.700 W
```

### Error description

Die Kommunikation von ESP und Inverter  funktioniert problemlos, ich erhalte regelmäßig in der Live Ansicht die aktuellen Leistungsdaten. Allerdings kann ich bei "Serial/Control" Commands kein Powerlimit setzen. Egal ob persistent , relativ oder absolut, es kommt immer der Fehler "CTRL Result Error: Deserialization failed". Am Inverter passiert dann gar nichts, der läuft weiterhin voll weiter. Ich brauche die Steuerung aber, da ich eine Nulleinspeisung plane.
[image]
[image]

knickohr, 2023-08-12:
Das Leerzeichen in deinem Inverternamen stört mich. Könnte vielleicht das Problem sein.

alauchner, 2023-08-12:
Hatte auch schon Namen ohne Leerzeichen, habe es noch einmal versucht mit einem neuen Namen ohne Leerzeichen, der Fehler bleibt

GeThi255, 2023-08-12:
Verwendest Du MQTT?

Dann könntest Du es mit diesen Befehlen versuchen:

# mqtt Befehle zum Steuern HM
# Limit in Watt
mosquitto_pub -u mqtt_Dein_User-P mqtt_Dein_Passwort -t 'inverter/ctrl/limit/0' -m "600W"
# Limit in %
mosquitto_pub -u mqtt_Dein_User-P mqtt_Dein_Passwort  -t 'inverter/ctrl/limit/0' -m "50%"

Natürlich User und Passwort entsprechend anpassen.

Ich habe einen HM-1500 und verwende die 0.7.30 ohne Probleme.

VG
Gerd

alauchner, 2023-08-12:
Bislang habe ich nur die Ahoy-Weboberfläche des ESP zum Testen verwendet. Solange das direkt nicht funktioniert, wollte ich  nicht von außen den ESP mit MQTT oder der Rest API ansprechen.  Ich habe mir auch den Sourecode angesehen. Die Fehlermeldung kommt aus der RestApi beim Parsen des Webrequests. Ich versuche morgen einmal mein Glück über den Aufruf aus einer externen Quelle.

alauchner, 2023-08-13:
Ich habe es jetzt über die Rest API zum Laufen gebracht. Aber: In der Doku steht, die URL des Rest-Endpunkts wäre "IP_Adresse/api". Damit geht es nicht. Die richtige URL ist "IP_Adresse/api/ctrl".

lumapu, 2023-08-13:
danke für den Hinweis mit der Doku, wird angepasst.

lumapu, 2023-08-13:
@alauchner wäre es möglich eine Web-Console mitlaufen zu lassen und hier den Fehler zu posten? In Firefox zB. `Extras->Browser-Werkzeuge->Werkzeuge für Web-Entwickler`?

alauchner, 2023-08-14:
Ich habe jetzt die Kommunikation mitgeschnitten. Die ersten drei Bilder sind das Setzten des Limits, was nicht funktioniert. Dann habe ich den Inverter über die Weboberfläche abgeschaltet, das hat funktioniert. Für mich sehen beide Requests ziemlich ähnlich aus. 

[image]
[image]
[image]
[image]
[image]
[image]

pamir93, 2023-09-10:
I had a similar problem with a HM600; it turned out that it depended on the used browser.
It worked with cromium and firefox beta (android) but not with samsung internet (android).
Configuration:
AHOY DTU on esp8266, fw git sha ba218ed::0.7.36
Chromium: 114.5735.196 ok
Firefox beta: 115.0b9 ok
Samsung Internet: 22.0.3.1 nok

stefan123t, 2024-10-29:
@alauchner kannst Du das nochmal mit einer aktuellen Version 0.8.2-152 überprüfen ob das Problem bei Dir noch existiert ?
Bitte auch mal den Vorschlag von pamir93 mit einem anderen Browser ausprobieren und dann hier berichten. Danke!

@lumapu die Dokumentation ist angepaßt oder ?