lumapu/ahoy #1087: Weboberfläche schlecht / nicht erreichbar mit MI-700 Inverter und Fusion Board

AKL-27, 2023-08-13:
### Platform

ESP32

### Assembly

the DTU was already assembled

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.30

### Github Hash

995c7ea5fbe72450009026a79f5a3998417dc06b

### Build & Flash Method

ESP Tools (flash)

### Setup

Modelname: OpenDTU Fusion - Community Edition
Retailer URL: https://shop.allianceapps.io/products/opendtu-fusion-community-edition

### Debug Serial Log output

```shell
11:29:14 I: [NTP]: 2023-08-13 09:29:14 UTC
11:29:14 I: MQTT connected
11:29:16 I: (#0) resetPayload
11:29:16 I: (#0) Requesting Inv SN 104152000xxx
11:29:16 I: (#0) enqueCommand: 0x01
11:29:16 I: (#0) enqueCommand: 0x09
11:29:16 I: (#0) prepareDevInformCmd 0x01
11:29:16 0f pid: 00
11:29:16 I: TX 11B Ch61 | 0f 52 00 09 44 82 15 12 52 00 c7 
11:29:17 I: (#0) enqueued cmd failed/timeout
11:29:17 I: (#0) resetPayload
11:29:17 I: (#0) Requesting Inv SN 104152000xxx
11:29:17 I: (#0) prepareDevInformCmd 0x09
11:29:17 09 pid: 09
11:29:17 I: TX 11B Ch75 | 09 52 00 09 44 82 15 12 52 09 c8 
11:29:17 I: RX 24B Ch40 | 89 52 00 09 44 52 00 09 44 01 40 00 23 09 4f 13 8b 04 33 00 bb 01 53 eb 
11:29:17 I: (#0) sth. missing: Request Retransmit 0x09
11:29:17 09 pid: 09
11:29:17 I: TX 11B Ch3 | 09 52 00 09 44 82 15 12 52 09 c8 
11:29:17 I: (#0) next request is 0x11
11:29:17 11 pid: 11
11:29:17 I: TX 11B Ch23 | 11 52 00 09 44 82 15 12 52 11 c8 
11:29:17 I: (#0) sth. missing: Request Retransmit 0x11
11:29:17 11 pid: 11
11:29:17 I: TX 11B Ch40 | 11 52 00 09 44 82 15 12 52 11 c8
```

### Error description

Habe das Fusion Board per Flash-Download-Tool (inkl. Erase) mit AhoyDTU V6.15 geflasht. Inverter ist ein älterer MI-700. Mit dieser Konstellation lief es wochenlang gut. Auch mit MQTT, aber ohne Display.

Nach diversen Update Versuchen (per OTA) auf Versionen 7.xx bis aktuell 7.30 zeigt sich immer das selbe Phänomen. Die Weboberfläche ist nicht oder nur nach langer Wartezeit erreichbar. Die Liveansicht zeigt immer nur 0 an. Das Board wird mit einem iPad 15W Netzteil versorgt und liegt 2m neben dem MI-700. Serialnr. habe ich hier verändert.

Habe ich etwas falsch konfiguriert? Ist die MI-Serie nicht mit Ahoy Versionen größer 6.15 kompatibel?

[image]

[image]

rmayergfx, 2023-08-13:
Bitte mal beide LED auf OFF stellen:
[image]
und dann die GPIOs kontrollieren.

knickohr, 2023-08-13:
3 Fragen hierzu :
- Du hast die richtige Firmware aus dem S3 Ordner genommen ?
- Das Setup (Hardware) hat sich nicht geändert, also nichts dazu gemacht oder anders ?
- Display angeschlossen ?  Welches ?  MQTT aktiv ?

AKL-27, 2023-08-13:
@rmayergfx, Das Board verwende ich, da ich mit meiner selbstgebastelter Einzel-Hardware nicht zufrieden war. Das Fusion Board hat alles drauf, was ich benötige und macht einen soliden Eindruck. Es ist mir sehr sympathisch.

- Die Antennen sind laut Anleitung richtig angeschlossen. Haben ja auch mit der 6.9 und mit 6.15 funktioniert.
- LED's OFF oder ON macht keinen Unterschied. Habe beides getestet.

@knickohr,

- Die Firmware ist korrekt aus dem S3 Ordner. Hatte mal die normale ESP32 Firmware versucht, da kommt beim Update gleich eine Fehlermeldung.
- Am Setup hat sich nichts geändert. Die 6.15 Konstellation lief normal. Dann per OTA Update auf 7.30. Dann tritt der Fehler auf.
- Display habe ich keines angeschlossen. MQTT ist aktiv. Habe es auch schon ohne MQTT konfiguriert, macht aber keinen Unterschied. Fehler unter den V7.xx ist immer da.

Wenn ich NRF in den Settings deaktiviere, ist die Weboberfläche wieder normal zu bedienen. Nur gibt es dann eben keine Daten :-)

Ich meine, zwischen der 6.15 und den 7.xx Versionen wurde mal die NRF lib erneuert. Kann ich in den Changes leider nicht mehr finden. Gefühlt besteht das Problem seit dem.

lumapu, 2023-08-13:
wenn man den Inverter-Typ von 1041 auf 1141 ändert, dann werden andere Routinen innerhalb von Ahoy verwendet.

rejoe2, 2023-08-16:
@AKL-27: Falls du das selbst bauen kannst, könntest du auch gleich https://github.com/lumapu/ahoy/pull/1095 mit testen...

Das mit der geänderten Seriennummer war nur um zu sehen, ob es an der Zahl der Inverter lag oder am Typ. Offenkundig letzteres.

AKL-27, 2023-08-16:
Selbst bauen übersteigt leider meine Kompetenz.

knickohr, 2023-08-16:
Jetzt warte doch mal ein oder zwei Tage 😱

AKL-27, 2023-08-16:
Na klar doch. Bin ja froh, dass die MI-Serie überhaupt supportet wird 👍

rejoe2, 2023-08-17:
> Na klar doch. Bin ja froh, dass die MI-Serie überhaupt supportet wird 👍

THX!
Sollte mit der aktuellen "action" ab incl. 0.7.34 wieder funktionieren. Sorry, dass es etwas gedauert hat.

AKL-27, 2023-08-17:
Habe heute upgedatet auf 0.7.34. Nach reboot war sofort die Liveansicht mit Daten da. Das hatte bisher mit der 0.6.15 immer einige Zeit gedauert. Alles läuft wie gewünscht und ich bin happy. Vielen Dank für eure Arbeit @rejoe2 @lumapu

lumapu, 2023-08-17:
Super, der Dank geht an @rejoe2 😊

Die blauen Boxen waren nicht sichtbar, weil du wahrscheinlich die "max Module power" Felder nicht ausgefüllt hättest. Freut mich zu hören, dass jetzt alles funktioniert.

closed 2023-08-17 by lumapu