lumapu/ahoy #1088: developer version 0.7.30 WLAN reconnect fail

rmayergfx, 2023-08-13:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.30

### Github Hash

995c7ea

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

standard, nothing special

### Error description

Access Point mit 2.4 GHz für optimale Reichweite.
Nach mehrfachem Reboot (Issues im WLAN, nunabhängig von AhoyDTU) ist das System weder per Ping noch über die Weboberfläche erreichbar. Display zeigt Werte an und scheint auch noch mit dem Wechselrichter verbunden zu sein.
Fallback Hotspot ist nicht aktiv, bzw. wird nicht angezeigt, selbst wenn man direkt vor der Platine steht.

Hier half nur stromlos und reboot.

rmayergfx, 2023-08-18:
Habe gerade ein Update auf 0.7.34 gemacht, 20min später war AhoyDTU im WLAN nicht mehr erreichbar, auf dem Display standen aber nach wie vor Werte, also scheint die Kommunikation mit dem WR noch funktioniert zu haben. 2min. stromlos gemacht und dann lief wieder alles.

lumapu, 2023-08-18:
echt komisch, kenne solche Problem überhaupt nicht. Hast du eine Möglichkeit das per seriellem Log (über USB) zu dokumentieren?

rmayergfx, 2023-08-18:
Nein, leider nicht

closed 2024-01-02 by lumapu