lumapu/ahoy #1089: [Bug]

You69Man, 2023-08-13:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.30

### Github Hash

995c7e

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

n.r.

### Error description

Beim Compilieren von  V0.7.30 mit PlatformIO sind mir ein paar Warnings aufgefallen.
Zumindest eines davon dürfte etwas Ernsteres sein:

publisher/pubMqtt.h:528:46: warning: format '%s' expects argument of type 'char*', but argument 4 has type 'String' [-Wformat=]
  528 |                         snprintf(mVal, 40, "%s", iv->getAlarmStr(iv->lastAlarm[j].code));

Hier fehlt meiner Meinung nach ein .c_str() am Ende der getAlarmStr(), sonst wirft es die snprint Funktion, und dann kann alles mögliche passieren.
Ist lt. GIT mit V0.7.29 reingewandert....

knickohr, 2023-08-13:
Ach deshalb bekomme ich keine vernünftigen Alarmmessages raus. Hab mich schon gewundert 🤪

lumapu, 2023-08-13:
danke für den Hinweis

You69Man, 2023-08-16:
Fixed in develop verson

closed 2023-08-16 by You69Man