lumapu/ahoy #1090: [Bug] [WEBSITE] Verkabelung in den neuen Fritzings Bildern entspricht nicht den Default Werten (CE & IRQ vertauscht)

MetaChuh, 2023-08-13:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.30

### Github Hash

995c7ea

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

web related

### Error description

hi @lumapu,

in den neuen schaltplänen (mit display) auf deiner web seite sind die ce (d4) und irq (d3) leitungen vertauscht.
dh sie entsprechen nicht den default werten.
d4 gelbes kabel geht auf irq statt ce
d3 graues kabel geht auf ce statt irq

auf den alten bildern ohne display hat das noch gepasst, soweit ich mich erinnere.

info für alle neuen user:
wenn diese beiden kabel vertauscht sind, und die pinbelegung in den settings auf default sind, leuchtet die blaue led (blau beim esp8266 d1 mini) beim ersten nrf24+ kommunikationsversuch auf, und geht nicht mehr aus.

thx & greetings,
metachuh

lumapu, 2023-08-13:
ja das war extra, da wir so oft die Empfehlung geben genau diese Pins zu vertauschen. @stefan123t erinnerst du dich?
Bei vielen ESP8266 geht der Interrupt nicht auf Pin D3, weiß nicht woran es liegt. Daher war meine Idee die Pins jetzt endlich zu korrigieren.

Wie denkt ihr darüber?

MetaChuh, 2023-08-13:
schwierig zu sagen,

obwohl ich mit den bisherigen pinbelegungen nie ein problem gehabt habe, bin ich, besonders für neue user, für parallele anpassung von schaltplan und zeitgleich identischer default belegung der jeweils aktuellen firmware.

gibt’ leider viele support technische faktoren die nach hinten für uns alle losgehen können:

1) neue user haben derzeit einen schaltplan, der nicht mit den default werten übereinstimmen, was tickets/issues produziert

2) einige haben breakout boards, und wenn sie mal ihren esp erasen und neu einspielen, werden sie wahrscheinlich garnicht daran denken, dass sich die default pinbelegung geändert hat, und/oder nicht wissen, dass man sie in den settings ändern kann, was wiederum tickets/issues produziert.

und sicher viele andere sachen, an die ich grad‘ nicht gedacht hab.
deshalb enthalte ich mal einer voreiligen empfehlung, die auch nach hinten losgeh‘n kann 😉

was meint ihr ?

knickohr, 2023-08-13:
Dinge änder sich, so ist das Leben 😉

Problematisch wird das allerdings bei diesen Black Boxen wo kein Aas weiß wie verkabelt wurde. Aber das ist es eh schon, seit der 0.7.x gibt es einige Einstellungen die hinzu kamen und die geändert werden müssen. Wäre in meinen Augen nur ein weiteres Steinchen, aber man sollte es wissen.

stefan123t, 2023-08-14:
Finde ich in Ordnung, sollten wir als Breaking Change bekanntmachen und rechfertigt evtl auch einen Major-Versionssprung. Nutzer von alten DTUs bzw Boards mit der alten Default Rinstellung könnten wir mit einem Preset/Template helfen.

knickohr, 2023-08-14:
Das größte Problem ist halt die Meißten lesen keine Changelogs, nicht mal das vom Release. Ansonsten würden solche Problemchen wesentlich seltener auftreten.

Vielleicht muß man in einer der jetzigen folgenden 0.7.x eine Routine einbauen die prüft ob man neu aufsetzt oder einfach nur updatet. Würde zumindest solche Probleme wie fehlende Haken in der NRF-Auswahl quasi von selbst beheben. Oder das Spielchen mit den zusätzlichen Pinouts bei NRF und die falschen LED-Einstellungen.

Bei neu muß man zu 95% funktionierende Einstellungen vorbelegen. Welche da sinnvoll sind, steht auf einem anderen Blatt. Man kann ja nicht alles abfangen, denn dann bräuchten wir die Einstellungen gar nicht mehr 😅 Ein bißchen Hirn muß schon noch dabei sein.

Und ehrlich gesagt, auf diese Black Boxen würde ich keine Rücksicht nehmen, das sollen die Verkäufer selbst supporten. Das Geld haben sie ja auch eingeschoben.

rmayergfx, 2023-08-15:
**OFFTOPIC***
Was sich da gerade am Markt an BlackBoxen und Platinen bewegt ist echt eine Katastrophe. Zusätzlich kommen noch die ESP32 und 8266 Module mit unterschiedlicher Anzahl an PINs 30,38. Da hat die passende Doku mit und ohne Display Varianten ja schon alleine 10 Seiten mit Bildern. 
Da dies ja noch nicht reicht sind auf dem Markt dann auch noch einige Fake Module, auch bei den NRF24L01+ und wenn es kein Fake ist gibt es oft genug Probleme mit defekten Modulen.

@lumapu 
Vielen Dank für die Umfangreiche FAQ Sammlung auf der Homepage. Ich finde man sollte hier im GIT ganz oben anpinnen, das Fehler erst gemeldet werden dürfen, wann man diese abgearbeitet hat.

MetaChuh, 2023-08-16:
@lumapu 

meine empfehlung und bitte:

first step:
in der rubrik `Ahoy Einrichten`auf der [`Getting Started`](https://ahoydtu.de/getting_started/) seite,
ergänzen, dass `d3` auf `ce`, und `d4` auf `irq` gestellt werden muss.

grund:
neue user sind derzeit, ohne viel eigenrecherche, schuldlos zum scheitern verurteilt.

coding aufwand:
keiner, nur doku ergänzung um eine zeile

@rmayergfx & @knickohr 
die black boxen und plug and play breakout boards, sowie deren hersteller, sind mir ebenfalls ziemlich egal.
wichtiges ziel ist imho: für den "successful first contact" von neuen usern, gemeinsam einfache lösungsansätze schnell zu finden, und umzusetzen.

thx und greetings
metachuh

MetaChuh, 2023-08-17:
@lumapu 
perfekt, danke 👍🙏

> Auszug aus Getting Started:
>
>4 Pinout prüfen
Die Pins für IRQ und CE müssen geprüft werden. Nach den Schaltbildern auf dieser Website müssen diese für Versionen vor 0.7.34 folgendermaßen gestellt werden: CE auf D3 und IRQ auf D4

closed 2023-08-17 by MetaChuh