lumapu/ahoy #1091: [Bug] Regression: Boxes for panels not blue anymore

LeSpocky, 2023-08-14:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.31

### Github Hash

4831945

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

two inverters:

1. HM-400 with one 400 Wp panel
2. HM-600 with two 240 Wp panels

### Error description

Web interface shows boxes for the panels in green now.  Those were blue before.  Looks like a regression after trying to implement #1084 maybe?  See attached screenshot.

[image]

lumapu, 2023-08-14:
correct, I will fix this. Copy-Paste issue 😇

knickohr, 2023-08-14:
Ich möchte bitte in pink 😂

lumapu, 2023-08-14:
ja, dafür muss es aber erst mal funktionieren 🤣
Will das auch haben - Stück für Stück wirds besser.

stefan123t, 2023-08-14:
Für die AlarmEvents hat Oberfritze im Discord seine Hilfe angeboten @lumapu schau mal wie Du ihm da ggf. Beim PR helfen kannst wenn er so weit ist.

LeSpocky, 2023-08-14:
Mit v0.7.32 bzw. commit 8544e86 sieht es wieder korrekt aus.  Danke für den schnellen Fix.

cibis8, 2023-08-17:
bei mir sehen die Farben etwas grau aus :( war ab 7.30 der Fall und hat sich seither nichts verändert...
[image]

lumapu, 2023-08-17:
ja das ist auch richtig so. Das graue soll darstellen, dass der Inverter gerade nicht verfügbar ist und von Ahoy als Offline erkannt wird. Quasi ein Status-Check auf den ersten Blick 😀

closed 2023-08-17 by lumapu