lumapu/ahoy #1094: HMT-2250T: Funkverbindung wird nach Sonnenaufgang nicht wiederhergestellt

Tarnatos, 2023-08-15:
## Hardware
  * [X] ESP32

Modelname: ESP32-D0WDQ5

### CMT2300A Module

### Antenna:
* [X] external antenna (SMA)

### Power Stabilization:
* [X]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module

### Version / Git SHA: 
Version: 0.7.26
Github Hash: https://github.com/lumapu/ahoy/commit/cbdb15050f8d0ff619af2ce16918dffc26aa76d8

### Build & Flash Method:
* [X] ESP Tools

### Debugging:
* [X] Setup settings (use our templates ... to be added)

Moin,

nach Sonnenaufgang schaltet der WR ja mangels DC Spannung ab. Nach Sonnenaufgang startet dieser wieder und beginnt mit seiner Arbeit.

Ahoy-DTU verliert natürlich wärend der Nacht die Verbindung, kann diese aber am Morgen nicht wiederherstellen. Ich muss dann morgens einen reboot des DTUs durchführen. Dann läuft wieder alles.

Sunrise Daten habe ich gepflegt und auch einen Offset sowie das Häckchen bei Pause gesetzt. NTP Zeit passt.

Was kann ich tun?

Gruß
Tarnatos

lumapu, 2023-08-15:
Habe selbst 2 HMT-2250-6T und kann den Fehler nicht bestätigen. Was sagt das Serielle Log? Dafür bitte den Haken bei "Serial Debug" unter "Settings->System" setzten.

Tarnatos, 2023-08-15:
Hab serial debug mal aktiviert und melde mich dann morgen dazu wieder. 

Nein Ahoy ist das einzige DTU im System und es gibt auch keine weiteren WR.

rmayergfx, 2023-08-15:
Welche Werte für die Nulleinspeisung gibst du per Skript weiter? Manche WR vertragen es nicht wenn Werte < 10% gesendet werden.

Tarnatos, 2023-08-15:
Hier die Konfig:

[code]
[COMMON]
INVERTER_COUNT = 1
SLOW_APPROX_LIMIT_IN_PERCENT = 20
SLOW_APPROX_FACTOR_IN_PERCENT = 20
LOOP_INTERVAL_IN_SECONDS = 20
SET_LIMIT_DELAY_IN_SECONDS = 5
SET_LIMIT_DELAY_IN_SECONDS_MULTIPLE_INVERTER = 2
POLL_INTERVAL_IN_SECONDS = 1
ON_GRID_USAGE_JUMP_TO_LIMIT_PERCENT = 100
MAX_DIFFERENCE_BETWEEN_LIMIT_AND_OUTPUTPOWER = 100
ENABLE_LOG_TO_FILE = false
LOG_BACKUP_COUNT = 30
SET_LIMIT_RETRY = 10
LOG_TEMPERATURE = false
SET_POWER_STATUS_DELAY_IN_SECONDS = 10

[CONTROL]
POWERMETER_TARGET_POINT = -75
POWERMETER_TOLERANCE = 25
POWERMETER_MAX_POINT = 0

[INVERTER_1]
HOY_MAX_WATT = 2250
HOY_MIN_WATT_IN_PERCENT = 5
HOY_COMPENSATE_WATT_FACTOR = 1[/code]

Die Regelung funktioniert auch ohne Probleme den Tag über.

rmayergfx, 2023-08-15:
Es ist aber nur die Verbindung zum WR weg? Das Webinterface ist nach wie vor erreichbar?

knickohr, 2023-08-15:
Mehrere Anmerkungen :

Ich würde das Problem beim Script sehen.
- Erstens mal donnert es immer Werte raus, egal ob sie sich geändert haben oder nicht
- Zweitens sollte das Intervall nicht zu kurz sein
- Drittens es sollte auf ein Achknowledge gewartet werden bis neue, veränderte Werte anstehen und geschrieben werden
- Nachts sollte das Script natürlich auch schlafen gehen !
- Und bei einen 2250VA WR sind Minimum 2% schon 45W, darunter sollte man nicht gehen !

All diese Punkte können den WR dazu veranlassen für immer zu schlafen. Es gab schon Fälle da haben sich Anwender den WR somit abgeschossen.

Da aber bei Dir offenbar nur die DTU den Dienst einstell, warum testest Du nicht einfach mal ohne das Script, ob es mit der DTau am nächsten Morgen funktioniert ? 🤔

Ollipop030, 2023-08-15:
> * Erstens mal donnert es immer Werte raus, egal ob sie sich geändert haben oder nicht

Lt seiner Config wird 10x hintereinander das gleiche Limit gesendet, danach nicht mehr. In der neuen Version wird aber auf ACK gewartet, und schon die erste Wiederholung wird nicht mehr gesendet. Ist aber auch nicht weiter schlimm, selbst das Senden eines Limits alle 11 Sekunden mit 2 WR verkraftet die DTU locker.

> * Zweitens sollte das Intervall nicht zu kurz sein

Sind 20 Sekunden, also 3 Änderungen pro Minute. Bei mir sind es sogar nur 11 Sekunden.

> * Drittens es sollte auf ein Achknowledge gewartet werden bevor neue, veränderte Werte anstehen und geschrieben werden

Gibt es in der aktuellen Dev des Scripts.

> * Nachts sollte das Script natürlich auch schlafen gehen !

Tut es. Vor jeder Änderung wird geprüft, ob der WR online ist. Das wird (wie in diesem Fall) alle 20 Sekunden geprüft. Sobald er wieder da ist läuft die Regelung weiter.

> * Und bei einen 2250VA WR sind Minimum 2% schon 45W, darunter sollte man nicht gehen !

Das Minimum in diesem Fall steht bei 5%. ABER: Ich habe schonmal einen HM600 zerschossen (bis ich ihn komplett stromlos gemacht habe), und das mit 7%. Ich selbst gehe bei dieser Regelung nicht mehr unter 10%. Denke aber nicht, dass das Auswirkungen haben wird, denn gerade Abends versucht das Script ja mehr Leistung aus dem WR zu holen, sprich das Limit wird immer weiter erhöht, bis dann irgendwann die Sonne ganz weg ist und der WR offline ist. 

Komisch ist aber, dass das Script den ganzen Tag seine Arbeit verrichtet, und nur der Reconnect nicht funktioniert. Ich würde aber auch mal einen Tag ohne probieren.

Des Weiteren lasse ich mich aber gern eines Besseren belehren, ich kann nämlich nur für die HM Serie sprechen. Damit funktioniert es nämlich bei mir seit Monaten. Aber vielleicht sind die HMT doch empfindlicher?

knickohr, 2023-08-15:
„Lt seiner Config wird 10x hintereinander das gleiche Limit gesendet, danach nicht mehr.“

Das ist gefährlich. Habe meinen WR auch schonmal damit blockiert. Auf Ack warten und nix gleiches mehr senden. Kommt kein Sack, stimmt sowieso was nicht.

„ selbst das Senden eines Limits alle 11 Sekunden mit 2 WR verkraftet die DTU locker.“

Nicht die DTU, der WR !

„ Aber vielleicht sind die HMT doch empfindlicher?“

Das ist natürlich noch ein Fragezeichen.

Tarnatos, 2023-08-16:
Guten Morgen in die Runde.

Habe nun als ersten Schritt das Offset von 60min entfernt und bisher läuft es nun durch. Werde es die nächsten Tage weiter beobachten und noch mal reporten.

Danke für den Tip mit dem Dev Branch des Nulleinspeisungsscripts. Habe das direkt eingebaut!

closed 2023-08-17 by lumapu

Tarnatos, 2023-08-18:
Guten Morgen,

das Problem ist bisher nicht behoben. Es läuft einige Zeit, dann wieder nicht mehr.  Im Rahmen meiner Tests hatte ich auch ein dev Image drauf. Hier trat das Problem mit der instabilen Verbindung bereits nach sehr kurzer Zeit auf.

OpenDTU hatte das Problem von vorn herein, weshalb ich auf Ahoy umgestiegen bin, was wesentlich stabiler läuft.

Nichts desto trotz, war heute morgen wieder ein reboot der DTU nötig, da eine Verbindung zum WR nicht zustande kam.