lumapu/ahoy #1096: Inverter alarm [Bug]

Peertje62, 2023-08-15:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

7.33

### Github Hash

?

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

-

### Debug Serial Log output

```shell
-
```

### Error description

Hi, happy so far with 7.33
I have tested, alarms inverter, by switching off mains.
some notifictions apear twice
see picture
[image]

Peertje62, 2023-08-15:
Or am I misunderstanding, which is also possible

kind regard Herman

lumapu, 2023-08-16:
I think they should only appear once. For now the alarms are read completely everytime a new alarm is triggered.
I'm in development of this feature so I think it lasts few Dev versions before I'm done with that 😎

Please keep reporting of your findings.

closed 2023-08-20 by Peertje62