lumapu/ahoy #1097: [Bug] Regression: Reboot at midnight not working anymore

LeSpocky, 2023-08-16:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.33

### Github Hash

9b250c5

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

System config "Reboot Ahoy at midnight" is set according to webpage.

### Error description

1. Installed releases in this order:
    1. v0.5.66
    2. v0.6.0
    3. v0.6.9
    4. v0.7.26
    5. v0.7.31
    6. v0.7.32
    7. v0.7.33
2. Feature worked with v0.7.26
3. Does not work anymore with v0.7.33, device does **not** reboot at midnight

In json settings export the field "rstMidNight" is always false, no matter how the checkbox in the webpage is set.

[image]
[image]

knickohr, 2023-08-16:
Try exporting the settings, edit manually, and import again. Maybe a workaround, but not a fix yet 😉

LeSpocky, 2023-08-16:
> In json settings export the field "rstMidNight" is always false, no matter how the checkbox in the webpage is set.

That's a different field, I suppose belonging to "Reset values and YieldDay at midnight".  

The json member which is actually changed is "wifi.reb".  According to source it is related to `mCfg.sys.schedReboot`.  That setting seems to be set, json export differs in that field, and that field only.  I did not realize that before the first coffee.  ☕

LeSpocky, 2023-08-16:
If I am not mistaken, the code necessary for the scheduled reboot was added with commit d7eb4a0b982f7ca8a10ecf5db51394229f4ccc2b but remove with merge commit e4e7e64861a798c9f90dc4a403fc2b04b7959a94 again.  Maybe by accident?

lumapu, 2023-08-16:
oh thanks for the insight, I will check that. That makes it easy to fix 👍

closed 2023-08-17 by lumapu