lumapu/ahoy #1099: [Bug] Alarm times one hour off

LeSpocky, 2023-08-17:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.34

### Github Hash

ac53925

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Two inverters, both start to slightly different times depending on sunlight conditions.  Time set through NTP.  Longitude/latitude set to Magdeburg/Germany.

### Error description

Alarm overlay entries show the wrong time.  Attached are three screenshots which show it.  I had a look at the webpage closely today between 6:00 and 7:00 local time, which is CEST currently.  The inverter 0 started today at around 6:15 local time, the inverter 1 around 6:37 local time.  The webpage however shows the related alarm events an hour later.  Maybe there's something wrong in UTC to localtime conversion?

[image]
[image]
[image]

knickohr, 2023-08-17:
Die Frage ist, wann und wo wird der Zeitstempel gesetzt ?

Kommt der Stempel aus dem Inverter, dann ist es wohl eine Stunde zu spät. Kommt er aus der DTU (nach einem Reboot), dann ist es eine Stunde zu spät ! 🤔

LeSpocky, 2023-08-17:
Note: the difference between UTC and CEST is +2 hours.  It is _not_ CET with +1 hour here, we currently have DST until end of October.

btw: I was wondering how the DTU determines the localtime anyway?  From the longitude/latitude?  As far as I can see there's no timezone setting on the AhoyDTU webpage, right?  Or does the browser do the correction based on browser settings?

lumapu, 2023-08-17:
yes the correction is done by the browser. There is no timezone setting until now.

The inverter transmits a timestamp starting at 0 but I don't know which timezone is the base. You can simply check `[IP]/api/inverter/alarm/0`. For me it shows `14545`. If you now paste this time to https://www.unixtimestamp.com/ then you will see that it will be `GMT | Thu Jan 01 1970 04:02:25 GMT+0000`. For me adding 2 hours looks correct. (the inverter only receives UTC time)

knickohr, 2023-08-18:
Aus der heutigen Release 0.7.36 :

[image]

[image]

[image]

Das letzte Foto zeigt den Status der einzelnen Inverter. Da sieht man sehr schön, das irgendetwas noch nicht stimmt. Um 7:31 wahren sie alle schon wach und haben fleißig produziert. 6:31 stimmt wohl 😉

lumapu, 2023-08-18:
hmmm, mal schauen wie das sein kann 😇

lumapu, 2023-08-18:
Problem gefunden 😀
Der Inveter liefert seinen Timestamp ja für den 01.01.1970. Wenn man für dieses Datum das Zeitzonenoffset berechnet, verwendet man automatisch Winterzeit, also +1 Stunde. Da wir aber aktuell Sommerzeit haben sollten es +2 Stunden sein. Leider wurden bisher beide Zeiten addiert sodass wir +3 Stunden hatten.
Wird gefixt in der nächsten dev `0.7.37`

knickohr, 2023-08-18:
Purrrfekt ! 👍

lumapu, 2023-08-18:
das gibt's bei OpenDTU schon eine Diskussion dazu:

https://github.com/tbnobody/OpenDTU/discussions/590

knickohr, 2023-08-18:
Ja, die kenne ich schon 👍

LeSpocky, 2023-08-19:
Mit 4d19e2d sehen die Zeiten heute früh erstmal korrekt aus.  Für beide "Inverter start" Ereignisse wird jetzt eine plausible Zeit angezeigt.

closed 2023-12-29 by lumapu