lumapu/ahoy #1100: [Bug]

mellow5, 2023-08-17:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.26

### Github Hash

cbdb150

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

used the stable build via webinstaller no changes

### Debug Serial Log output

```shell
I: (#0) resetPayload
I: (#0) Requesting Inv SN 
I: (#0) enqueCommand: 0x0B
I: (#0) prepareDevInformCmd 0x0B
15 pid: 80
I: TX 27B Ch61 | 15 90 75 85 20 82 88 47 96 80 0B 00 64 DE A3 B9 00 00 00 0A 00 00 00 00 B5 7D 67 
I: RX 27B Ch40 | 95 90 75 85 20 90 75 85 20 01 00 01 00 E8 00 03 00 01 00 07 00 02 00 00 00 5A 20 
I: RX 27B Ch40 | 95 90 75 85 20 90 75 85 20 02 00 00 00 00 00 00 00 00 00 0B 00 01 00 00 00 00 9D 
I: RX 27B Ch40 | 95 90 75 85 20 90 75 85 20 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 92 
I: RX 27B Ch40 | 95 90 75 85 20 90 75 85 20 84 00 00 00 00 00 00 00 00 00 00 01 16 00 0A 80 30 BC 
I: procPyld: cmd:  0x0B
I: procPyld: txid: 0x95
I: Payload (62): 00 01 00 E8 00 03 00 01 00 07 00 02 00 00 00 5A 00 00 00 00 00 00 00 00 00 0B 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 00 00 00 00 00 00 00 00 00 00 01 16 00 0A 
I: Iv: 0
I: INV/ch1/U_DC: 23.200 V
I: INV/ch1/I_DC: 0.030 A
I: INV/ch1/P_DC: 0.700 W
I: INV/ch1/YieldTotal: 0.090 kWh
I: INV/ch1/Irradiation: 0.184 %
I: INV/ch2/U_DC: 23.200 V
I: INV/ch2/I_DC: 0.010 A
I: INV/ch2/P_DC: 0.200 W
I: INV/ch2/Irradiation: 0.053 %
I: INV/ch3/U_DC: 1.100 V
I: INV/ch3/I_DC: 0.010 A
I: INV/ch4/U_DC: 1.100 V
I: INV/ch0/U_AC: 0.400 V
I: INV/ch0/Temp: 27.800 °C
I: INV/ch0/ALARM_MES_ID: 10.000 
I: INV/ch0/YieldTotal: 0.090 kWh
I: INV/ch0/P_DC: 0.900 W
I: 
I: (#0) resetPayload
I: (#0) Requesting Inv SN 
I: (#0) enqueCommand: 0x0B
I: (#0) prepareDevInformCmd 0x0B
15 pid: 80
I: TX 27B Ch75 | 15 90 75 85 20 82 88 47 96 80 0B 00 64 DE A3 BA 00 00 00 0A 00 00 00 00 45 69 80 
I: RX 27B Ch40 | 95 90 75 85 20 90 75 85 20 01 00 01 00 E8 00 03 00 01 00 07 00 02 00 00 00 5A 20 
I: RX 27B Ch40 | 95 90 75 85 20 90 75 85 20 02 00 00 00 00 00 00 00 00 00 0B 00 01 00 00 00 00 9D 
I: RX 27B Ch40 | 95 90 75 85 20 90 75 85 20 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 92 
I: RX 27B Ch40 | 95 90 75 85 20 90 75 85 20 84 00 00 00 00 00 00 00 00 00 00 01 17 00 0A 40 61 2C 
I: procPyld: cmd:  0x0B
I: procPyld: txid: 0x95
I: Payload (62): 00 01 00 E8 00 03 00 01 00 07 00 02 00 00 00 5A 00 00 00 00 00 00 00 00 00 0B 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 00 00 00 00 00 00 00 00 00 00 01 17 00 0A 
I: Iv: 0
I: INV/ch1/U_DC: 23.200 V
I: INV/ch1/I_DC: 0.030 A
I: INV/ch1/P_DC: 0.700 W
I: INV/ch1/YieldTotal: 0.090 kWh
I: INV/ch1/Irradiation: 0.184 %
I: INV/ch2/U_DC: 23.200 V
I: INV/ch2/I_DC: 0.010 A
I: INV/ch2/P_DC: 0.200 W
I: INV/ch2/Irradiation: 0.053 %
I: INV/ch3/U_DC: 1.100 V
I: INV/ch3/I_DC: 0.010 A
I: INV/ch4/U_DC: 1.100 V
I: INV/ch0/U_AC: 0.400 V
I: INV/ch0/Temp: 27.900 °C
I: INV/ch0/ALARM_MES_ID: 10.000 
I: INV/ch0/YieldTotal: 0.090 kWh
I: INV/ch0/P_DC: 0.900 W
I: 
I: (#0) resetPayload
I: (#0) Requesting Inv SN 
I: (#0) enqueCommand: 0x0B
I: (#0) prepareDevInformCmd 0x0B
15 pid: 80
I: TX 27B Ch3 | 15 90 75 85 20 82 88 47 96 80 0B 00 64 DE A3 BB 00 00 00 0A 00 00 00 00 D5 64 1C 
I: RX 27B Ch40 | 95 90 75 85 20 90 75 85 20 01 00 01 00 E8 00 03 00 01 00 07 00 02 00 00 00 5A 20 
I: RX 27B Ch40 | 95 90 75 85 20 90 75 85 20 02 00 00 00 00 00 00 00 00 00 0B 00 01 00 00 00 00 9D 
I: RX 27B Ch40 | 95 90 75 85 20 90 75 85 20 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 92 
I: RX 27B Ch40 | 95 90 75 85 20 90 75 85 20 84 00 00 00 00 00 00 00 00 00 00 01 17 00 0A 40 61 2C 
I: procPyld: cmd:  0x0B
I: procPyld: txid: 0x95
I: Payload (62): 00 01 00 E8 00 03 00 01 00 07 00 02 00 00 00 5A 00 00 00 00 00 00 00 00 00 0B 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 00 00 00 00 00 00 00 00 00 00 01 17 00 0A 
I: Iv: 0
I: INV/ch1/U_DC: 23.200 V
I: INV/ch1/I_DC: 0.030 A
I: INV/ch1/P_DC: 0.700 W
I: INV/ch1/YieldTotal: 0.090 kWh
I: INV/ch1/Irradiation: 0.184 %
I: INV/ch2/U_DC: 23.200 V
I: INV/ch2/I_DC: 0.010 A
I: INV/ch2/P_DC: 0.200 W
I: INV/ch2/Irradiation: 0.053 %
I: INV/ch3/U_DC: 1.100 V
I: INV/ch3/I_DC: 0.010 A
I: INV/ch4/U_DC: 1.100 V
I: INV/ch0/U_AC: 0.400 V
I: INV/ch0/Temp: 27.900 °C
I: INV/ch0/ALARM_MES_ID: 10.000 
I: INV/ch0/YieldTotal: 0.090 kWh
I: INV/ch0/P_DC: 0.900 W
```

### Error description

I think, the maximum nonpersistent power transmitted via MQTT/serial  is stored persistent. After a reboot of the inverter the non persistent value is still there.

I intenionally changed the persistent value e.g. to 80 %. It gets updated in the live view immediatly. Than I change the non persistent value to 100 %. It changes in the live view to 100 %. Than I reboot the inverter. As soon as the live view starts updating it is at 100 % not 80 % as expected.

I'm using the HM 1500.

I deleted my serial number from the logs.

Ollipop030, 2023-08-18:
A simple reboot will not reset the nonpersistent limit. To reset the nonpersistent power limit you have to cut power from the inverter, AC and DC for about min. 10 seconds.

mellow5, 2023-08-18:
> A simple reboot will not reset the nonpersistent limit. To reset the nonpersistent power limit you have to cut power from the inverter, AC and DC for about min. 10 seconds.

Thats what I die to reboot the Inverter.

I realized it took way more than 10 seconds. Switching it off for 15 min worked.

closed 2023-08-18 by mellow5

mellow5, 2023-08-18:
It is just the displayed value of the power limit in LiveView that does not update after a restart of the inverter.

lumapu, 2023-08-18:
it's only read when Ahoy was not able to communicate for 15 minutes with the inverter