lumapu/ahoy #1101: [Bug]

Loetnase, 2023-08-19:
### Platform

ESP32

### Assembly

the DTU was already assembled

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.37

### Github Hash

ba218ed

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Guten Morgen.
[image]
Ich habe die latest 0.7.37 gerade auf ein neues System ESP 32 mit Erease installiert.
Das meldet sich als 7.36 und die Vorbelegung vom ePaper fehlt wieder.
Das wurde doch schon im Issue 1075 behandelt bzw. gefixt

### Error description

Ich habe die latest 0.7.37 gerade auf ein neues System ESP 32 mit Erease installiert.
Das meldet sich als 7.36 und die Vorbelegung vom ePaper fehlt wieder.
Das wurde doch schon im Issue 1075 behandelt bzw. gefixt

knickohr, 2023-08-19:
Die Meldung von 0.7.36 kommt wohl noch aus Deinem Browsercache.

Du hast die richtige Belegung vom ePaper am nativen ESP32 (nicht S3 !) ?

Ansonsten :

[image]

Loetnase, 2023-08-19:
Ja habe ich gesehen. Die 7.36 habe ich versehentlich mit dem Webinstaller installiert und dann wurde die 7.37 nicht übernommen. 
Das mit der Display Config ePaper Vorbelegung ist aber so wie beschrieben, die fehlt leider wieder.

lumapu, 2023-08-19:
das habe ich auch noch nicht implementiert

Loetnase, 2023-08-19:
Ich dachte du hättest das schon reingenommen.
Dann mach ich das Issue gleich wieder zu.
Danke dir für deine tolle Arbeit

knickohr, 2023-08-19:
Ich glaube wir sollten hier noch nicht zu machen. Den Issue könnte man prima für weitere default Display-Pinout Settings verwenden 😉

Hier das für das Nokia am (nativen) ESP32 :

[image]

Loetnase, 2023-08-19:
da machen wir hier die Sammlung auf 😁

knickohr, 2023-08-19:
Ich suche dann noch die I2C Displays raus

knickohr, 2023-08-20:
Soo, hier das Default-Setting für **alle** OLEDs, sind für alle gleich, egal wie groß und welcher Controller. Den Controller sollte man natürlich korrekt auswählen 😉

[image]

Man beachte das Pixel-Shifting und die nicht zu hoch eingestellte Luminance. Das verhindert ein Einbrennen des OLEDs.

Loetnase, 2023-08-24:
Hier ist das Default Setting für ESP8266 mit Nokia 5110 bis V 0.7.33 und altem Schaltplan.
[image]

lumapu, 2023-08-27:
opendtu-fusion v2.x ePaper Pinout:

[image]

knickohr, 2023-08-28:
Ja ?  Aber funktioniert doch noch nicht mit  Ahoy 0.7.x & nRF & ePaper. Oder jetzt doch mit der .42 ? @lumapu

Loetnase, 2023-08-28:
@lumapu, @knickohr 
macht Luminance 60 beim ePaper Sinn?
Oder wäre 100 besser 🤔
Was ist als Max überhaupt einstellbar, auch 255 wie beim Nokia.

lumapu, 2023-08-28:
Luminance wird doch beim ePaper garnicht verwendet?! Ist glaube ich nur fehlerhaft hier angezeigt, man braucht es nur für alle anderen Displays

Loetnase, 2023-08-28:
@lumapu 
du hast recht,
0 bis 255 ist eingebbar, verändern tut sich aber nichts. 
Macht es Sinn das beim ePaper auszublenden?

knickohr, 2023-08-28:
Luminance ist für OLED, Contrast für das Nokia, dem ePaper ist das Setting Wurst.

Genauso wie das Pixelshifting (Screensaver) nur für das OLED relevant ist, für alle anderen Displays belanglos.

Ich glaube ich muß mal ein Issue aufmachen, damit man unsinnige Settings bei diversen Displays ausblendet.

stefan123t, 2024-10-29:
@knickohr @Loetnase prima Sammlung, ist die in der Zwischenzeit auch schon im Code als Defaults für die jeweiligen Displays hinterlegt ?
Müssen wir da noch verschiedene Varianten berücksichtigen, z.B. Display XY am ESP32-S3 oder am OpenDTU-Fusion Board ?
Das wäre dann ja eventuell auch für die Dokumentation der Pin Mappings wichtig und richtig ?