lumapu/ahoy #1102: Max Power?

Karlo49, 2023-08-19:
ESP8266  Software 0.7.36  Firefox  Linux Mint 20.3
Kann es sein, daß die Max Power Werte vom Vortag übernommen werden und erst überschrieben werden wenn sie höher sind?
Ist das gewollt? Bei mir ist das so.
Kann man die Max Power in in den Settings abschaltbar machen?
Ich brauche das nicht, weil ich mir das sowieso in Grafana anschaue.

lumapu, 2023-08-19:
da brauchen wir noch settings dafür, bisher wird ab Ahoy start gezählt.
Wenn du es nicht brauchst, dann übersehe es doch einfach, ich würde nur eine Einstellung "reset max values on midnight" vorsehen

knickohr, 2023-08-20:
Ja, beides hat seinen Charm. Sowohl die max. Leistung am Tage, wie auch über die DTU Laufzeit.

Gute Idee mit dem Häkchen 😉

DanielR92, 2023-08-25:
wie ich sehe wurde es von Lumapu eingebaut. 
Darf man es closen? :)

closed 2023-08-25 by DanielR92

lumapu, 2023-08-26:
das finde ich verfrüht, ich schließe die `fixed dev` erst beim nächsten release. Es könnte sonst sein, dass wir wieder issues mit dem gleichen Thema reinbekommen