lumapu/ahoy #1103: [Bug]

SeibeGitHub, 2023-08-19:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.36

### Github Hash

ba218ed

### Build & Flash Method

ESP Tools (flash)

### Setup

Only:
Network: Setting (DHCP)
Inverter: one inverter, Polling 15s, Reset values when inverter polling pauses at sunset = yes
Sunrise&Sunset: Geodaata & pause polling during night=yes

### Error description

Hallo, ich habe eben von ESP8266 auf ESP32 umgestellt, dabei habe ich versehentlich als Dezimaltrennzeichen den . (Punkt) anstatt das , (Komma) bei den Geodaten in Sunrise & Sunset angegeben, Danach ist der ESP nicht mehr zu erreichen als Version habe ich V0.7.36 wobei dies glaub unerheblich ist. Das verhalten ist reproduzierbar.

lumapu, 2023-08-20:
Habe es gerade getestet - mit einem ESP8266. Ich kann es nicht reproduzieren. Nach dem Speichern stehen für beide Kooridnaten wieder die `.` statt den `,` drin.

SeibeGitHub, 2023-08-21:
Ich habe es gerade nochmals mit einem ESP32 getestet, 230817_ahoy_0.7.36_ba218ed_esp32.bin neu neu geflasht zugriff via AP Mode problemlosmöglich, dann bei Netzwerk mein Lokales gescannt und PW eigetragen mehr nicht, Zugriff mit vom DHCP vergebener IP problemlos möglich nun nur bei SUNSET folgenden Geodaten eingetragen 48.80865342748568     9.278663963477005 und den haken bei pause polling gesetzt mehr nicht. Danach save und ab dann ist der Zugriff bzw. das pingen nicht mehr möglich.   
[image]
[image]

lumapu, 2023-08-21:
könntest du am ESP bitte das Serielle Log mitlaufen lassen? Ich habe es gerade eben nochmal mit der `0.7.38` getestet (auf einem ESP8266) mit deinen Daten.

SeibeGitHub, 2023-08-21:
Ich habe mal über HTerm ausgelesen, hoffe das ist was du sehen wolltest. Der log wurde nach dem Eintrag in SUNSET gemacht, da sieht man dass der ESP sich verbindet kurz da ist und dann wieder neu verbindet aber eigentlich nicht abschmiert sprich kein WD oder ähnliches auslöst. Wenn du ein anderes log meinst müsstest du mir sagen wie ich das erstellen soll.

[output_2023-08-21_13-37-05.log](https://github.com/lumapu/ahoy/files/12395530/output_2023-08-21_13-37-05.log)

SeibeGitHub, 2023-08-21:
Es ist gerade ein nackter ESP32 kein Port ist aktuell angeschlossen

lumapu, 2023-08-21:
danke für das Log, das hat aus meiner Sicht nichts mit den Sunrise-Settings zu tun. Es sieht mir nach reinen WLAN-Problemen aus, woran auch immer das liegt. 

hier noch die Übersetzungen der Event IDs:
WIFI event 1 = ARDUINO_EVENT_WIFI_SCAN_DONE
WIFI event 4 = `ARDUINO_EVENT_WIFI_STA_CONNECTED`
WIFI event 5 = `ARDUINO_EVENT_WIFI_STA_DISCONNECTED`
WIFI event 7 = `ARDUINO_EVENT_WIFI_STA_GOT_IP`
WIFI event 10 = ARDUINO_EVENT_WIFI_AP_START
WIFI event 11 = ARDUINO_EVENT_WIFI_AP_STOP

SeibeGitHub, 2023-08-21:
Ok, wie auch immer mit den anderen Settings läuft es ja Problemlos. Der RSSI an der Fritze ist im übrigen vom ESP -44 was ja völlig okay ist. Evtl. hat der ESP ja einen schuss in welcher form auch immer. Habe eh für die Bastelkiste neue 32er bestellt und werde es dann nochmal nachstellen in der HF spukt es eh ab und an :-). Danke für deine Mühe. Noch ne kurze Frage bezüglich dem Intervall zum Abruf der WR , ich nehme ja an, dass diese versetzt abgerufen werden so dass sie sich beim senden nicht gegenseitig stören aktuell habe ich 6 Inverter und einen  Intervall von 5sec eingestellt und eine Wiederholung von 5. Was würdest du empfehlen?

rmayergfx, 2023-08-21:
Kleiner Tipp von mir, gehe auf die Seite https://esphome.github.io/esp-web-tools/ und flashe den ESP mal dort und verbinde dich mit deinem gewählten Netzwerk, da siehst du ganz schnell ob der ESP einen Hau weg hat. Dort sind die WLAN Settings wesentlich einfacher zu setzen und zu testen. Nicht vergessen, Chrome benutzen. Wenn ESPHome stabil läuft kannst du einfach die passende bin von AhoyDTU über OTA Update einspielen.

SeibeGitHub, 2023-08-21:
Ich habe schon eigene Projekte wo ich einst ne Nulleispeisung realisiert habe auf den selben ESP aufgespielt, da sieht man dieses merkwürdige Verhalten nicht. Auch das bekannte WLED hab ich testweise auf diesem ESP gehabt und es läuft Stundenlang. Wie gesagt ich belass es vorerst und teste es die Tage wenn die neuen ESP eintreffen nochmal scheint ja auch ein einmaliges Verhalten zu sein.

lumapu, 2023-08-21:
> Noch ne kurze Frage bezüglich dem Intervall zum Abruf der WR , ich nehme ja an, dass diese versetzt abgerufen werden so dass sie sich beim senden nicht gegenseitig stören aktuell habe ich 6 Inverter und einen Intervall von 5sec eingestellt und eine Wiederholung von 5. Was würdest du empfehlen?

Das ist schwierig zu sagen, aber allgemein ist alles im Bereich von 5s auf einem ESP32 sicher. Wir du richtig angemerkt hast, stören sich die Inverter nicht gegenseitig, da sie nacheinander abgefragt werden. Jede Abfrage dauert ca 300ms, danach kommen noch die Berechnungen und die MqTT-Übertragung.
5 Retries ist Standard, daher würde ich es erst mal so lassen, nur wenn es Probleme gibt hier schrauben.

closed 2024-01-02 by lumapu