lumapu/ahoy #1104: [Bug] HA MQTT entity name starts with the device name in your config

3gfisch, 2023-08-19:
### Platform

ESP8266

### Assembly

the DTU was already assembled

### Antenna

circuit board

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.9

### Github Hash

15ec6a0

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

MQTT for HA added

### Error description

Home Assistant with the newer update complains about the MQTT names that it prefixes the inverter name:

Log:
Logger: homeassistant.components.mqtt.mixins
Source: components/mqtt/mixins.py:294 
Integration: MQTT (documentation, issues) 
First occurred: 15:10:19 (36 occurrences) 
Last logged: 15:10:19

MQTT entity name starts with the device name in your config {'name': 'Haas ALARM_MES_ID', 'state_topic': 'inverter/Haas/ch0/ALARM_MES_ID', 'unit_of_measurement': '', 'unique_id': '116181603236_ch0_ALARM_MES_ID', 'device': {'name': 'Haas', 'identifiers': ['116181603236'], 'model': 'Haas', 'configuration_url': 'http://192.168.1.103', 'manufacturer': 'Hoymiles', 'connections': []}, 'expire_after': 95, 'qos': 0, 'payload_not_available': 'offline', 'enabled_by_default': True, 'payload_available': 'online', 'availability_mode': 'latest', 'encoding': 'utf-8', 'force_update': False}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'ALARM_MES_ID'
MQTT entity name starts with the device name in your config {'name': 'Haas YieldDay', 'state_topic': 'inverter/Haas/ch0/YieldDay', 'unit_of_measurement': 'Wh', 'unique_id': '116181603236_ch0_YieldDay', 'device': {'name': 'Haas', 'identifiers': ['116181603236'], 'model': 'Haas', 'configuration_url': 'http://192.168.1.103', 'manufacturer': 'Hoymiles', 'connections': []}, 'device_class': <SensorDeviceClass.ENERGY: 'energy'>, 'state_class': <SensorStateClass.TOTAL_INCREASING: 'total_increasing'>, 'qos': 0, 'payload_not_available': 'offline', 'enabled_by_default': True, 'payload_available': 'online', 'availability_mode': 'latest', 'encoding': 'utf-8', 'force_update': False}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'YieldDay'
MQTT entity name starts with the device name in your config {'name': 'Haas YieldTotal', 'state_topic': 'inverter/Haas/ch0/YieldTotal', 'unit_of_measurement': 'kWh', 'unique_id': '116181603236_ch0_YieldTotal', 'device': {'name': 'Haas', 'identifiers': ['116181603236'], 'model': 'Haas', 'configuration_url': 'http://192.168.1.103', 'manufacturer': 'Hoymiles', 'connections': []}, 'device_class': <SensorDeviceClass.ENERGY: 'energy'>, 'state_class': <SensorStateClass.TOTAL_INCREASING: 'total_increasing'>, 'qos': 0, 'payload_not_available': 'offline', 'enabled_by_default': True, 'payload_available': 'online', 'availability_mode': 'latest', 'encoding': 'utf-8', 'force_update': False}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'YieldTotal'
MQTT entity name starts with the device name in your config {'name': 'Haas P_DC', 'state_topic': 'inverter/Haas/ch0/P_DC', 'unit_of_measurement': 'W', 'unique_id': '116181603236_ch0_P_DC', 'device': {'name': 'Haas', 'identifiers': ['116181603236'], 'model': 'Haas', 'configuration_url': 'http://192.168.1.103', 'manufacturer': 'Hoymiles', 'connections': []}, 'expire_after': 95, 'device_class': <SensorDeviceClass.POWER: 'power'>, 'state_class': <SensorStateClass.MEASUREMENT: 'measurement'>, 'qos': 0, 'payload_not_available': 'offline', 'enabled_by_default': True, 'payload_available': 'online', 'availability_mode': 'latest', 'encoding': 'utf-8', 'force_update': False}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'P_DC'
MQTT entity name starts with the device name in your config {'name': 'Haas Efficiency', 'state_topic': 'inverter/Haas/ch0/Efficiency', 'unit_of_measurement': '%', 'unique_id': '116181603236_ch0_Efficiency', 'device': {'name': 'Haas', 'identifiers': ['116181603236'], 'model': 'Haas', 'configuration_url': 'http://192.168.1.103', 'manufacturer': 'Hoymiles', 'connections': []}, 'expire_after': 95, 'qos': 0, 'payload_not_available': 'offline', 'enabled_by_default': True, 'payload_available': 'online', 'availability_mode': 'latest', 'encoding': 'utf-8', 'force_update': False}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'Efficiency'

lumapu, 2023-08-19:
I think you should test the latest release before reporting about it. There is already a closed issue with the same topic #1066

3gfisch, 2023-08-20:
@lumapu thanks i did as mentioned there, delete the MQTT device form HA integration, updated and clicked send on the DTU web interface to discover the new device in HA looks good so far. Thanks!

closed 2023-08-20 by 3gfisch