lumapu/ahoy #1105: [Bug]

DejanBukovec, 2023-08-19:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.36

### Github Hash

ba218ed

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

-

### Error description

Hi,
Is possible that there is bug in new "Alarms" function?
I have 3 inverters and for example today first two of them get information of "Alarm" and count is 1 and last one do not get information and it stay at 0...

[image]
[image]

[image]
[image]

Problematic one:
[image]
[image]

Ollipop030, 2023-08-20:
Ist mir gar nicht aufgefallen, aber bei mir hat der letzte Inverter auch keinen Alarm. Sollte ja zumindest der Start heute morgen drin sein.

knickohr, 2023-08-20:
Nein, das ist nur blöder Zufall. 

Könnte auch behaupten, der erste Inverter ist der der am letzten online kommt. Das beobachte ich bei meinen 3vDTUs 😉

DejanBukovec, 2023-08-20:
I can confirm that today again Inverter #2 didn't get Alarm data and stay at 0. Inverter #0 and #1 get one alarm(Inverter start) and this is ok. Also it looks like it is old data because time is exactly same as yesterday which is not possible... By experience with OpenDTU every day Alarms reset and I receive new Inverter Start and other Alarms...
Question now is also why there is no new data(Inverter Start)? Is possible that Inverter start producing power before DTU start pooling and DTU do not accept data? Does DTU periodically pool Alarm data(Read also old data from inverter)?

@knickohr Using multiple AhoyDTU's at same time for same inverters as I know is not ok and produce communication problems...

Peertje62, 2023-08-20:
Good morning,
I also have two inverters, HM-300 & HM-600, ESP 3266, works as expected, both inverters show their start alarms.
( except #1096) 
regards Herman

DejanBukovec, 2023-08-20:
Im remove AC power and now I have for Inverter 0 and 1 I have 5 Alarms and for inverter 2 I have 4 Alarms...

[image]
[image]
[image]

I think that Sunrise/Sunset function pooling do not work ok and my inverters start before location sunrise and work longer than location sunset... For Sunset Im 100% sure because I see that inverters has produce small power when DTU already stop pooling(I have enabled "Pause polling inverters during night")... Then Im also set "Offset (pre sunrise, post sunset)" to 10 minutes but I will now try 1 hour to see if tomorrow DTU will catch Alarms...

Peertje62, 2023-08-20:
Hi, correct me if I'am wrong but:
Inverter starts have nothing to do with DTU start or pooling and readout.
as far as you **don't** switch the mains of the inverters separate! (Shelly?)
The fact that you get multiple notifications is allready in the pipeline (#1096 )
regards Herman

DejanBukovec, 2023-08-20:
@Peertje62 I don't have problems like you... I completly understand my events... It looks like AhoyDTU keep in memory old data...
Inverter 0:
Inverter start 7:02:33 is **YESTERDAY** start event
Inverter start 7:06:35 is **TODAY** start event
Other 3 events happen when Im remove AC power.

1 hour ago Im reboot DTU an now Im lost YESTERDAY events and see 4 Alarms for ALL 3 Inverters:
[image]
[image]
[image]

Now I have 60 minute offset and will see if geting alarms(Problematic Inverter Start) from Inverter have some connection with pooling... I think that it have so if pooling is disabled for inverter you will not receive Alarms until some other alarm happen(Which normaly do not happen)...

Peertje62, 2023-08-20:
Reboot Ahoy at midnight

knickohr, 2023-08-20:
Yes, Sunrise, Sunset can bei an issue if it is too Late/early.

No, I‘m using different Inverters on the different DTUs 😉

DejanBukovec, 2023-08-20:
@Peertje62 Can you please tell me how reset at midnight will solve that DTU do not pool alarm data in the morning? Only thing which will help will be reset after all inverters are online…

@knickohr ok understand you use something like one DTU for one inverter… Does this help you with “power change”(For example zero feed option) response times?

@lumapu can you please check code if is possible to add read alarms when inverters became online? Thanks.

Peertje62, 2023-08-20:
Hi,
the assumption was that the DTU was showing the old alarm messages again as they were apparently stored in the DTU, a reboot would fix that

DejanBukovec, 2023-08-20:
@Peertje62 Im open issue about missing alarms and not duplicated or old data... For old data is already opened issue #1096 which is different as my problem...

lumapu, 2023-08-21:
do you use power limit control? Checking the code I saw, that once you apply a new power limit control command the entiere queue will be cleared. If this happens before the alarms were read the read will not be triggered again.
Sure it's a problem of desing - I will work on it.

DejanBukovec, 2023-08-21:
Today with 60 minute offset it didn't receive alarm data. 
After reboot Alarms stay at 0.
I will try later connect PC to DTU and get logs after reboot... Because there will be a lot of data is there some specific string/hex which I need to wait?

I use power limit because have "zero export" solar system So I send probably imediately when inverter became online power limit command which reset alarm receive command... Can you somehow re-add alarm command together with statistics command into queue if it is deleted from queue?

knickohr, 2023-08-21:
With the v0.7.39 all of my Inverters now have a Start Alarm.

DanielR92, 2023-08-25:
Is that fixed now?

DejanBukovec, 2023-08-25:
No it is not. I m report this issue with latest release 0.7.36 and there is no new release yet...

DanielR92, 2023-08-25:
Thanks, then I try to solve it. :)

DejanBukovec, 2023-08-26:
Im yesterday install prebuilded dev version 0.7.40 and problem is not fixed in it. Alarm for Inverter Start is still missing(Cleared from queue because of limit commands)...

DejanBukovec, 2023-08-26:
Today in the morning when Inverters already produce some small power I'm install prebulded dev version 0.7.41 and issue is not fixed. Only first inverter has get Alarm data after DTU reboot...

stefan123t, 2023-10-08:
@DejanBukovec Danke für die Meldung, ja es gibt noch ein paar Unbekannte / unknown Fehlermeldungen.
U.a. 
* 12 (https://github.com/tbnobody/OpenDTU/issues/1068),
* 14 (https://github.com/tbnobody/OpenDTU/issues/131),
* 12, 13, 14, 36,  (https://github.com/tbnobody/OpenDTU/issues/121),
* 12, 14, 36, 46, 47, 73 https://github.com/tbnobody/OpenDTU/discussions/590

Ich bin in den `assets/warn_code.json` der Hoymiles S-Miles Installer App fündig geworden. 🥳 

Hier mal die bisher aufgetretenen Fehler oder Warnings aus dem in [#121](https://github.com/tbnobody/OpenDTU/issues/121) angehängten warn_code.json (GitHub versteht nur ZIP ?).

11: Grid voltage surge
12: Grid voltage sharp drop
13: Grid frequency mutation
14: Grid phase mutation
36: INV overvoltage or overcurrent
46: FB overvoltage
47: FB overcurrent
71: VW function enable
72: FW function enable
73: TW function enable

Es wird von @lumapu im Rahmen von #1105, #177 oder #608 in Ahoy implementiert und daher kann dieses Issue evtl. auch bald geschlossen werden ?

MrFr33man123, 2023-10-13:
Ich versuche mich mal an das Issue ran zuhängen. In einer älteren AHOY Version habe ich hochzählende Nummern als Alarm bekommen. Jeden Inverter Start von der Nummer des letzten wieder hoch.
20 => 21 => 24 => 29 => 30 => 31 => ...
Aus den Daten hatte ich abgeleitet das die Auswertung den Unterschied zwischen der Vorherigen Nummer und der Aktuellen die Eigentliche Nachricht ist. Für meinen HM-600 war das:
0 | Inverter Start
1 | Grid OK Starting Production
3 | Producing Power
5 | Grid Connected

Mit der aktuellen Version bekomme ich immer nur die Erhöhung um 1.

Was wird denn aktuell genutzt als Ausgabe?

lumapu, 2023-10-15:
die ID ansich sagt nichts zum Alarm, sie gibt nur an wie viele Alarme getrackt wurden. Anhand dieser ID liest Ahoy dann die Alarme der Reihe nach aus.

closed 2024-05-30 by lumapu