lumapu/ahoy #1106: [Bug] Luminance of SSD1306 cannot be changed

technics42, 2023-08-20:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.26

### Github Hash

cbdb150

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Nothing special

### Error description

Changing luminance of display in settings actually does not change luminance/contrast of display.
Regardless of value, the brightness of display is always the same.

knickohr, 2023-08-20:
Is it really a SSD1306 Controller on the display, or you using a display with another Controller, maybe the 2.42“ OLED with SSD1309 Controller ?

technics42, 2023-08-20:
Yes, it's really a SSD1306 with 0.96"

DanielR92, 2023-08-26:
I have a SSD1306 too. But the Luminance makes here no effect... I can reproduce it. 
I use the branch in dev03.

kiu77, 2023-08-28:
Had this even with an 1,3" display and SH1106. Thought this parameter just nor works. But will double check that now since I know that others have a similar problem.

lumapu, 2023-09-04:
please test with latest dev version `0.7.46` available in ~20min (https;//fw.ahoydtu.de)

lumapu, 2023-09-12:
@technics42 I think that will be a new feature-request, maybe you create another issue and we collect the interest before implementation.

technics42, 2023-09-13:
Ok, let's do it this way.

closed 2023-09-13 by technics42