lumapu/ahoy #1107: [Bug] ePaper FullRefresh Routine vermutlich fehlerhaft

knickohr, 2023-08-21:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.38 und davor

### Github Hash

Blahblah

### Build & Flash Method

ESP Tools (flash)

### Setup

ePaper

### Error description

Wurde an der FullRefresh Routine des ePapers was verändert, ergänzt, optimiert ?

Mir ist aufgefallen das der FullRefresh beim Booten der DTU wesentlich schneller stattfindet. Das wäre eigentlich kein Problem, sogar wünschenswert, aber es entstehen dann häßliche „Ghostings“ auf dem Display. Der Hintergrung ist nicht mehr weiß, sondern grau, einige Pixel bleiben hängen und vor allem entsteht eine Art Wasserzeichen im Hintergrund. Das Ahoy-Logo erscheint eingebrannt 😲

[image]

Das Verhalten ist sein 2 oder 3 DEVs vorhanden. Kann leider den genauen Beginn nicht mehr sagen. Früher gestaltete sich der Refresh vom ePaper so das er im Sekundentakt mal schwarz, dann weiß, immer mit negativen Logo war. Ab diesem Zeitpunkt ist es eher ein wildes Flackern und das Logo bleibt im Hintergrund irgendwie hängen ☹️

lumapu, 2023-08-21:
das muss ab der eingesetzten library liegen, in Ahoy hat sich hier nichts geändert.

HorstyS, 2023-08-21:
also, ich hab das bei mit nicht. sowohl nicht mit 0.7.26 oder .36

knickohr, 2023-08-21:
Wir sind mittlerweile bei 38 !

HorstyS, 2023-08-21:
korrekt, aber es wurde

Version
0.7.38 und davor

angegeben

HorstyS, 2023-08-21:
und ich versuche mal die .39 ;)

knickohr, 2023-08-21:
Ich bin jetzt zurück bis zur 0.6.9 😲 Und immer das gleiche Verhalten. So lange kann das noch nicht geändert worden sein 🤔

Hab die 39 jetzt drauf und soweit OK, muß halt aufpassen beim Booten. Seltsam finde ich das Verhalten schon, zumal es ja kein Einzelfall ist. Alle meine 3 DTUs haben das „Flattern“ am Anfang 😵

https://github.com/lumapu/ahoy/assets/63048210/a4347595-4ee3-4afd-b30a-c3ee4939e3ab

Früher war dieses Flattern eher ein gemütliches Pumpen im Sekundentakt.

Ich schätze mal @lumapu  hat Recht, da wurde wahrscheinlich was an der Lib verändert 😪

lumapu, 2023-08-21:
dann schauen wir doch mal die Änderungen durch, evtl. ist @dAjaY85 auch schon informiert?

dAjaY85, 2023-08-21:
in OpenDTU ist immer noch die GxEPD2 integriert. Es wurde halt für die Displays ein eigener Task erstellt, damit dieser den gesamten SPI Bus nicht blockiert.

Zusätzlich ist in OpenDTU das ePaper Display nun als SW-SPI integriert.

knickohr, 2023-08-21:
Ich dachte Du hast das komplett umgekrempelt als das CMT rein kam und die Bildchen ? 🤔

Hast doch was von der Waveshare-Lib gesagt 🤪

dAjaY85, 2023-08-21:
hatte kein Bock das ganze Layout neu zu machen, war mir nach paar Tagen zu viel Arbeit :-)

dAjaY85, 2023-08-21:
liegt vermutlich an der Bearbeitungszeit, deswegen ist der Prozess in OpenDTU in einem eigenen Task.

HorstyS, 2023-08-22:
also....ich habe jetzt die .39 seit gestern nachmittag. ich habe kein ghosting oder ähnliches auf meinem ePaper display. der refresh ist auch genauso wie mit den älteren versionen. auch mehrere reboots gestern produzieren das nicht.
just to mention...

dAjaY85, 2023-08-22:
Hängt von vielen Faktoren ab, Anzahl WR etc. jede zusätzliche Komponente bremst die DTU ab.
Wie gesagt, wir hatten festgestellt, dass das Display den DTU Task um paar s bremst, weil während der Bearbeitung des Displays die NRF usw. nicht angesprochen werden.
Es wäre schon Sinnvoll die von @LennartF22 vorgeschlagene Implementierung auch in Ahoy umzusetzen.

Anbei der damalige PR: https://github.com/dAjaY85/OpenDTU/pull/1

lumapu, 2023-08-22:
Habe gestern mein epaper auch wieder in Betrieb genommen (an Fusion Board). Ich habe auch kein ghosting gesehen, war allerdings auch die `0.7.40`.

Ich habe bzgl. des Displays nichts geändert. @knickohr evtl. liegt es tatsächlich an der Tageszeit zu der du Ahoy neu startest.

dAjaY85, 2023-08-22:
Naja Knickohr hat auch seine 12 WR dran.
Nicht das es ein Timing Problem ist.

lumapu, 2023-08-22:
es geht ja eher zu schnell, nach deiner Theorie würde aber alles langsamer sein oder?

HorstyS, 2023-08-22:
ja, ich habe eine einfache konfiguration mit nur einem WR an einem ESP32. das könnte natürlich der unterschied sein.

dAjaY85, 2023-08-22:
Die Display Prozedur blockiert der main Task, das hatten wir Anfang dieses Jahres festgestellt, deswegen auch der eigene Task fürs Display.

knickohr, 2023-08-22:
> Habe gestern mein epaper auch wieder in Betrieb genommen (an Fusion Board). Ich habe auch kein ghosting gesehen, war allerdings auch die `0.7.40`.
> 
Moment !

Heißt das das das Fusion mit ePaper, nRF und Ahoy 0.7.40 wieder läuft ?  Ab der 0.7.x hat es das nämlich nicht mehr gemacht 😲

lumapu, 2023-08-22:
nein nicht mit ePaper aber das kommt hoffentlich bald

knickohr, 2023-09-01:
Fixed !  😎

Die Lösung findet ihr hier im Bild :

[image]

knickohr, 2023-10-15:
Ich mache das wieder auf weil es sich immer mehr verhärtet das die Full-Refresh Routine doch fehlerhaft oder unvollständig zu sein scheint.

Wie ich darauf komme ?
Die DTU hat jetzt eine Uptime von mehreren Tagen und der Hintergrund des ePapers wird immer dunkler. Mit jedem Full-Refresh scheint etwas mehr grau im Hintergrund hängen zu bleiben. Kurioserweise ist an den Stellen sn denen ein Partia-Refresh stattfinden der Hintergrund schön weiß.

Meiner Meinung ist da was im argen 😕

knickohr, 2023-10-26:
Was gibt’s Neues ?

Nun ja, nach 2 Wochen Testbetrieb mit einer Diode in der Versorgungsspannung muß ich berichten das es jetzt zu funktionieren scheint. Ich habe eine alte Germaniumdiode (AA119) genommen weil die einen sehr geringen Spannungsabfall von nur ca. 0,15V hat. Ja, das Ding ist alt und kann nicht viel ab, aber offenbar reicht es aus.

Das würde auch meine Vermutung bekräftigen das es nicht das Display selbst und vermutlich auch die die Refresh-Routine ist, sondern das hier wohl was anderes im Argen ist. Den Verdacht habe ich schon länger das der Strom nicht vom Display benötigt wird, sondern in der restlichen Schaltung gebraucht wird. Nun stellt sich für mich die Frage was beim Refresh noch alles nebenher läuft, bzw. gerade zu diesem Zeitpunkt aktiv ist, das die Spannung runter zieht ?

Ich werde weiter beobachten 😉

MetaChuh, 2023-10-29:
hab' mir grad das e-paper display zum testen bestellt, und bin jetzt schon dankbar für deine vorab infos 👍
wird etwas dauern, bis ich zum testen komme, weil ich wechsle gerade alle esp8266 gegen die form factor und pin kompatiblen esp32 d1 mini module von az.....y und in den ferien der kids bleibt nur noch die zeit, wenn alle schlafen und ich selbst noch nicht zu ko bin 😉

knickohr, 2023-10-29:
Das Problem vom ePaper ist momentan nur ein Einzelfall. Bis jetzt hat noch niemand auch nur ähnliche Probleme gehabt. Es ist nicht hardwareabhängig da ich mit mehreren unterschiedlichen Konfigurationen getestet habe. Ich schätze mal, es ist Alterungs- oder Laufzeitproblem. Da ich das Display seit den allerersten Tests quasi rund um die Uhr am Laufen habe. Kurioserweise scheint das Verhalten nicht das Display selbst auszulösen, sondern etwas von der anderen Hardware. Vielleicht ist es ein ähnlicher Effekt wie beim NRF-Modul und dem Elko. Oder die Library hat doch einen Schuß ? 🤔

Es ist auch noch nicht sicher ob es nich5 irgendwann wieder auftritt. Das wäre aber dann der Beweis das es mit der Alterung des Displays zu tun hat. Deshalb lasse ich auch vorerst mal hier auf.

knickohr, 2023-11-15:
Nach weiteren 2 Wochen und etlichen Reboots mache ich jetzt zu. Das Problem ist seither nicht mehr aufgetreten.

Die Lösung war anscheinend die Diode. Ob der Kondensator zusätzlich notwendig ist, weiß ich nicht, aber schaden tut er auch nicht.

MetaChuh, 2023-11-15:
@knickohr
danke für die info 👍
ps: bei mir liegt das e-paper display seit 2 wochen auch aus zeitmangel noch originialverpackt da 🙈

plus ich finde meine alte germanium diode nicht, die ich 100pro noch irgend wo aus einem  kosmos elektronikbaukasten der 80er habe.

also nicht wundern, wenn in den nächsten paar jahren in diesem issue aus heiterem himmel steht: ha ich hab die diode endlich gefunden 😂

glg und danke dir für die infos 👍

knickohr, 2023-12-22:
Nachtrag hierzu :

Mir is5 vor ein paar Tagen die Germaniumdiode hops gegangen. Keine Ahnung warum. Vielleicht weil sie schon zu alt war oder aber das Display zieht doch hin und wieder mal mehr als 20mA Strom.

Egal, es ist jetzt eine BAT43 drin und die tut auch wunderbar, verträgt jetzt 200mA.

MetaChuh, 2023-12-23:
mein waveshare rev2.1 hat bis jetzt keine issues, auch ohne diode.

lustiges was mir aufgefallen ist (absolut irrelevant aber irgendwie faszinieeerend):

wenn die rotation 0° ist, und man den strom trennt, bleibt das e-paper bild 1:1 eingefroren wie es ist ... so wie erwartet ...
aaaaber: wenn die rotation auf 90° gestellt ist, und man dann den strom trennt, minimiert sich der kontrast des eingefrorenen bildes auf 50%, also verwaschen grauer.

ist voll wurst, aber faszinierend ...
hirnwichsen: im 90° mode erkennt man durch den hellen grauton statt schwarz, dass die spannung weg ist und alles inaktiv ist.
wenn dies auch bei 0° wäre, hätte ich gedacht es ist per design.
jetzt denke ich, per accidental "design" ist auch gut zum wissen und nutzen 😃

knickohr, 2023-12-23:
Wo nimmst Du den Strom weg ?  Am Display oder am USB. Den Effekt habe ich hin und wieder auch wenn ich das von USB nehme. Habe aber noch nicht darauf aufgepaßt ob es von der Rotation abhängt. Ich drehe 180 Grad.

knickohr, 2024-01-10:
Ich mache hier mal wieder auf 😢

Es wird wieder schlimmer, trotz eingebauter Diode. Das „Eingrauen“ und Ghosting kommt ausschließlich nach einem Softboot. Boote ich kalt und warte vorher ein paar Minuten ist es wieder OK.

Deshalb @lumapu , bitte schnellstmöglich den FullRefresh beim Softboot unterdrücken. iTunes mittlerweile 2 weitere User im Discord aufgetaucht sind, die ebenfalls von Ghosting-Problemen beim ePaper berichten 😱

MetaChuh, 2024-01-10:
> Wo nimmst Du den Strom weg ? Am Display oder am USB. Den Effekt habe ich hin und wieder auch wenn ich das von USB nehme. Habe aber noch nicht darauf aufgepaßt ob es von der Rotation abhängt. Ich drehe 180 Grad.

hi @knickohr 
prosit neujahr und sorry für die späte antwort
(ist im trubel untergegangen bis zum re-opening)

ich hab's e-paper noch immer nur fliegend angestöpselt an einen esp32 d1 mini (die mag ich gerne, weil passt "pin kompatibel" rein, wo eigentlich ein esp 8266 d1 mini rein soll)

strom ist direkt auf der 3.3v leitung vom d1 board, sogar ohne extra cap.
nrf und e-paper werden beide vom onboard 3.3v wandler befeuert.
strom kommt von einem front usb port einer synology, oder einem normalen iphone ladeadapter (je nach dem was grad frei ist)

displaymäßig hab' ich keine troubles bis jetzt, aber das ist auch nur eine test dtu mit 0.6.9, die nicht stakkato über api abgefragt wird.

zum display greyout und stabilität kann ich nichts sagen, das teil rennt mit uralt firmware und dient nur dem zweck, dass ich schnell was am epaper testen konnte, also ahoy dtu damals drauf, und danach nie wirklich verwendet.

[image]

glg,
metachuh

knickohr, 2024-01-10:
Ich glaube Dir das. Wenn Du das Display nur hin und wieder mal verwendest, dann funktioniert es.

Tanakkara, 2024-02-04:
[image]
mit der 0.8.64

knickohr, 2024-02-04:
Nein, das war schon weit vorher da, es tritt irgendwann mal auf, und dann geht es nicht mehr weg. Die Geister die ich rief …

ich008, 2024-05-03:
[image]
ist in den letzten 2 Wochen aufgetreten. Softwareupdate von v0.8.36 zu v0.8.114 über v0.8.83 brachten keine Abhilfe. Versuche mich nächste Woche mal am troubleshooting.

Loetnase, 2024-06-19:
Hallo Freunde der geisternden ePaper,

hier mal wieder ein Update von meinen Geistern. Ich habe einmal 3 ePaper zusammen gekauft, 
eines kam gleich in Einsatz -> Aufkleber A (Auffällig) 79, das mit der längsten Betriebszeit, das steht in der Mitte.
das zweite etwas später in den Einsatz  -> Aufkleber 78, steht rechts.
und das dritte mit blauem Rand erst zur Geistersuche vor kurzem im Einsatz.
Je nach verwendeter Softwareversion Ahoy zeigt das A 79 sehr schnell Geister, das 78 später und das ohne mit blauem Rand erst spät.
Das sagt mir, mit steigender Betriebszeit steigt die Geisteranfälligkeit, neue ePapers werden wohl zuerst keine Geister zeigen. @MetaChuh   
Mit höheren Softwareversionen kommen die Geister auch schneller. Das ist auf den Bildern gut zu sehen, z.B. bei der 0.8.126 ist nach 13,5 Stunden alles begeistert.
[image]
Ich probiere gerade mal diverse Versionen rückwärts durch um vieleicht einen Startzeitpunkt zu finden.
[image]
Die uralte 0.6.0 hat nach über 6 Tagen Uptime zwar leichte aber akzeptable Geister. Ich habe da mal die Gehäuse gewechselt aber A ist immer A
[image]
[image]
Ich denke die ePapers werden da irgendwie langsamer und könnten längere Refreshzeiten gebrauchen.
Es macht auch einen Unterschied beim "cleanen" des ePapers ob man einen System -> Reboot macht oder oder Settings -> Save,
Settings save ist da besser, das ePaper ist da sauberer. 
Es macht auch einen großen Unterschied die Versorgungsspannung für ca. eine halbe Stunde weg zu nehmen und dann zu starten wie @knickohr auch schon schrieb. Das lässt die Geister später kommen. 
Ist da im Code irgendwo etwas einstellbar/veränderbar? Ich habe da leider keine brauchbare Ahnung von Software und kann da wohl nicht viel helfen.
Wenn aber jemand mit Ahnung da etwas machen kann aber ein geisterndes ePaper hat, einfach bitte melden.
Er oder sie kann gerne eine oder zwei komplette DTUs von mir bekommen, ich helfe da gerne.
Ich danke euch für eure Mithilfe und geleistete Arbeit.  
mkG eure Lötnase

semicuda, 2024-06-22:
Hi zusammen,

ich hatte auch das Problem mit Geisterbildern und immer weiter sinkendem Kontrast bei jedem Update - testweise habe ich nun mal (sehr amateurhaft) die Partial-Refreshs entfernt und damit das Problem ersteinmal gefixt. Läuft seither problemlos:

https://github.com/semicuda/ahoy/commit/cf715da282b72ee14e72d454aba34935ff291555

Cheers

knickohr, 2024-06-22:
Etwas Erklärung bitte !
Wenn Du die Partial Refresh entfernt hast, dann schreibt es doch immer über die gleiche Stelle. Würde bedeuten das ein schwarzer Punkt nicht mehr gelöscht wird und im Laufe der Zeit das ganze Feld schwarz wird.

semicuda, 2024-06-22:
Verstehe was du meinst - aber in dem Kontext bedeutet Partial Refresh, dass nur ein bestimmter Teil des Framebuffers (also des Bildschirminhalts im RAM) auf das Display geschrieben wird; der Rest bleibt stehen. Bei einem Full Refresh werden alle Pixel neu ins Display geschrieben, egal ob sie verändert werden oder nicht. Pixelzustände vom vorherigen Inhalt bleiben da definitiv nicht stehen.

knickohr, 2024-06-22:
Ja korrekt.

Man soll aber Full Refresh nicht so oft machen da anscheinend dort das Display scheller „ausgebrannt“ wird. Das es länger dauert ist klar.

semicuda, 2024-06-22:
Eben - und das mit dem Ausbrennen ist mir jetzt ersteinmal egal, funktioniert und sieht gut aus.

Loetnase, 2024-07-16:
Hallo Freunde der geisternden ePaper, 

Ich möchte jetzt mal meine neueren Erkentnisse zu den ePapern weitergeben.
Ich bin ziemlich weit zurück gegangen und dann wieder schrittweise nach vorne, habe dann vor meinem Urlaub noch die 0.8.10-0.8.12 aufgespielt und nach etwas über 16 Tagen sahen die noch richtig gut aus. 
Dann kam der nächste Dreierblock dran und ab 0.8.13 kommen die Geister wieder, ich habe dann mehrmals die Software auf den 3 Geräten gewechselt um die Hardware auszuschliesen. 
Ich würde sagen bis 0.8.12 ist die Software i.O. und ab 0.8.13 starten die Geister.
Im Change Log der 0.8.113 ist da für mich nichts zum ePaper erkennbar.
Kann nur @lumapu die Unterschiede im Quelltext zwischen 0.8.12 und 0.8.13 bezüglich ePaper herausfinden oder ist da auch jemand anderes noch dazu fähig?
Vieleicht kann dann jemand damit eine Version mit einem Lösungsansatz erstellen, welchen ich dann sehr gerne wieder testen werde. Meine Softwarekentnisse sind da leider zu dürftig.  

Ich denke die Abarbeitung dieses Geisterproblems sollte hier im Git erfolgen, wobei ich ab und zu ein paar Infos dazu auch im Discord bekannt geben würde.

[image]

[image]

ich008, 2024-07-17:
Sehen kann die commits jeder. Das einzige zwischen 0.8.12 und 0.8.13 das mir jetzt bezüglich refresh ins Auge springt ist https://github.com/lumapu/ahoy/commit/d3c0232e06ee0211c9d656c46913b357830e2431
Der Rest ist eher Bildschirmschoner, neue Hardware und Zeichensatz kann natürlich auch noch irgendwas in einer Lib sein die jetzt nicht direkt hier im src geändert wurde.

You69Man, 2024-07-18:
Hi all,
ich habe ja damals die Änderungen für die OLED Displays (screensaver, symbols...) gemacht, wollte aber eigentlich das ePaper gar nicht angreifen, da ich selbst keines zum Testen habe. Wie ich nun sehe, gab es aber offenbar einen kleinen Nebeneffekt auf das ePaper, indem der FullRefresh nun alle 480*5 Sekunden, anstatt wie vorher alle 480*10 Sekunden durchgeführt wird.

Ich kann mir das Fehlverhalten des ePaper Refresh rein vom Lesen des Sourcecodes zwar nicht wirklich erklären, zumal nämlich auch vorher schon bei neuen Inverter-Daten (mNewPayload) zusätzliche, azyklische ePaper refreshs durchgeführt wurden. Außerdem hatte Knickohr ja die Ghostings schon mit "0.7.38 und davor" gemeldet, also lange vor meinen OLED-Änderungen. Andererseits sind aber die sehr guten Tests von Lötnase nicht von der Hand zu weisen, das sieht eindeutig nach etwas Systematischem aus.

Ich hätte hier daher einmal einen Vorschlag für einen Fix (#1706), um die Wartung der im Timing doch recht unterschiedlichen OLED und ePaper Displays besser zu trennen, und wieder die ursprüngliche 10s Taktung des ePapers herzustellen. Ob die ePaper Phänomene damit komplett verschwinden würde ich bezweifeln, aber vielleicht treten sie damit wieder seltener auf.

Ein kleiner zusätzlicher Bugfix ist auch mit dabei (mEpaper.tickerSecond() und mEpaper.fullRefresh() wurden immer aufgerufen, auch wenn kein ePaper konfiguriert und initialisiert war, welche Auswirkungen immer das auch hatte...)

Bitte um Review (@lumapu) und Tests durch die ePaper Besitzer (kann es selbst leider wie gesagt nicht testen)!

knickohr, 2024-07-18:
Ich glaube auch nicht das es daran liegt. Denn es gibt Versionen da funktioniert das ePaper besser und dann gibt es wieder Versionen wo es richtig übel ist.

Ich glaube eher da spukt was in den Takt des ePaper rein. Aber man weiß ja nie 😉

Loetnase, 2024-07-19:
Ich denke es liegt auch etwas an der Betriebsdauer der ePaper, ich habe ja alle drei zusammen gekauft und hatte zuerst das mit dem Aufkleber (A) im normalen Betrieb, das zweite zum Ausprobieren, das dritte lag noch so in der Schublade.
Zuerst hat nur das A gegeistert, dann beim Spielen kam langsam das zweite dazu, zum Verifizieren kam dann des dritte auch noch in den Einsatz und das fing auch langsam an zu geistern. Jetzt geistern alle 3 zuverlässig bei den hohen Versionen.
Da spielen Alterung und Software zusammen.

Sobald jemand eine testbare Version gebacken hat, dürft ihr mir sagen wo ich die finde, die teste ich euch gerne.
Ich bin leider kein Held der Software um das selbst zu backen aber lasst es uns probieren.

You69Man, 2024-07-20:
Ich habe hier (#1706) mal zwei Testversionen verlinkt (wroom32 und openDtu)

knickohr, 2024-08-13:
Die Hoffnung stirbt zuletzt 😁

Mit der 132 scheint das Problem endgültig behoben zu sein. Ich beobachte noch ein paar Tage 😉

Loetnase, 2024-08-13:
Ich lass die 132 auch noch ein paar Tage laufen 🏃🏃🏃‍♀️
Dann hoffen wir mal, dass das keine Eintagsfliege war 😇

closed 2024-08-16 by lumapu