lumapu/ahoy #1110: after update to 0.7.36 inverter #5 is missing and can not be added manually

gudruninfranken, 2023-08-22:
With my setup I was using Ahoy DTU for monitoring 5 inverters. After update there are only 4 visible. I am not able to add the missing one manually. I tried to import my last system setup, but it also shows 4 instead of 5 inverters. Do you have any idea to solve it?

## Hardware
  * [ ] ESP8266

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode

### Antenna:
* [ ] circuit board

knickohr, 2023-08-22:
As you see in the Documentation/Changelog, the ESP8266 now only supports Max. 4 Inverters since Version 0.7.26.

https://github.com/lumapu/ahoy/releases

lumapu, 2023-08-22:
You have to move to ESP32 which is in general the better choice. There you can add 16 inverters.
Maybe we can support more inverters than 4 in future, but for now there are no plans to do so.

tux011, 2023-08-27:
I have the same Problem, the "Add Inverter Button" is available but without any function - no error.. No hint in the Docu. Please write down clear this **NEW** limitation for the ESP8266. Thx. Just a hint for the ESP266 Users the version 0.5.66 works fine also with 6 inverters if you are Using: "interval":"5","retries":"1". So I have to do a downgrade - to prevent to change the Hardware. Danke! ....

knickohr, 2023-08-27:
Dazu gibt es (leider) nicht mehr zu sagen, bzw. zu schreiben 🤔 Ist schon seit 3 Releases so  :

https://github.com/lumapu/ahoy/releases

[image]

gudruninfranken, 2023-08-27:
Thank you very much. I did not read it.  I ordered the hardware for ESP32 and will check if the solution is more satisfying than the former solution with two ESP8266 with each 5 inverters.

knickohr, 2023-08-27:
I‘m running 13 Inverters on an ESP32 😉

closed 2023-12-29 by lumapu