lumapu/ahoy #1111: MI - extend Model support

rejoe2, 2023-08-22:
Got some additional fields for the MI-600 (2nd gen) working:
[image]

Pls. help to "close the circle", as it's unclear to me (bitshifting noob...), how these information parts have to play together to get that completely solved...

Fragments:
- add MI's to const devInfo_t devInfo[] in hmDefines.h
Tried with 
```
 const devInfo_t devInfo[] = {
    // MI 2nd gen
    { 0x000013, 600 },
    { 0x007041, 600 },

```
- modify miPayload.h (see at the end) 

This will output sth. like:
```
11:15:37 I: (#6) resetPayload
11:15:37 I: (#6) Requesting Inv SN 104160100013
11:15:37 I: (#6) prepareDevInformCmd 0x05
11:15:37 I: TX 11B Ch40 | 0f 60 10 00 13 86 09 76 68 00 fd 
11:15:37 I: RX 27B Ch75 | 8f 60 10 00 13 60 10 00 13 00 00 ce 00 ec 07 e2 04 67 03 52 00 11 00 01 70 41 5b 
11:15:37 I: (#6) I: HW_VER is 28737
11:15:37 70 41
11:15:37 I: RX 27B Ch61 | 8f 60 10 00 13 60 10 00 13 01 00 13 11 8f 0e 00 04 cd 00 14 22 60 00 11 00 01 82 
11:15:37 I: (#6) got 2nd frame (hw info)
11:15:37 I: HW_PartNo 1249679
![Screenshot_20230822_110950](https://github.com/lumapu/ahoy/assets/18676801/32346fb0-0212-40a7-8868-b80e2e419c4f)

11:15:37 I: HW_FB_TLmValue 3584
11:15:37 I: HW_FB_ReSPRT 1229
11:15:37 I: HW_GridSamp_ResValule 20
11:15:37 I: HW_ECapValue 8800

```

miPayload.h:
```//-----------------------------------------------------------------------------
// 2023 Ahoy, https://ahoydtu.de
// Creative Commons - http://creativecommons.org/licenses/by-nc-sa/3.0/de/
//-----------------------------------------------------------------------------

#ifndef __MI_PAYLOAD_H__
#define __MI_PAYLOAD_H__

//#include "hmInverter.h"
#include "../utils/dbg.h"
#include "../utils/crc.h"
#include "../config/config.h"
#include <Arduino.h>

typedef struct {
    uint32_t ts;
    bool requested;
    bool limitrequested;
    uint8_t txCmd;
    uint8_t len[MAX_PAYLOAD_ENTRIES];
    bool complete;
    bool dataAB[3];
    bool stsAB[3];
    uint16_t sts[6];
    uint8_t txId;
    uint8_t invId;
    uint8_t retransmits;
    bool gotFragment;
    /*
    uint8_t data[MAX_PAYLOAD_ENTRIES][MAX_RF_PAYLOAD_SIZE];
    uint8_t maxPackId;
[... 730 lines cut ...]
            if (clrSts) {                    // only clear channel states at startup
                mPayload[id].sts[CH1]    = 0;
                mPayload[id].sts[CH2]    = 0;
                mPayload[id].sts[CH3]    = 0;
                mPayload[id].sts[CH4]    = 0;
                mPayload[id].sts[5]      = 0; //remember last summarized state
            }
        }

        IApp *mApp;
        HMSYSTEM *mSys;
        HMRADIO *mRadio;
        statistics_t *mStat;
        uint8_t mMaxRetrans;
        uint32_t *mTimestamp;
        miPayload_t mPayload[MAX_NUM_INVERTERS];
        bool mSerialDebug;

        Inverter<> *mHighPrioIv;
        alarmListenerType mCbAlarm;
        payloadListenerType mCbMiPayload;
};

#endif /*__MI_PAYLOAD_H__*/
```

lumapu, 2023-08-22:
cool, dass du hier auch an die Mi Wechselrichter denkst 👍. Sieht doch schon sehr gut aus, wo genau kann man dir helfen?

stefan123t, 2023-08-22:
@lumapu also das ist die HW_PartNo 1249679 als Hex ?
Wundere mich dass der andere Wert 0x7041 und nicht 0x1041 ist. Hmm ?
@rejoe2 Da gab es doch ein Excel mit den ganzen Gen2 Commandos und Frames/Paketen, hast Du das evtl grade vorliegen ?

lumapu, 2023-08-23:
genau, es wird die HW PartNo. mit der maximalen Leistung verknüpft.

rejoe2, 2023-08-23:
https://github.com/rejoe2/ahoy/commit/20713c9ffbf378a968bb77222616d40d77509dc1

Leider scheint da in der Tat noch was zu fehlen, das ist der output mit dem obigen Code-Stand:
[image]

rejoe2, 2023-08-23:
Nachtrag: habe jetzt nochmal die xlsx durchgesehen und die betreffenden Abschnitte zu 0x0f und den Antworten im Code als Kommentar hinterlegt, so ist es einfacher zu finden (Achtung: die Byte-Zählung weicht um 1 ab!). 
Weiter sollte jetzt sichergestellt sein, dass auch der 2. Frame abgefragt wird, falls er "durchgerutscht" war und nur die FW-Infos ausgewertet werden konnten.

rejoe2, 2023-08-24:
Habe wieder etwas rumgespielt, jetzt ist die HW-Part-Nr. wirklich da und zum Schluss wird auch noch "notify()" aufgerufen (habe mit diversen Varianten gespielt) - bisher ist die angezeigte Leistung immer "0"... Irgendein Bruchstückchen fehlt also immer noch.

beegee3, 2023-08-29:
hallo zusammen, hab' mal wieder Zeit gefunden, die issues zu lesen und @rejoe2 hat immer spannende Sachen!

Der Eintrag in hmDefines.h muss m.E. so lauten:
```
const devInfo_t devInfo[] = { // MI 2nd gen
    { 0x001311, 600 },
    ... 
```
denn das letzte Byte der HW-PN (hier `8f`) gibt "nur" die Hardwarevariante an.
Alle HW-PN kann man in `sunspec.c` des China Sourcecodes nachlesen.
Hier der relevante Teil:
```
                InverterDetail_Read((InverterDetail *)&MI_Detail, mi_place, 1);
                HW_PN = ((u32)MI_Detail.Property.HW_PNH << 16) + MI_Detail.Property.HW_PNL;

                switch(HW_PN)
                {
                    case 0x0001000F:
                        sprintf((char *)&sunspec_buf[x * 2 + 0], "MI-250T");
                        break;

                    case 0x0003000F:
                        sprintf((char *)&sunspec_buf[x * 2 + 0], "MI-300T");
                        break;

                    case 0x0011000F:
                        sprintf((char *)&sunspec_buf[x * 2 + 0], "MI-500T");
                        break;

                    case 0x0013000F:
                        sprintf((char *)&sunspec_buf[x * 2 + 0], "MI-600T");
                        break;

                    case 0x0001118F:
                        sprintf((char *)&sunspec_buf[x * 2 + 0], "MI-250");
                        break;

                    case 0x0003118F:
                        sprintf((char *)&sunspec_buf[x * 2 + 0], "MI-300");
                        break;

                    case 0x0004118F:
                        sprintf((char *)&sunspec_buf[x * 2 + 0], "MI-350");
                        break;

                    case 0x0011118F:
                        sprintf((char *)&sunspec_buf[x * 2 + 0], "MI-500");
                        break;

                    case 0x0013118F:
                        sprintf((char *)&sunspec_buf[x * 2 + 0], "MI-600");
                        break;

                    case 0x0013218F:
                        sprintf((char *)&sunspec_buf[x * 2 + 0], "MI-600LV");
                        break;

                    case 0x0014218F:
                        sprintf((char *)&sunspec_buf[x * 2 + 0], "MI-700LV");
                        break;

                    case 0x0014118F:
                        sprintf((char *)&sunspec_buf[x * 2 + 0], "MI-700");
                        break;

                    case 0x0021118F:
                        sprintf((char *)&sunspec_buf[x * 2 + 0], "MI-1000");
                        break;

                    case 0x0023118F:
[... 575 lines cut ...]

                    case 0x10001007:
                        sprintf((char *)&sunspec_buf[x * 2 + 0], "MI-300(R)(SO)");
                        break;

                    case 0x10000000:
                        sprintf((char *)&sunspec_buf[x * 2 + 0], "MI-250");
                        break;

                    case 0x10000001:
                        sprintf((char *)&sunspec_buf[x * 2 + 0], "MI-250(R)");
                        break;

                    case 0x10000002:
                        sprintf((char *)&sunspec_buf[x * 2 + 0], "MI-250(R)(SO)");
                        break;

                    default :
                        sprintf((char *)&sunspec_buf[x * 2 + 0], "---");
                        break;
                }
```

Hab's nicht geprüft, aber evtl. sind da noch Inverter angegeben, die bisher nicht in `devInfo` angegeben sind.
Ggfs. kann man das nutzen, um `devInfo` in `hmDefines.h` zu ergänzen. - Wer auch immer Zeit und Lust dazu hat 😄

lumapu, 2023-08-29:
Hey @beegee3 willkommen zurück! Schön, dass du wieder bei uns bist 😊.
Die Liste ist ja genial, werde gleich mal durchschauen was noch fehlt, bzw. welche Infos wir noch "rausziehen" können. 
Btw. gibt es das China-Repo eigentlich noch, ich habe es nie "gezogen".

beegee3, 2023-08-29:
@lumapu: ich lese mit, wann immer ich Zeit habe, was mittlerweile aber leider sehr selten ist.
Noch seltener find' ich Zeit, auch selber was zu schreiben (wie neulich zum FTP). Hoffe, das ändert sich mal wieder.
Habe vor langer Zeit mal die (meiner Meinung nach) wesentlichen Dateien aus dem China-Repo kopiert.
Aber das gibt's immer noch: https://gitee.com/iotloves/hoymiles-DTU-PRO/

lumapu, 2023-08-29:
@stefan123t gibt es von deiner Seite hierzu auch schon Dokumentation?
Ich habe mal angefange zu "interpretieren". Geht aber erst ab der 2.ten? MI Generation (Start mit `0x10...`)

https://github.com/lumapu/ahoy/blob/201098ae0bc1874c5b4a81a1187d0a1b840aca12/src/hm/hmDefines.h#L262-L329

rejoe2, 2023-08-30:
Thx (einmal mehr! und auch für die "warmen Worte") @beegee3 ; das mit 0x001311 war der richtige Ansatz!

PR ist eingereicht :grin: .

PS: vielleicht reicht die Zeit, mal https://github.com/lumapu/ahoy/pull/1080/files anzusehen?
Werde vermutlich nicht so schnell dazu kommen, mir das in Ruhe anzusehen und ggf. testweise in den Code einzupflegen, ist aber vom Ansatz her m.E. durchaus interessant....

rejoe2, 2023-08-30:
Ähm, ich sehe grade, dass es da noch eine Reihe weiterer MI-Varianten gibt...

Bei den 0x1-er Nummern dürfte es sich um 3rd gen. Modelle handeln, beim Rest ist es unklar. Würde ggf. einfach mal diesen Teil "as is" einpflegen, und die MI-Community dann mal auf discord befragen, inwie weit das paßt bzw. noch korrekturbedürftig ist?

rejoe2, 2023-08-30:
@beegee3 - Im Hinblick auf die MI sind im Moment noch zwei Themenkreise offen:
a) vollständige Auswertung (wenigstens) der Infos, die die Dinger liefern. Da gibt es außer der HWPartNr noch einige Infos mehr, die miPayload.h jetzt halt einfach mal seriell rausgibt. Eigentlich müßte sich dahinter auch sowas wie ein grid profile etc. verbergen, aber jedenfalls ich habe keine Idee, wie man das in die übliche HM-Struktur "übersetzt" bekommen könnte.
b) "alarms": Die MI kennen ja jeweils nur einen Status pro Eingang, aber wenn der "problematisch" ist (insbesondere undervoltage kennen wir mit Code 2), könnte man daraus einen "alarm" generieren und in die normale Struktur schreiben? (ggf. mache ich dazu einen gesonderten issue auf, wenn wir den Rest hier erledigt haben?)

rejoe2, 2023-08-31:
> @rejoe2: zu a) zunächst muss man wissen, was sich hinter den Informationen der 3 Rückgabeframes zur Anfrage 0x0f verbirgt (was bedeutet HW_FB_TLmValue, HW_FB_ReSPRT, HW_GridSamp_ResValule, HW_ECapValue, usw.) Aber inwiefern nutzen diese Infos? (braucht man das?) 
Was sich dahinter verbirgt, kann ich auch nicht beantworten - das sind einfach die Feldbezeichnungen, die aus der xlsx kommen. Was ich ggf. liefern kann, sind die (übersetzten) Echt-Werte, wobei das im Rohformat bereits im ersten Beitrag hier stand.
Kurz: Keine Ahnung, aber irgendwie müßte es ja auch bei den MI sowas geben wie ein "grid profile", und wenn, dann müßte es sich irgendwie darin finden - so jedenfalls meine "noobische" Denke.... 

>zu b) ich dachte, in den Antworten zu 0x09 (A channel) und 0x11 (B channel) wird mit Status, Fault_Num und Cnt_Fault alles notwendige geliefert. Was ist anders als bei HM Invertern?
Habe das bisher nicht geloggt, aber bei den Fault-Infos nie was anderes gesehen wie Nullen. 
Das bedeutet, dass man eigentlich nur die Status-Info wirklich nutzen kann, aber da ist es eben pro Kanal und "jetzt"; Änderungen müßte man daher selbst tracken, eine (abrufbare) history gibt es nicht, und (andere) Zeitpunkte eben auch nicht.

rejoe2, 2023-08-31:
@beegee3 
(sorry, die vorherige Nachricht hat es beim Editieren zerschreddert...)

ad a) Als noob in diesen Themen erschließen sich mir diese Kürzel nicht (die stammen aus dem xlsx), aber es ist doch davon auszugehen, dass "irgendwo" die Daten zum gridprofile oä. stehen. Mir fällt kein anderer Ort ein wie da....

ad b) Na ja, es gibt einfach keine abfragbare history (so verstehe ich den HM-Code?). Das bedeutet, man muss alle Infos aus dem "hier und jetzt" ziehen.

Habe grade mal den Test gemacht und die Inverter von AC genommen. Das macht der MI-600 im Moment draus (Ende dann mit 3 - inverter start, ohne dass irgendwelche Errors noch berichtet würden):
```
15:09:41 I: (#6) next request is 0x11
15:09:41 I: TX 11 CH40 | 11 60 10 00 13 86 09 76 68 11 f2 
15:09:41 I: RX 16 CH75, -75dBm | 92 60 10 00 13 60 10 00 13 00 05 00 08 00 01 9e 
15:09:41 W: Status change for CH2 (5080): Grid Overvoltage/Undervoltage
15:09:41 I: (#6) alarm ID incremented to 5080
15:09:41 I: (#6) sth. missing: Request Retransmit 0x11
15:09:41 I: TX 11 CH61 | 11 60 10 00 13 86 09 76 68 11 f2 
15:09:42 I: RX 24 CH23, -75dBm | 91 60 10 00 13 60 10 00 13 01 8a 00 00 01 70 00 00 03 d6 04 de 01 89 ec 
15:09:42 I: (#6) next request is 0x09
15:09:42 I: TX 11 CH75 | 09 60 10 00 13 86 09 76 68 09 f2 
15:09:42 I: RX 16 CH23, -75dBm | 88 60 10 00 13 60 10 00 13 00 05 00 08 00 01 84 
15:09:42 W: Status change for CH1 (5080): Grid Overvoltage/Undervoltage

15:11:33 I: (#6) enqueCommand: 0x09
15:11:33 I: (#6) enqueCommand: 0x05
15:11:33 I: TX 11 CH75 | 09 60 10 00 13 86 09 76 68 05 fe 
15:11:33 I: RX 16 CH23, -75dBm | 88 60 10 00 13 60 10 00 13 00 05 00 09 00 01 85 
15:11:33 W: Status change for CH1 (5090): Grid Overfrequency/Underfrequency
15:11:33 I: (#6) alarm ID incremented to 5090
15:11:33 I: (#6) sth. missing: Request Retransmit 0x09
15:11:33 I: TX 11 CH3 | 09 60 10 00 13 86 09 76 68 09 f2 
15:11:33 I: RX 16 CH23, -75dBm | 88 60 10 00 13 60 10 00 13 00 05 00 09 00 01 85 
15:11:33 I: (#6) sth. missing: Request Retransmit 0x09
15:11:33 I: TX 11 CH23 | 09 60 10 00 13 86 09 76 68 09 f2 
15:11:33 I: RX 24 CH75, -75dBm | 89 60 10 00 13 60 10 00 13 01 8b 00 00 09 2f 13 88 00 09 04 e7 01 83 d6 
15:11:33 I: (#6) next request is 0x11
15:11:33 I: TX 11 CH40 | 11 60 10 00 13 86 09 76 68 11 f2 
15:11:34 I: RX 16 CH75, -75dBm | 92 60 10 00 13 60 10 00 13 00 05 00 09 00 01 9f 
15:11:34 W: Status change for CH2 (5090): Grid Overfrequency/Underfrequency
15:11:34 I: RX 24 CH3, -75dBm | 91 60 10 00 13 60 10 00 13 01 89 00 00 09 2f 13 88 00 06 04 de 01 83 fa 
15:11:34 I: (#6) got all msgs
```

beegee3, 2023-08-31:
zu a) hab' ich Info im China-Repo gefunden:
```
USFWBuild_VER                //bootloader version
AppFWBuild_VER               //application version
AppFWBuild_YYYY              //App creation time
AppFWBuild_MMDD
AppFWBuild_HHMM
AppFW_PN                     //software part number
HW_VER                       //hardware version
HW_PN                        //Hardware part number
HW_FB_TLmValue               //The value of the excitation inductance
HW_FB_ReSPRT                 //Resonance period
HW_GridSamp_ResValule        //Use the resistance value
HW_ECapValue                 //Electrolytic capacitor value
Matching_AppFW_PN            //The software part number for hardware matching
AppFW_MINVER                 //The minimum version number of the software required by the hardware

GPFCode                      //Grid protection file code
GPFVer                       //The version of the grid protection file
```
Laut Excel File gibt es zum Gridprofile die Info (GPFCode und GPFVer) nur in den rot markierten Bereichen (not Legacy) über die Abfrage `0x0f xx xx xx xx YY YY YY YY 0x01` also eigentlich das gleiche wie 'Polling the device software and hardware version number command' nur mit User data = 0x01 statt 0x00. Probier' das doch mal aus, vielleicht kommt ja etwas zurück.

zu b)
die Antworten 88 und 92 in deinem Protokoll: `... 00 05 00 08 00 01` bzw. `... 00 05 00 09 00 01` heißen doch
Status = 00 05
Fault_Num = 00 08 bzw. 00 09
Cnt_Fault = 00 01

Du hast einen Fehler produziert und bekommst ihn angezeigt, soweit also alles richtig!
Ich verstehe aber die `Status change for CH1 (5080): Grid Overvoltage/Undervoltage` und `Status change for CH1 (5090): Grid Overfrequency/Underfrequency` Meldungen nicht (entsprechend für CH2).
Wo kommen die her? Sind irgendwie merkwürdige Verschiebungen (Kombinationen) von Status und Fault_Num, d.h. 05 und 08 bzw. 05 und 09! Wobei die Status change Zahlen dezimal, die Fault_Num Werte aber hexadezimal sind. Passt nicht!!!
Die Fehler 8 = 00 08 und 9 = 00 09 sind allerdings nicht in der `getAlarmStr` Liste.
Sie könnten - deinem provoziertem Fehler entsprechend - so etwas wie 'AC disconnected' und 'AC reconnected' bedeuten.
Btw. - woher stammt eigentlich die `getAlarmStr` Liste?

rejoe2, 2023-08-31:
Das mit user data 0x01 probiere ich mal aus, allerdings könnte (!) es sein, dass das mal eingebaut gewesen war, aber keine Antworten kamen (muss ich nachsehen, und es wird vermutlich einige Tage dauern).

Das mit den Fehlern ist "einfach": Da keiner auf meine diesbezüglichen Rückfragen reagiert hatte, habe ich irgendwann eben den Versuch unternommen, das "irgendwie" auf die "allgemeine Alarm-Code-Liste" umzubiegen (das ist in den miPayload-sts-message-Auswertungen zu finden). Kann schon sein, dass da Fehler drin sind, Ziel war es erst mal, überhaupt eine Brücke zu haben, um das ganze am Ende eben möglichst ähnlich zu haben....

LeSpocky, 2023-09-19:
Wieso ist denn in 201098ae0bc1874c5b4a81a1187d0a1b840aca12 die 0x101040 rausgeflogen?

lumapu, 2023-09-19:
keine Ahnung, evtl. ist die in der China Liste nicht aufgetaucht, machst du einen PR oder schreibst einen neuen issue, hier finde ich die Info nie wieder ... 😇

beegee3, 2023-09-19:
ist nicht in der China Liste, aber einfach
`{ 0x101040,  400 },`
bei `hmDefines` nach Zeile 309 einfügen

LeSpocky, 2023-09-20:
> keine Ahnung, evtl. ist die in der China Liste nicht aufgetaucht, machst du einen PR oder schreibst einen neuen issue, hier finde ich die Info nie wieder ... 😇

Siehe PR #1169 dann.

rejoe2, 2023-09-21:
> zu a) hab' ich Info im China-Repo gefunden:
> 
> GPFCode                      //Grid protection file code
> GPFVer                       //The version of the grid protection file
> ```
> 
> Laut Excel File gibt es zum Gridprofile die Info (GPFCode und GPFVer) nur in den rot markierten Bereichen (not Legacy) über die Abfrage `0x0f xx xx xx xx YY YY YY YY 0x01` also eigentlich das gleiche wie 'Polling the device software and hardware version number command' nur mit User data = 0x01 statt 0x00. Probier' das doch mal aus, vielleicht kommt ja etwas zurück. Ansonsten gibt es ja noch den 'Polling Grid Protection Parameter File Command' 0x10, der den 'grid-connected protection parameter status' und den 'national standard code' liefert.
> 

Das mit der 0x01 war schon eingebaut gewesen, da kommen einfach nur wieder alle frames wie bei subcmd 0x00 auch.

Das hier liefert die legacy-Abfrage: 
```
11:35:55 I: (#6) prepareDevInformCmd 0x10
11:35:55 I: TX 11 CH3 | 10 60 10 00 13 86 09 76 68 00 e2 
11:35:55 I: RX 15 CH40, -75dBm | 90 60 10 00 13 60 10 00 13 02 08 07 11 00 8c 
11:35:55 I: ST1 2
11:35:55 I: CountryStd 2055
11:35:55 I: Version 4352
```

Code-Schnippsel dazu wäre:
```
void miGPFDecode(Inverter<> *iv, packet_t *p ) {
            mPayload[iv->id].gotFragment = true;
            mPayload[iv->id].fragments++;
            mPayload[iv->id].gotGPF = true;

/* according to xlsx (different start byte -1!)
 Polling Grid-connected Protection Parameter File Command - Receipt
 byte[10]               ST1 indicates the status of the grid-connected protection file. ST1=1 indicates the default grid-connected protection file, ST=2 indicates that the grid-connected protection file is configured and normal, ST=3 indicates that the grid-connected protection file cannot be recognized, ST=4 indicates that the grid-connected protection file is damaged
 byte[11]	 byte[12]   CountryStd variable indicates the national standard code of the grid-connected protection file
 byte[13]	 byte[14]   Version indicates the version of the grid-connected protection file
*/
            if(mSerialDebug) {
                DPRINT(DBG_INFO,F("ST1 "));
                DBGPRINTLN(String(p->packet[9]));
                DPRINT(DBG_INFO,F("CountryStd "));
                DBGPRINTLN(String((p->packet[10] << 8) + p->packet[11]));
                DPRINT(DBG_INFO,F("Version "));
                DBGPRINTLN(String((p->packet[12] << 8) + p->packet[13]));
            }
        }
```
Gibt es irgendwelche HM-Datenfelder, die man damit sinnvoll befüllen könnte?

stefan123t, 2024-09-10:
> @stefan123t gibt es von deiner Seite hierzu auch schon Dokumentation?
> 
> Ich habe mal angefange zu "interpretieren". Geht aber erst ab der 2.ten? MI Generation (Start mit `0x10...`)
> 
> 
> 
> https://github.com/lumapu/ahoy/blob/201098ae0bc1874c5b4a81a1187d0a1b840aca12/src/hm/hmDefines.h#L262-L329

Ja das sieht schon ganz übersichtlich aus im Code.

Außer den Excel Listen mit den MI/HM und DTU/Repeater Model Serial IDs aus dem gitee.com Repo habe ich da auch nichts.

Für die neuen OEM Versionen gibt es ja dann so eine krude Serial ID Conversion Routine, die aus Buchstaben wieder Hex/Serial IDs macht.

stefan123t, 2024-10-29:
## Micro-inverse system related product ID coding rules

### 1. Introduction
Hemai micro-inverter system products include micro-inverters, DTUs, repeaters, smart meters (outsourced), etc., and their coding rules are consistent.
The serial number of the related products of the micro-inverse system is a total of 6 bytes, that is, a 12-digit hexadecimal number, which is defined as follows:
`FFFF FFFF FFFF`
Different subdivision products have slightly different definitions. For details, please refer to the relevant regulations in specific chapters.

### 2. ID coding rules
#### 1. The first four rules

The 1st to 4th digits are the **model code**, among which
* the **1st** digit is the **product category**, and `1` represents the related products of the micro-inverse system.
* the **2nd~3rd** digit is the **product category**. At present, the related products of the micro-inversion system are:

**2-3** digit value | Product Category | Involved product model
--- | --- | ---
`01` | One drag series (four-core bus version) | MI-250T1, MI-300T1, MI-350T1
`02` | one drag series | MI-250, MI-300, MI-350, MI-250T, MI-300T, MI-350T
`03` | One for two series (four-core bus version) | MI-500T1, MI-600T1
`04` | One for two HM, MI series | MI-500,MI-600, MI-700, MI-500T,MI-600T, MI-700T, HM-500,HM-600, HM-700, HM-500T,HM-600T, HM-700T
`06` | One for four HM, MI series | MI-1000,MI-1200, MI-1500, MI-1000T,MI-1200T, MI-1500T, HM-1000,HM-1200, HM-1500, HM-1000T,HM-1200T, HM-1500T
`16` | HM Pro one drag four series | HM-1200 Pro, HM-1500 Pro, HM-1200T Pro, HM-1500T Pro
`0C` | smart meter | Chint DDSU666, DTSU666, etc.
`0D` | Minimalist DTU | DTU-G100, DTU-W100, DTU-Lite-GPRS, DTU-Lite-WIFI
`0E` | repeater | RP-433-ICR
`0F` | DTU | DTU-MI, DTU-433, DTU-MI-GPRS, DTU-433-GPRS, DTU-MI-AR, DTU-MI-ARW, DTU-MI-GPRS-ARW, DTU-Pro-GPRS, DTU-Pro-WIFI
`9X` | Production test fixture | 

* the **4th** digit represents the small **generation** of the product, please refer to the following two tables for detailed description.
At present, the model codes of each version of the micro-inverse are as follows:

Micro inverse series | One drag series (four-core bus version) | one drag series | One drag two series<br>(four-core bus version) | One drag two series | One to four series | HM Pro one drag four series
--- | --- | --- | --- | --- | --- | ---
2M baud rate communication version | `1010` | `1020` | `1030` | `1040` | `1060`
250k baud rate communication version |  | `1021` |  | `1041` | `1061` | `1161`
No communication version | `1019` | `1029` | `1039` | `1049` | `1069` |  | 

At present, the **model codes** of each version of repeater, DTU and smart meter are as follows:

version | model codes
--- | ---
DTU-MI (2.4G RF version) | `10F0`
DTU-433 (433M RF version) | `10F1`
DTU-MI-GPRS (2.4G RF version with GPRS) | `10F2`
DTU-433-GPRS (433M RF version with GPRS) | `10F3`
DTU-MI-AR (anti-backflow version) | `10F4`
DTU-MI-ARW (three-phase wired anti-backflow version) | `10F5`
DTU-MI-GPRS-ARW (three-phase wired GPRS anti-backflow version) | `10F6`
DTU-Pro-GPRS | `10F7`
DTU-Pro-WIFI | `10F8`
Common DTU reserved version | `10F5`-`10FF`
DTU-G100 (GPRS minimalist DTU) | `10D0`
DTU-W100 (WIFI minimalist DTU) | `10D1`
DTU-Lite-GPRS | `10D2`
DTU-Lite-WIFI | `10D3`
Minimalist reserved version | `10D4`-`10DF`
repeater | `10E0`
Relay reserved version | `10E1`-`10EF`
Smart meter (outsourced) | `10C0`
Smart meter reservation | `10C0`~`10CF`

Note: IDs for reserved versions are also open.

#### 2. Last eight coding rules

For products such as micro-inverters, DTUs, repeaters, etc., the rules are as follows:
The **5th** to **12th** digits are the **production serial number** and are used as the **RF communication address** of the micro-inverter/DTU/repeater, among which:
The **5th** digit is the **year of production** (time provided by the welder), the `1` is for 2015, and so on
The **6th~7th** digits are the **production week** (the time provided by the welding factory), and the range is `1~52`.
The **8th~12th** digits are the **pipeline coding** (currently in decimal), the micro-inverter, repeater, and DTU share the same pipeline sequence, which is coded in sequence and must not be repeated, a total of 99999 digits.

For smart meter products, the rules are as follows:
The **5th** to **12th** digits are the **production serial number**, and include the **RS485 communication address** of the smart meter, among which:
The **5th** digit is the **production year** (the time when the smart meter was shipped in our company), `1` represents 2015, and so on
The **6th~7th** digits are the **production week** (the time when the smart meter is shipped in our company), and the range is `1~52`.
The **8th** digit is the name of the **meter supplier**, `1` represents Chint, and `2~F` is reserved.
The **9th** digit represents the **meter type**, which may be different for different suppliers. For Chint smart meters, the regulations are as follows:

Description of Chint Smart Meter | **8th~9th** digits
--- | ---
Single-phase smart meter DDSU666 (conventional product, without Ct) | `11`
Three-phase four-wire smart meter DTSU666 (conventional product, with Ct, variable ratio can be set) | `12`
Three-phase four-wire smart meter DTSU666 (conventional product, without Ct) | `13`
Three-phase four-wire smart meter DTSU666 (conventional product, with Ct, the transformation ratio cannot be set, the default is 100A/40mA) | `14`
[... cut ...]
Three-phase four-wire smart meter DTSU666 (conventional product, with Ct, the transformation ratio cannot be set, the transformation ratio is to be determined, reserved) | `16~1F`

The **10th~12th** digits are the **serial code** (represented in decimal), representing the **communication address** of the meter (range `001~247`).

Note: 1. The outsourced Chint smart meter needs to be configured with a communication address and affixed with our label when shipping. Different types of meters must not be confused; The number of electricity meters is less than 247).