lumapu/ahoy #1113: [Bug]

Gubi2023, 2023-08-22:
### Platform

ESP8266

### Assembly

the DTU was already assembled

### Antenna

circuit board

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.39

### Github Hash

0.7.39

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Inverter updated all 15 sec

### Error description

ist die Anzeige der Alarme so korrekt? angezeigt werden 18 Alarme, in der Liste sind aber nur 10
[image]
[image]

Gubi2023, 2023-08-22:
> Es werden nur die letzten 10 angezeigt. Doppelte zeitgleiche sollten
> eigentlich nicht vorkommen 😲 Bitte mal auf die 0.7.40 updaten.
> <https://github.com/lumapu/ahoy/issues/1113#issuecomment-1688094088> , or
> unsubscribe
> [ { ***@***.***": "http://schema.org", ***@***.***": "EmailMessage",
> "https://github.com/lumapu/ahoy/issues/1113#issuecomment-1688094088",
> "url":
> "https://github.com/lumapu/ahoy/issues/1113#issuecomment-1688094088",
> "name": "View Issue" }, "description": "View this Issue on GitHub",
> "https://github.com" } } ]

Hallo, habe jetzt 0.7.40 drauf, an der Sortierung solltet ihr noch etwas 
feilen  [wink]
Siehe Screenshot
 
Grüsse
Günter Czarny

Gubi2023, 2023-08-30:
version 0.7.44: Alarme werden zyklisch wiederholt: 
[image]

Gubi2023, 2023-09-07:
Ende-Datum falsch:
siehe Screenshot
[image]

lumapu, 2023-09-07:
die Zeiten kommen vom Wechselrichter. Darauf haben wir keinen Einfluss.

closed 2023-10-02 by Gubi2023