lumapu/ahoy #1123: Verbesserungsvorschlag: wäre es möglich bei  "Max power" die Uhrzeit mit anzuzeigen?

Gubi2023, 2023-08-26:
## Hardware
  * [ ] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Gubi2023, 2023-08-26:
Verbesserungsvorschlag: wäre es möglich bei "Max power" die Uhrzeit mit anzuzeigen?

DanielR92, 2023-08-26:
An sich wäre das möglich. Jedoch aktuell noch nicht ganz dringlich hier. :)
Oder wie sehen das die anderen?

MetaChuh, 2023-08-26:
@DanielR92

nichts ist wirklich dringlich und alles ist wirklich cool 😉

ideen sind immer gut.
viele haben max power mit timestamp sowieso in ihrer steuerung, jedoch wenn es schon standalone diesen wert ab jetzt gibt, ist die idee nicht dringlich, aber ergänzend.

reines brainstorming

Gubi2023, 2024-01-05:
danke, passt.

closed 2024-01-05 by Gubi2023