lumapu/ahoy #1124: [Bug] last_success kommt mir zu oft 😅

knickohr, 2023-08-26:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.41 und davor

### Github Hash

1234567890

### Build & Flash Method

ESP Tools (flash)

### Setup

Multiple Inverters, ePaper, MQTT

### Error description

Das last_success kommt mir ein bißchen zu oft. Bei jedem Inverterticket werden nicht nur die für diesen Inverter spezifischen Werte und sein eigenes last_success ausgegeben, sondern auch die last_success der anderen an der DTU angeschlossenen (obwohl sich deren Werte noch nicht geändert haben).

[image]

knickohr, 2023-08-28:
Nein @lumapu , noch nicht fixed 😲

lumapu, 2023-08-28:
und warum?

knickohr, 2023-08-28:
Hier mal die „Tickets“ von 2 unterschiedlichen Invertern. Warum kommen beim Inverter 1 auch die last_success von 2 bis 6, bzw. beim Inverter 2, die von 1 und 3 bis 6 ?

[image]

Wie man sieht ändern sich auch nur die Werte von dem betroffenen Inverter, die anderen nicht.

knickohr, 2023-09-19:
Leider mit der neuen DEV 57 immer noch nicht ganz sauber. Die Alarms sind zwar jetzt weg, nur noch die nötigen, aber das last_success dominiert immer noch heftig.

[image]

Mir kommt es vor, als wenn etwas nicht stimmt mit einem Inverter (keine Antwort, Fehler in der Übertragung, Retransmit, …8 dann ballert er immer einen kompletten Satz last_success für alle Inverter raus.

knickohr, 2023-09-20:
DEV 58 zeigt Ähnliches Startverhalten wie die 51. bis sich alle Inverter ordentlich gemeldet haben ist es durchsetzt mit unendlich vielen last_success. Danach läuft es recht gut, nur wenn ein Inverter sich nicht meldet donnert er für alle Inverter das last_success raus. Da ist noch Optimierungsbedarf 😁

knickohr, 2023-09-20:
@lumapu 

Workaround von @rejoe2 um zumindest die last_success mit Wert 0 zu unterbinden, die ja eh keine wirklichen Information bereitstellen :

Das mit der "0" bekomme ich in den Griff, allerdings über einen workaround. Für eine wirkliche Lösung müßte imo die publish-Funktion prüfen, zu welchem Inverter der tx-Command eigentlich gehört, und nicht einfach davon ausgehen, dass es für alle dasselbe ist....
Zeile 106 in pubMqttIvData.h
```
  if(mIv->getLastTs(rec) != mIvLastPublish[mIv->id] && mIv->getLastTs(rec) != 0) { 
```

knickohr, 2023-09-21:
Noch nicht ganz glücklich damit, aber schon wesentlich besser. Mich stört halt immer noch das wenn ein Inverter mal nicht erreichbar ist, immer die llast_success von allen Inverttern raus geballert werden, anstatt nur das vom betroffenen.

rejoe2, 2023-09-22:
> Noch nicht ganz glücklich damit, aber schon wesentlich besser. Mich stört halt immer noch das wenn ein Inverter mal nicht erreichbar ist, immer die llast_success von allen Inverttern raus geballert werden, anstatt nur das vom betroffenen.

Sagte ja im vorhinein, dass das nur ein workaround ist und man das (auch) imo grundlegender fixen müßte, indem man das letzte Sende-Kommando inverterspezifisch prüft. Leider fehlen mir dazu die Kenntnisse (oder ich müßte mich zu tief eindenken?)...

knickohr, 2023-09-22:
Ja, man sollte es nur nich in Vergessenheit geraten lassen 🫣

knickohr, 2023-11-15:
Ja, seit der neuen 0.8.4 ist es besser, muß mir das aber nochmal in Ruhe zu Gemüte führen wenn die 0.8.x stabil bei mir läuft.

closed 2023-12-29 by lumapu