lumapu/ahoy #1126: [Bug] no inverter polling if no wifi available

technics42, 2023-08-27:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.36

### Github Hash

ba218ed

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Inverter interval set to 20 seconds

### Error description

After polling inverter is disabled due to sunset, inverter polling is not enabled any more at sunrise, if there is no wifi available.
You can easiliy check this:
If no wifi available, OLED is not turned on again (long time after sunrise). Then switch on wifi, connect to webserver of Ahoy and you can see the inverter still was not polled. Then after some seconds, the polling starts. You are going to receive values and OLED is activated.

Ollipop030, 2023-08-27:
Do you reboot yout dtu at midnight? If so, and without wifi, the time can´t be synced. And with no correct time polling inverters will not work.

technics42, 2023-08-27:
No, no reboot is done.
Dtu is running and time is synced.

MetaChuh, 2023-08-27:
@technics42 
too many issue tickets for one person at once.
5 open, 1 closed.

please select the most important issue to you and post them one by one until fulfilled at any open source project.
only freelancers around here and at other projects things get pretty cluttered by multi issues.

this does not mean, that your issues might not be relevant or existing, but it's the same as if your wife tells you 20 things at the same time and gets angry if you forgot, or did not do one of them.

technics42, 2023-08-27:
Sorry, but I think this is some kind of backlog and reporting issues, especially bug-reporting, is welcome and one part of a successful project!
Priority has to be selected by developers/project-owners, not by reporters.
But usually, bugs need to be fixed prior before implementing new stuff.

Shouldn't I report bugs any more?
So, if different people report those bugs, that would be ok, but if they are only reported by one person they shouldn't be reported?
That really sounds strange to me...
It's really hard for me to look into future (maybe not for you), but if I report an issue I don't know when I will encounter the next bug and if that bug would be more important to be fixed.

I am also not getting angry about those bugs, I just tell them - so please, don't compare me with your angry wife ;)

@lumapu I really would be interested in your opinion!

lumapu, 2023-09-30:
we did some changes here - can you check again with the latest development build?
https://fw.ahoydtu.de

technics42, 2023-10-01:
I'd like to - I am following issues on latest versions, but all seem to have some other "major issues" - especially 0.7.61...

lumapu, 2023-10-01:
sure. Next version will be published soon, then the communication should work as expected since than you can test `0.7.59`

technics42, 2023-10-07:
Version 0.7.65 is doing the job - inverter polling is started in the morning without wifi available - can be seen on display.

closed 2023-10-07 by technics42