lumapu/ahoy #1129: Weak NRF Signal

Otto-Normal, 2023-08-28:
## Hardware
  * [x] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: Wemos D1 mini
Retailer URL: ______

### nRF24L01+ Module
* [x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [x] external antenna (SMA)

### Power Stabilization:
* [x]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0.7.42
Github Hash: _______

### Build & Flash Method:
* [x] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Hallo,
super in v0.7.42 kann nun mit Anzeige der Signalstärke die Position zum Wechselrichter optimiert werden!
Klappt bei mir aber leid nicht……
Abstand zum HM 1500 circa 4 Meter.
Auch eine Änderung des Power Levels auf HIGH, oder MAX zeigt unverändert „Weak signal < 64dBm“.
Stärkeres Netzteil mit 2,5 A macht keine Änderung.

Übrigens, müsste es nicht heißen (größer) > 64dBm?

Danke!

knickohr, 2023-08-28:
Ahhh, da war jemand schneller 😉

Ja, es kommt immer „weak“, obwohl, ich nur ein paar Zentimeter durch eine Wand muß. Und es scheint wohl so, das es nur für einen Inverter gemacht wird. Was ist bei mehreren Invertern ?  Eigentlich müßte man das für jeden Inverter abfragen 🤔

Ollipop030, 2023-08-28:
> Übrigens, müsste es nicht heißen (größer) > 64dBm?

Da fehlt eigentlich das Minus Zeichen, oder irre ich mich? Aber das Selbe zeigt er bei mir auch, drei Meter Entfernung mit externer Antenne.

Karlo49, 2023-08-28:
Es muß heißen < -64dbm
0 dbm = 1mW
64 dbm = 2,51 kW  das ist schon ein starker Sender

rmayergfx, 2023-08-28:
Also ich finde die Bezeichnung etwas unglücklich gewählt.
Für das "normale" WLAN wird wifi_rssi angegeben mit korrektem? Wert -62
Warum dann beim nrf24l01+ nund die Bezeichnung in dBm und Weak Signal? 
Jeder Wert näher an der 0 ist besser, siehe:
[image]

Somit steht hier eine verwirrende Information "-" fehlt und es ist ein "Sehr gut" statt "Weak signal".
AhoyDTU wurde zum Test gerade mal 2m daneben platziert, mit externer Antenne:
[image]

rmayergfx, 2023-08-28:
@lumapu 
Habe gerade mal mit der FW GIT SHA: a515dbb :: 0.7.42 an verschiedenen Standorten im Haus getestet. Da ändert sich nichts, es wird immer "NRF Signal: Weak signal < 64dBm" angezeigt. Kann man irgendwo in der API den richtigen Wert ansehen. 
Diese Anzeige kann definitiv nicht stimmen, denn ich habe es an Positionen getestet wo ich einen Meter weiter keinen Connect mehr zum WR aufbauen kann.

knickohr, 2023-08-28:
😉

[image]

Ob die Anzeige stimmt oder nicht, steht auf einem anderen Blatt. Aber sie funktioniert im Prinzip. Und Ja, es kommt mir auch zu wenig „Strong“.  Das obige Bild stammt von einem Fusion Board, offenbar hat es eine bessere Empfangsempfindlichkeit als ein EByte Modul.

lumapu, 2023-08-28:
-75dBm heißt `weak`, -64dBm sind `strong`. Habe ich leider noch nicht sehen können, obwohl der Inverter ca. 3m entfernt (durchs Dach) ist.

[image]

knickohr, 2023-08-28:
Wo kommen die -75 her ?  Ich dachte es gibt nur die Fallungerscheidung

< -64
> -64

😲

lumapu, 2023-08-28:
ja die habe ich definiert, Vorschläge sind willkommen. Ich könnte auch -65 hinschreiben

lumapu, 2023-08-28:
@rmayergfx ja Version `0.7.43` - steckt noch in der Pipeline 😂

knickohr, 2023-08-28:
Nein, bitte keine Werte da rein interpretieren. Die Funktion liefert nur die Aussage <-64dBm oder >-64dBm, siehe bitte den PR #1050 

Alles andere ist gequirlte Sch… !

lumapu, 2023-08-28:
auf der anderen Seite haben wir aber auch ein gegebenes Konstrukt (von den HMS / HMT Wechselrichtern). Es gibt extra ein Feld RSSI für jeden Inverter, das würde ich gerne auch verwenden. Ich weiß dass absolute Werte falsch sind, in der WebUI wird es auch korrekt dargestellt:

[image]

rmayergfx, 2023-08-28:
@lumapu 
Jetzt bin ich noch verwirrter wie vorher. Woher nimmt er bei mir diesen Wert, ich kann da absolut nichts finden.
Habe auch mal auf Debug umgeschaltet um mehr angezeigt zu bekommen. Ist das vielleicht ein Wert der nur von den Hoymiles zurückgegeben wird? Mit einem TSUN sehe ich hier nichts.

knickohr, 2023-08-28:
Es gibt keinen Wert zurück, im Prinzip nur gutund schlecht. In der NRF24 Lib ist das als >-64dBm und <-64dBm definiert.

knickohr, 2023-08-28:
Ahhh sooo, jetzt verstehe ich das.

nur in der seriellen Kondole kommt das mit -75 und -64 🫣

Akzeptiert. Jetzt noch in MQTT und API 🙉🙈🙊

lumapu, 2023-08-28:
> Akzeptiert. Jetzt noch in MQTT und API 🙉🙈🙊

ok

rmayergfx, 2023-08-28:
Ich seh den Wert nicht, warum? Ab welcher FW Version wird das angezeigt?

knickohr, 2023-08-28:
Es ist eine experimentelle Funktion in der Library vom NRF. Warum liest Du den betreffenden PR #1050 dazu nicht ?

Ob es real ist, oder ob es richtige Werte sind, das kann Dir nur der Author der Library sagen. Wir versuchen das nur hier mal zu implementieren.

ich bin auch noch nicht so glücklich damit, weil bei me8nen 13 Invertern nur hin und wieder mal ein gutes Empfangssignal kommt. Da kann aber weder @lumapu  noch die Library noch Ahoy was dafür.

rmayergfx, 2023-08-28:
Mir ging es doch darum zu verstehen was da passiert, habe nun etwas weiter gegraben und es ist wohl so, das es sich um den Wert -64 dBm dreht, der irgendwo auf einem Register verfügbar ist und 0 oder 1 zurückgibt. Da ein genauer Wert hier wohl nicht ausgelesen werden kann, versteckt es bitte gerne wieder im System Menü, damit man sich nicht an einem Wert festklammert, der nicht wirklich aussagekräftig ist. Anscheinend hat der Entwickler des Chipset da nur mit WLAN "gut" und "schlecht" experimentiert. Wenn es keine wirklich nachvollziehbaren realen Werte gibt, würde ich dieses Goodie erstmal auf Eis legen.

lumapu, 2023-08-28:
im System Menü hat's nichts verloren, da es wechselrichterspezifisch ist.
Jetzt spielt mal damit rum, als nächstes müsste man sich dann die Lösung von @oberfrizte mal anschauen, da gibt es mW. 10 Stufen

beegee3, 2023-09-01:
die Lösung von @Oberfritze hat nichts mit der Signalstärke zu tun. Bei ihm geht es darum, auf welchen Kanälen vorrangig gesendet werden soll. Dazu wird eine Sende-Erfolgs-Statistik angelegt, um die Kanäle mit den meisten Antworten bevorzugt zu nutzen. Das ist sicher hilfreich, wenn man eine permanente Störquelle in der näheren Umgebung hat, entspricht aber nicht dem von den Invertern genutztem Gazell-channel-hopping, wonach der Empfänger für eine gewisse (feste) Zeit auf einem Kanal horcht bevor er auf den nächsten wechselt. Dabei werden immer alle Kanäle benutzt! Der Sender hingegen startet laut Gazell-Protokoll zunächst mit einem zufälligen Kanal und versucht die Empfänger Wechsel nachzubilden, sobald er eine Antwort erhält. Aufgrund von Asynchronität wird dieses Nachbilden aber nicht lange durchgeführt (höchstens 2 Sek), dann nimmt der Sender wieder einen zufälligen Kanal. Daher ist dieses Nachbilden für uns uninteressant. Die Zeiträume zwischen zwei Sendevorgängen würden in der Regel zur Auswahl eines zufälligen Kanals führen. Wegen meiner kann ich den Gazell-Pseudo-Algorithmus raussuchen (ist ein 5-Zeiler) und für hmRadio aufbereiten. Hilft vielleicht, wenn man mehrere Ahoy ESPs im Einsatz hat, so dass die sich weniger in die Quere kommen können.

knickohr, 2023-09-01:
Schon, er hat aber auch eine Bewertung der Kanäle durchgeführt. In der yheuristik kann man eine Art Pseudo-rssi errechnen. Ist zwar kein echtes rssi, aber immerhin.

knickohr, 2023-09-01:
#

lumapu, 2023-09-30:
gehört für mich irgendwie zu issue #718, glaube hier geht es nicht mehr weiter

DanielR92, 2024-05-19:
Die https://github.com/lumapu/ahoy/issues/718 ist doch closed, kann man dieses Issue auch closen?

closed 2024-05-30 by lumapu