lumapu/ahoy #1130: [Bug]

kiu77, 2023-08-29:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.44

### Github Hash

3e5fe74

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

all on 1 board

### Debug Serial Log output

```shell
---
```

### Error description

Gestern Abend nach Sonnenuntergang von 0,7,36 auf 0.7.44 upgedatet - klappte problemlos.
Heute morgen wollte die DTU meine Inverter nicht auslesen. Siehe:

[image]

Internetverbindung und Zeit waren okay.

kiu77, 2023-08-29:
Oha, zurück zu 0.7.36 klappte nicht wegen Fehler. Aber jetzt geht die DTU unter 0.7.44!
Merkwürdig

knickohr, 2023-08-29:
Nochmal ein oder zwei Tage und Nächte beobachten. Vielleicht nur verhaspelt. Ich empfehle Updates nur unter Tags zu machen, da sieht Mannschaft schnell ob (noch) alles funktioniert.

Ollipop030, 2023-08-29:
> Vielleicht nur verhaspelt.

Denke auch. Die 0.7.44 (gestern Nachmittag installiert) hat heute morgen beide WR bei mir direkt wieder erkannt.

kiu77, 2023-08-30:
Beim Update auf 0.7.45 - ebenfalls nachts - hat es geklappt, dass die Werte der Inverter am heutigen Morgen abgefragt werden. Insofern ist das "Problem" vermutlich obsolet.

closed 2023-12-29 by lumapu