lumapu/ahoy #1134: D1 Mini Pro does not boot

killah78, 2023-08-30:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.9

### Github Hash

15ec6a0

### Build & Flash Method

was already installed

### Setup

new device, installed 0.6.0 with nodemcu. no problems. 
then 0.6.9 via OTA -> does not boot anymore

### Debug Serial Log output

```shell
during OTA update:

Update Start: 230419_ahoy_0.6.9_15ec6a0_esp8266.bin
                                                   Update Success: 551776B
                                                                          I: Rebooting...
I: connectionEvent
connect to network 'Buchfink' ...
I: [WiFi] Connection Lost

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3424, room 16
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8
tail 0
chksum 0x2b
csum 0x2b
v00086490
@cp:0
ld
e:3
 ets Jan  8 2013,rst cause:3, boot mode:(3,6)

ets_main.c
```

### Error description

it seems same like in #931 
A new D1 Mini Pro (purchased at berrybase.de) was flashed via web installer. No booting. 
I tried another device (I purchased 2 pc), same situation. 
I flashed tasmota and easyesp WITHOUT issues. 
The I recognized issue #931 and tried with 0.6.0 via nodemcu. That worked fined. 
Then I tried updating to 0.6.9 via OTA. OTA said success. 
Debug console said:

Update Start: 230419_ahoy_0.6.9_15ec6a0_esp8266.bin
                                                   Update Success: 551776B
                                                                          I: Rebooting...
I: connectionEvent
connect to network 'Buchfink' ...
I: [WiFi] Connection Lost

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3424, room 16
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8
tail 0
chksum 0x2b
csum 0x2b
v00086490
@cp:0
ld
e:3
 ets Jan  8 2013,rst cause:3, boot mode:(3,6)

ets_main.c

And then freeze and not booting anymore. 
So question is what happend between 0.6.0 and 0.6.9? I don't think this is a broken device as i tried it with 2 and other people also have same issue. 
PS: I also tried flashing in Linux via esptool. Same result.

rejoe2, 2023-08-30:
Are you sure to open up a bug wrt. to problems with 0.6.9?

Go ahead and try to flash last official or recent dev version! You may have to activate nRF module then in settings, rest should work fine (there are also some changes in default pin assignments that may be related to just the problem you are about to report).

killah78, 2023-08-30:
That is exactly the problem. The 0.6.0 is the only/latest that works. Starting from 0.6.9 until latest 7.36 or dev doesn't work. Neither web install, nor via windows NodeMCU flasher, nor Linux esptool. With or without installed nrf. It just doesn't boot. Only 0.6.0 works fine.

rejoe2, 2023-08-30:
Did you try to erase the chip first prior to flashing? 

Afaik, there had been some reports on problems when updating without erase; not sure about that as I'm using an ESP32.
Note: default PIN assignment for ESP8266 has also changed (somewhen later as 0.7.30!) , so you may have to resolder lines just to avoid this problem (see recent discussions on discord if possible).

killah78, 2023-08-30:
Yes, tried with and without erase. 
Pin assignment you mean the nrf connection? Actually it should work even without nrf connection. The debug log should at least start with setting up the AP. This is what happens when I flashed 0.6.0. But in the debug console it just says nothing...
Or do you really mean it won't work without connected nrf?? Do you have a link for the disord channel?

rejoe2, 2023-08-30:
Yes, as (to my understanding) the IRQ PIN was to often assigned to the reset PIN. See esp. https://discord.com/channels/984173303147155506/1029761098381017098/1141612251380449280

killah78, 2023-08-31:
Ok, found out what the problem is. I needed to flash via DOUT flash mode. I think that is not the usual mode. I tried always with DIO, in the web installer you can't even select it. But with NodeMCU and DOUT and 57k baud it finally works. So I close this issue.

closed 2023-08-31 by killah78