lumapu/ahoy #1137: Settings Sunset Long lat decimal point

hpd96, 2023-09-02:
Question
Are the fields Long lat accepting persisting 
German decimal Point Comma ,

E.g.
52,38685

Ist that working to calculate sun Set?

Should ahoy convert , to '.' ?

Currently seems to convert . To ,    ( May be irritating for non Germans )

Or ist ahoy fetching Region settings fromm web Browser to usw decimal Point setting?

Or ist this simply documentation issue ?

Grüße
Best regards,
Peter

PS very fine Software Project!!

## Hardware
  * [ X] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [X ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0.7.36
Github Hash: _______

### Build & Flash Method:
* [x ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

MetaChuh, 2023-09-02:
@hpd96 

> 52,38685
Ist that working to calculate sun Set?

yes, input can be any of these decimal separators (<kbd>.</kbd> or <kbd>,</kbd>)

> Should ahoy convert , to '.' ?

yes, internally <kbd>.</kbd> is used as a decimal separator.
```
[sun] => Array
                (
                    [lat] => 48.15120
                    [lon] => 16.26398
                    [disnightcom] => 
                    [offs] => 0
                )
````

> Or ist ahoy fetching Region settings from web Browser to use decimal Point setting?

yes, afaik it's just the browser/os locale.

greetings,
metachuh

MetaChuh, 2023-09-05:
@hpd96 
do you have further questions about this request, or are you satisfied with the given information about tihis topic ?

any answer ist appreciated, and if you are satisfied, please close pending issues, as it often may cause stress, if an issue is closed by the devs due to inactivity, and the issue is not yet resolved to your needs.

many thanks for your understanding, as well as the easy wordings you used, to transport your questions.

greetings,
metachuh

closed 2023-09-05 by hpd96

hpd96, 2023-09-05:
Will Provide a Patch PR for the documentation

Great project