lumapu/ahoy #1138: [Bug] Inverter Connection bricht immer wieder ab und verbindet sich danach nicht mehr

theunknownstarman, 2023-09-03:
### Platform

ESP32

### Assembly

the DTU was already assembled

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

board

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

Unabhängig 7.35 od. 7.45

### Github Hash

201098a

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

MqTT zu HA
Alle Werte auf Standardinstallation
ESP32 WROOM

### Debug Serial Log output

```shell
09:27:24 I: (#0) prepareDevInformCmd 0x0b
09:27:24 I: TX 27 CH75 | 15 84 99 80 68 89 97 42 64 80 0b 00 64 f4 35 5c 00 00 00 00 00 00 00 00 5a 7f 8f 
09:27:24 I: (#0) nothing received
09:27:54 I: (#0) enqueCommand: 0x01
09:27:54 I: (#0) enqueCommand: 0x0b
09:27:54 I: (#0) prepareDevInformCmd 0x01
09:27:54 I: TX 27 CH3 | 15 84 99 80 68 89 97 42 64 80 01 00 64 f4 35 7a 00 00 00 00 00 00 00 00 31 c7 70 
09:27:54 I: (#0) nothing received
09:28:24 I: (#0) prepareDevInformCmd 0x0b
09:28:24 I: TX 27 CH23 | 15 84 99 80 68 89 97 42 64 80 0b 00 64 f4 35 98 00 00 00 00 00 00 00 00 9f 1d ec 
09:28:24 I: (#0) nothing received
09:28:54 I: (#0) enqueCommand: 0x01
09:28:54 I: (#0) enqueCommand: 0x0b
09:28:54 I: (#0) prepareDevInformCmd 0x01
09:28:54 I: TX 27 CH40 | 15 84 99 80 68 89 97 42 64 80 01 00 64 f4 35 b6 00 00 00 00 00 00 00 00 34 c2 bc 
09:28:54 I: (#0) nothing received
09:29:24 I: (#0) prepareDevInformCmd 0x0b
09:29:24 I: TX 27 CH61 | 15 84 99 80 68 89 97 42 64 80 0b 00 64 f4 35 d4 00 00 00 00 00 00 00 00 5c 79 07 
09:29:24 I: (#0) nothing received
```

### Error description

Ich habe zwei Inverter an der DTU konfiguriert.

Egal ob ich nur einen od. beide einrichte hält im Schnitt die Verbindung maximal einen Tag. Am nächsten Morgen funktioniert die Verbindung noch für max. eine Stunde.  Danach bricht die Verbindung zu beiden Invertern ab und verbindet sich nicht wieder.

Nach mehrmaligem Reboot (mehr als 10 mal etwa) und dann schrittweiser Einrichtung der Konfigurationen
- zuerst nur einen Inverter
- dann zweiten
- dann MqTT
klappt manchmal Verbindung und ist auch stabil bis zum nächsten Ausfall (sowohl WebUI als auch MqTT Verbindung)

Ich habe folgende Lösungsansätze ausprobiert:
- Neuflashen der DTU
- Test mit 7.35 
- Test mit 7.45
- MqTT Konfig gelöscht
- Unterschiedliche Netzteile ausprobiert
- Nur einen Inverter konfiguriert
- Konfig. Einrichtung mit Chrome anstatt Safari

Könnte es an dem ESP32 WROOM liegen?
Gibt es Erfahrungen über Stabilität?

Ich habe noch eine DTU mit 8266 unter 0.6.9 laufen. Diese ist völlig stabil.

rejoe2, 2023-09-03:
>Ich habe noch eine DTU mit 8266 unter 0.6.9 laufen. Diese ist völlig stabil.

Läuft die zeitgleich?!? Die Inverter scheinen irgendwie ihre Lausch-Zeiten auf bestimmten Kanälen mit "ihrer" DTU zu syncen. Wenn dann eine weitere DTU dazukommt, kann das funktionieren, muss aber nicht.

theunknownstarman, 2023-09-03:
Ist zwar eine gute Idee, aber die 8266 DTU ist außerhalb des Empfangsbereiches der anderen DTU. (Eine ist auf dem Dachboden, eine im Keller. 2 Stöckiges Haus).
Die DTUs laufen alle mit LOW Übertragungsleistung.

rejoe2, 2023-09-03:
Es geht nicht darum, dass sich die DTU's gegenseitig sehen, sondern ob die mit den Invertern "sprechen"... 
Schalte den ESP8266 ab und erhöhe das Abfrageintervall auf dem ESP32 (30 sec. default sind sehr wenige messages).

theunknownstarman, 2023-09-03:
Ich bin jetzt mal auf ein 10 Sekunden Intervall für die Inverter.
Mit dieser Einstellung steht die Verbindung zu den beiden am ESP32 konfigurierten Invertern innerhalb kürzester Zeit.

Ich lasse das jetzt mal ein bis zwei Tage laufen und beobachte.

Sollte das Problem noch bestehen, dann schalte ich im zweiten Schritt den 8266 ab um das Problem etwas einzugrenzen.

kiu77, 2023-09-03:
Nur als Hinweis: Ich habe an einer DTU mit ESP32 zwei Inverter mit 5s Intervall der Abfrage.
Aktiviere ich eine zweite DTU innerhalb 1 m Distanz zur ersten mit 5s Interval und einem der beiden Inverter, klappt das trotzdem zuverlässig. Muss also nicht an einer zweiten DTU liegen.

knickohr, 2023-09-03:
Wieso müßt ihr eigentlich immer Kuddelmuddel machen. Zum Test EINE DTU, nicht mehr und nicht weniger, alles andere kann, muß aber nicht funktionieren.

Ist ein Elko verbaut ?

theunknownstarman, 2023-09-04:
Ergebnis bis heute:

Die DTU lief absolut stabil bis heute Morgen um ca. 7:15 Uhr. Danach waren wieder beide Inverter nicht erreichbar. (Vor dieser Uhrzeit wurden allerdings bereits von beiden Anlagen über Solarerzeugung gemeldet)
Die Deaktivierung der zweiten DTU (auf dem Dachboden) brachte keine Besserung.

Seit 07:15 Uhr ist keinerlei Verbindung mehr zu den Invertern mehr möglich.
Die DTU wurde mehr als 10 mal rebootet. Längere Zeit stromlos gemacht.

Die WLAN Verbindung steht stabil. MqTT Übertragung (senden) zum Home Assistant ist stabil - ohne Werte - vorhanden.

lumapu, 2023-09-04:
zwei DTUs zur gleichen Zeit ist definitiv `unsupported` und kann deshalb hier nicht als `bug` geführt werden. Das ist definitiv experimentell - ich bin nicht dagegen, aber dann führen wir es als `question` oä.

theunknownstarman, 2023-09-04:
Scheinbar habe ich mich falsch ausgedrückt. Die zweite DTU ist bereits offline und das Problem ist nach wie vor vorhanden. Insofern kann ich ausschließen, dass es an einer zweiten DTU liegt (die über zwei Stockwerke mit entsprechenden Betondecken gegeneinander abgeschirmt ist).

Im Übrigen ist mein Szenario nicht so abwegig, denn es kann ja durchaus zwei Reihenhäuser nebeneinander geben, wo beide Parteien mittels einer DTU ihre Wechselrichter überwachen wollen ...

Mein nächster Plan ist ein Austausch der DTU gegen ein neues Gerät. In den nächsten ca. 2 Tagen werde ich berichten.

lumapu, 2023-09-04:
achso, ich finde deine letzte Antwort schon wieder zweideutig 😉, weiß jetzt aber sicher, dass nur eine DTU Strom hat.
Lass einfach den Satz mit den Betondecken und gegenseitiger Abschirmung weg, dann ist jedem klar wie deine Situation ist.

Natürlich ist das ein gegebenes Szenario mit den Reihenhäusern und da gehe ich auch davon aus, dass es funktioniert. Es dürfen halt jeweils nur die eigenen Wechselrichter abgefragt werden, denn es ist immer eine Peer-to-Peer Kommunikation.
Das ganze sollte sich wie WLAN verhalten, da darf auch jeder eins (oder mehrere) haben, trotzdem funktonieren sie 'nebeneinander'.

knickohr, 2023-09-04:
Du hast immer noch nicht die Frage nach dem Elko beantwortet. Oder habe ich was überlesen ?

theunknownstarman, 2023-09-04:
sorry. Habe ICH überlesen.

Bei der DTU handelt es sich um ein Fertiggerät von einem Händler. Deshalb gehe ich davon aus, dass der ELKO vorhanden ist.

lumapu, 2023-09-04:
diese Händler ... wenn du wüstest was da draußen los ist 🙄

knickohr, 2023-09-04:
Eigentlich will ich es genau genommen gar nicht wirklich wissen 😞

rmayergfx, 2023-09-05:
Eigentlich wieder mal der Klassiker. Händler verkauft AhoyDTU als Fertiggerät @lumapu sieht keinen Cent davon und soll hier im Forum noch die Probleme lösen.
@theunknownstarman 
Stell hier doch mal einen Link zu dem gekauften "Fertigmodul" ein, damit man sich davon ein Bild machen kann. Alles andere ist Glaskugel lesen.

theunknownstarman, 2023-09-05:
Das Modul wurde nicht als "Fertigmodul inkl. Software" verkauft sondern NUR die Hardware!

rmayergfx, 2023-09-05:
Naja, was anderes darf er ja auch nicht ohne mit den Lizenznbestimmungen zu kollidieren.
Hilft dir aber auch nicht weiter, denn du hast die Probleme. Daher wäre es schon wichtig was da verkauft wurde und wie es aufgebaut ist.

theunknownstarman, 2023-09-05:
Ich schließe das Thema und vermerke für mich, dass ich ein Einzelfall bin und es hier im Thread für MICH zu diesem Problem keine Hilfe gibt.

closed 2023-09-05 by theunknownstarman

rmayergfx, 2023-09-05:
Warum denn das? Es wurde nirgendwo gesagt das du keine Hilfe bekommst, aber du musst uns schon dabei helfen. Warum stellst du nicht die nötigen Informationen dazu ein?

theunknownstarman, 2023-09-05:
Bitte nichts reininterpretieren.

Ich habe weder eine Anspruchshaltung auf Hilfe, noch bin ich beleidigt. 

Mir ging es in meinem Ticket alleinig um eine mögliche Lösung für mein Problem bzw. die Hoffnung, dass ich nicht der einzige bin, der so eine Fehlersituation hat. Offensichtlich bin ich mit diesem Problem eher ein Einzelfall oder sogar der einzige.

Die Tips, die hier formuliert wurden, habe ich alle ausprobiert.  

Zu der DTU bzw. dessen Inhalt (ob mit oder ohne Elko) kann ich nichts sagen, da das Gerät verschlossen und versiegelt ist. Da auf dem Gerät noch Garantie ist tausche ich es aus, um zu sehen ob es evtl. ein hardwarebedingter Fehler vorliegt. Bilder über dessen Inhalt gibt es nicht. Dies ist auch nicht ungewöhnlich, da Händler selten ihre Geräte in offenem Zustand fotographieren.

An einem Handlerbashing werde ich mich nicht beteiligen, da dies nicht der Lösungsfindung dient.

rmayergfx, 2023-09-05:
Handlerbashing, was soll das denn nun schon wieder? Es geht einzig und alleine darum zu wissen was da verbaut wurde, nicht mehr und nicht weniger. Zur Info, Espressif ESP-WROOM-32 findet man auf einigen Boards, leider gibt es auch da einiges an Spreu. Daher wäre es immer sinnvoll zu wissen was im Endeffekt da verbaut/gekauft wurde. Wenn das Gerät verschlossen und versiegelt ist und man nicht reinsehen kann, sollte es zumindest in der Doku stehen.

knickohr, 2023-09-05:
Ein ähnliches Problem besteht wohl auch mit dem ePaper Display. Zumindest erstmals bei mir. Ein Elko rein und gut ist. Ob es jetzt noch besser ist, da einen Widerstand rein zu machen wäre ein Versuch Wert.

Aber was wir hier mittlerweile alle gelernt haben :  Es geht nicht ohne Elko !  Nicht wenn man Lanzeitstabilität und Funktionssicherheit gewährleisten will. Das sollten sic/ alle PCB-Entwickler und DTU-Bauer hinter die Ohren schreiben !  Umso wichtiger ist es auch, das wir keine Schaltpläne (Fritzings) mehr raus geben ohne Elko (und Widerstand).

theunknownstarman, 2023-09-08:
Finaler Endstand:

Die DTU wurde gegen eine Neue ausgetauscht und seither ist kein Ausfall der Inverter Verbindungen mehr feststellbar.
Die zweite DTU ist ebenfalls wieder in Betrieb und es sind keine Störungen feststellbar.