lumapu/ahoy #1140: [Bug] DTU bootet in Endlosschleife wenn alle Inverter disabled sind

knickohr, 2023-09-05:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.46

### Github Hash

Latent DEV

### Build & Flash Method

ESP Tools (flash)

### Setup

MQTT & ePaper

### Error description

Gestern Abend noch die letzte DEV aufgespielt und heute Morgen mit Erschrecken festgestellt das sich die DTU in einer Endlos-Rebootschleife befindet. Die Inverter sind momentan deaktiviert, da ich auf einer anderen DTU einen Test mit der Firmware von Oberfritze mache.

DTU in den AP gezogen, so das sie keine Zeit bekommt, danach konnte ich auf die 45 downgraden. Offenbar gibt es einen Fehler, sobald die Inverter nach gültiger Zeit anfangen zu arbeiten sollen,  aber eben kein Inverter aktiviert ist. Sobald mindestens einer aktiviert funktioniert alles wieder.

MetaChuh, 2023-09-07:
hi @knickohr 
ich hatte ähnliche beobachtungen mit nur 4 invertern.

was ich bisher an reproduzierbaren auslösern gefunden hab, ist:
wenn ich die api poll abfrage auf die dtu ip während der reboot phase (wenn die dtu nicht erreichbar ist) ausschalte, dann rebootet sie wieder normal, und danach (spätestens wenn alle inverterdaten wieder eingelesen sind) kann ich die dtu auch wieder in 1-5s brutal stakkato intervallen abfragen, ohne dass sie in den beobachteten boot loop zustand geht.

ob ein mqtt setup, welches intern von der dtu im angegebenen intervall durchgeführt wird, das gleiche produzieren kann weiß ich leider noch nicht.

note:
zwar bei mir ein anderes setup, wegen kleinerer inverteranzahl und bare metal api steuerung, aber vielleicht findest du durch die info was, was uns alle evtl auf die ursache hindeuten kann 👍

thx & greetings

knickohr, 2023-10-01:
Kann man zu machen, habe es eben nochmal getestet, funzt !

closed 2023-10-01 by knickohr