lumapu/ahoy #1141: [Bug] Nach flashen des Fusionsboard V1 oder ESP S3 geht nix mehr weder Ota noch VSC

teletapi, 2023-09-05:
### Platform

ESP32

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.46

### Github Hash

6b721cb

### Build & Flash Method

VSCode - Platform IO (build & flash)

### Setup

Fusionsboad V1 mit Epaper und ESP32 S3 devkit mit SSH1106

Beim ESP32 funktioniert alles

### Debug Serial Log output

```shell
Kein log möglich, zeigt nix an
```

### Error description

Nach flashen des ESP32 S3 geht garnix mehr, beim Fusionsboard ist es ähnlich nach mehrfachen abstecken ist das Board zwar wieder erreichbar jedoch endlos langsam weder ein reboot funktioniert noch andere änderungen werden übernommen der

rmayergfx, 2023-09-05:
Probiere doch mal mit den https://esphome.github.io/esp-web-tools/ ob damit das Board wieder funktioniert.
Danach kannst du die passende Firmware auch per OTA einspielen. Wenn es mit ESPHome auch nicht mehr geht, hat vllt. das Board einen Bug/Defekt.

rmayergfx, 2023-09-05:
Das ging aus deiner Fehlermeldung nicht hervor. Bitte in Zukunft eindeutig dazuschreiben wenn es mit Version 7.45 noch geht und mit 7.46 nicht mehr und welche Fehler genau auftreten! 
Mit "endlos langsam" kann man nichts anfangen, bitte mal ins Log schauen und dementsprechend hier posten!

lumapu, 2023-09-05:
kann das auch jemand anderes bestätigen? Ja gestern war dort eine Änderung, siehe issue #1106

DGK110, 2023-09-05:
Kann das leider auch bestätigen.

Meine Fusion DTU ist nach Update auf 0.7.4.6 tod!

closed 2023-09-06 by teletapi

knickohr, 2023-09-07:
https://discord.com/channels/984173303147155506/1029761098381017098/1149113574980071585