lumapu/ahoy #1142: [Bug] Mi-1200 - No Channels but total solar feed

TeckyTec, 2023-09-05:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.36

### Github Hash

ba218ed

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

-

### Error description

Mi-1200 - no channels but total solar feed
[image]
[image]

lumapu, 2023-09-05:
@rejoe2 is correct, `maximum module power` is missing in your settings. No bug 😉

TeckyTec, 2023-09-06:
@lumapu @rejoe2 
Cant find it - no box to configure ....

[image]

if i want to add one more inverter or readd the original - i dont have these options

[image]

rejoe2, 2023-09-06:
Strange...
Might be a problem in v0.5.36? At least in (my recent version of) v0.5.45, these option shows up as soon as a valid inverter nr is entered (for 4 panels,  starting with1061, 1062 and 1161 work.

As a workaround, you could use the JSON file option and do the corrections there.

@lumapu: 
a) In general, as soon as there's more than one converter, changing the settings for only the last inverter is possible. afair, this has been different in the past. Is there an option to get back to the earlier behaviour?

b) As this is one of the common user problems with the default "0": Wouldn't it be a good idea to change the default to e.g. 300 or 350?

@TeckyTec :
ATM I'm doing some testing with quality parameters, see https://github.com/lumapu/ahoy/compare/main...rejoe2:ahoy:nRF-radio-quality-array. As you are one of the few owners of a 4-channel -MI (2nd gen.), I'd kindly ask you to do some testing as well wrt. to this. So if you are able to compile the firmware, just clone my repo, otherwise I also could provide a binary for ESP8266 as well. There's also some discussion on that on discord in case you want to join.

TeckyTec, 2023-09-06:
@rejoe2 
Would I have the possibility to perform a full reset? possibly then the max power fields come up

Okay how and where can I configure this JSON?

Of course we can run tests and I am happy to install test firmwares

I find the project very exciting and will join the Discord

BTW I have the following 4x modules installed:
Yingli YL250P-29b

ELECTRICAL BEHAVIOR
module type Yingli YL250P-29b
power Pmax W: 250
Power tolerance ΔPmax W: 0 / + 5
Module efficiency ηm %: 15.3
Voltage at Pmax Vmpp V: 29.8
Current at Pmax Impp A: 8,39
Open circuit voltage Voc V: 37,6
Short circuit current Isc A: 8,92

MATERIALS AND COMPONENTS
Front cover (material / thickness): Low iron annealed glass / 3.2mm
Back cover film (color): White
Cell Type (Quantity / Technology / Dimensions): 60 / Multicrystalline Silicon / 156mm x 156mm
Cell Embedding (Material): Ethylene vinyl acetate (EVA)
Frame (material / color / anodization color): Anodized aluminum alloy / Silver / Clear
Junction box (protection class): ≥ IP65
Cable (length / conductor cross-section ): 1100mm / 4mm2
Connector (type / protection class): MC4 / IP67
Dimensions (length / width / thickness): 1650mm / 990mm / 40mm
Weight: 18,5kg

rejoe2, 2023-09-06:
So the discord channel should be this one: https://discord.com/channels/984173303147155506/1144045244166443159

As soon as you install a newer version (there's also .45 available via github actions (avoid .46 for the moment!)), you should get back the option to edit at least the last inverter panel data. JSON is available via "Settings" menue item (last section, Export/Import, but would require to add the WiFi password to the JSON as well).

TeckyTec, 2023-09-06:
Do you mean 0.7.**45**? Where can I find this?

[image]

rejoe2, 2023-09-06:
Yes, 0.7.45 was meant. That's (almost) the lastest version in the development03-tree. You have to look for "actions" there:
[image]

lumapu, 2023-09-07:
or easier at https://fw.ahoydtu.de

@rejoe2 I didn't know that only the last inverter can be edited - I will check this. Is there already a seperate issue for that?

rejoe2, 2023-09-07:
> or easier at https://fw.ahoydtu.de
> 
> @rejoe2 I didn't know that only the last inverter can be edited - I will check this. Is there already a seperate issue for that?

There's no issue for that. Until you asked, I was not even sure this was not the intented behaviour...

lumapu, 2023-09-07:
fixed in `0.7.47`

rejoe2, 2023-09-08:
Thanks for fixing that!

Btw: I like the new "System" view! 

@TeckyTec: Did you find some time for testing on https://github.com/rejoe2/ahoy/tree/nRF-radio-quality-array?

TeckyTec, 2023-09-11:
@rejoe2 @lumapu 
thx 0.7.47 has fix the problem

closed 2023-09-11 by TeckyTec