lumapu/ahoy #1144: TSOL-M800 G2 -- Inverter nicht erreichbar

Loewe1991, 2023-09-08:
## Hardware
  * [ X] ESP8266
  * [ X] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ X] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [X] circuit board
* [ X] external antenna (SMA)

### Power Stabilization:
* [X ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0.7.36
Github Hash: _______

### Build & Flash Method:
* [X ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ X] USB Serial Log (attached)
* [ X] Setup settings (use our templates ... to be added)

Servus in die Runde!
Habe ein Problem was ich leider nicht alleine gelöst bekomme.
Wie im Titel schon geschrieben handelt es sich um einen TSOL-M800 Inverter bei dem die SerienNr. mit "11" beginnt was ja Gen2 entsprechen sollte und somit kompatibel sein müsste.

Ich habe die Ahoy-DTU auf basis von 2 unterschiedlichen Kontrollern aufgebaut (ESP32 und 8266) bei der Antenne habe ich eine mit und eine ohne den Kondensator.
Bei dem 8266 habe ich auch die PIN-Belegung in verschiedenen Formen und Varianten probiert. Leider alles ohne Erfolg.

Bei einem Hoymiles funktionieren beide Aufbauten ohne Probleme, nur bei der M800 WR leider nicht.

Hat noch jemand eine Idee wie ich das Problem lösen könnte ?

`
12:28:25 I: (#0) nothing received
12:28:35 I: (#0) enqueued cmd failed/timeout
12:28:35 I: (#0) resetPayload
12:28:35 I: (#0) Requesting Inv SN 11418XXXXXXX
12:28:35 I: (#0) prepareDevInformCmd 0x0b
12:28:35 I: TX 27B Ch61 | 15 82 03 78 41 84 29 35 65 80 0b 00 64 fa f7 53 00 00 00 00 00 00 00 00 b4 ae fb
12:28:35 I: (#0) nothing received
12:28:45 I: (#0) enqueued cmd failed/timeout
12:28:45 I: (#0) resetPayload
12:28:45 I: (#0) Requesting Inv SN 11418XXXXXXX
12:28:45 I: (#0) enqueCommand: 0x01
12:28:45 I: (#0) enqueCommand: 0x0b
12:28:45 I: (#0) prepareDevInformCmd 0x01
12:28:45 I: TX 27B Ch75 | 15 82 03 78 41 84 29 35 65 80 01 00 64 fa f7 5d 00 00 00 00 00 00 00 00 de e8 d3
12:28:45 I: (#0) nothing received
12:28:55 I: (#0) enqueued cmd failed/timeout
12:28:55 I: (#0) resetPayload
12:28:55 I: (#0) Requesting Inv SN 11418XXXXXXX
12:28:55 I: (#0) prepareDevInformCmd 0x0b
12:28:55 I: TX 27B Ch3 | 15 82 03 78 41 84 29 35 65 80 0b 00 64 fa f7 67 00 00 00 00 00 00 00 00 75 c8 68
12:28:55 I: (#0) nothing received
12:29:05 I: (#0) enqueued cmd failed/timeout
12:29:05 I: (#0) resetPayload
12:29:05 I: (#0) Requesting Inv SN 11418XXXXXXX
12:29:05 I: (#0) enqueCommand: 0x01
12:29:05 I: (#0) enqueCommand: 0x0b
12:29:05 I: (#0) prepareDevInformCmd 0x01
12:29:05 I: TX 27B Ch23 | 15 82 03 78 41 84 29 35 65 80 01 00 64 fa f7 71 00 00 00 00 00 00 00 00 1f 24 f2
12:29:05 I: (#0) nothing received
12:29:15 I: (#0) enqueued cmd failed/timeout
12:29:15 I: (#0) resetPayload
12:29:15 I: (#0) Requesting Inv SN 11418XXXXXXX
12:29:15 I: (#0) prepareDevInformCmd 0x0b
12:29:15 I: TX 27B Ch40 | 15 82 03 78 41 84 29 35 65 80 0b 00 64 fa f7 7b 00 00 00 00 00 00 00 00 b5 50 2c
12:29:15 I: (#0) nothing received
12:29:25 I: (#0) enqueued cmd failed/timeout
12:29:25 I: (#0) resetPayload
12:29:25 I: (#0) Requesting Inv SN 11418XXXXXXX
12:29:25 I: (#0) enqueCommand: 0x01
12:29:25 I: (#0) enqueCommand: 0x0b
12:29:25 I: (#0) prepareDevInformCmd 0x01
12:29:25 I: TX 27B Ch61 | 15 82 03 78 41 84 29 35 65 80 01 00 64 fa f7 85 00 00 00 00 00 00 00 00 db 12 f4
12:29:25 I: (#0) nothing received
12:29:35 I: (#0) enqueued cmd failed/timeout
12:29:35 I: (#0) resetPayload
12:29:35 I: (#0) Requesting Inv SN 11418XXXXXXX
12:29:35 I: (#0) prepareDevInformCmd 0x0b
12:29:35 I: TX 27B Ch75 | 15 82 03 78 41 84 29 35 65 80 0b 00 64 fa f7 8f 00 00 00 00 00 00 00 00 71 66 2a
12:29:35 I: (#0) nothing received
`

beegee3, 2023-09-08:
Da sind mehrere Dinge, die nicht zusammenpassen: Inverter, Seriennummer und Generation.
Laut https://github.com/lumapu/ahoy/blob/main/doc/hoymiles-format-description.md ist der TSOL-M800 ein Inverter der 2.Gen und hat eine Seriennummer mit 1041 beginnend.
Wenn deine Seriennummer mit 1141 beginnt, wäre es 3.Gen.
Wenn es ein 1.Gen Inverter ist, müsste die Seriennummer mit 1040 beginnen (und würde aktuell nicht von Ahoy unterstützt).
Bitte prüfe nochmal die Seriennummer.
Wird denn angezeigt (Webanzeige -> System), dass der NRF24L01 'connected' ist? Wenn nicht musst du die Konfiguration des NRF24L01 in den Settings anpassen.

Loewe1991, 2023-09-08:
Hallo beegee3,
danke für die Rückmeldung. Da habe ich wohl falsche Informationen irgendwo aufgeschnappt. 
Dachte der Evo wäre die 3. Gen und hat diese R-Seriennummer, aber nun gut. WIeder etwas gelernt.

NRF24L01+ hat den Status "is connected" Datarate liegt bei "250 kbps" und Power Level "low". 

Die Seriennummer habe ich jetzt mehrmals gegengeprüft und die fängt "1141" an.

beegee3, 2023-09-09:
Hallo Loewe1991,
interessant, dann ist es 3.Gen und dein Inverter entspricht wohl einem HM-800. Soweit also gut!
Du benutzt die Version 0.7.36. Die hat einen Bug in der Default Belegung des NRF24L01+, s. #1132.
Wenn deine Verkabelung Standard ist, solltest du also die Einstellung CE auf GPIO4 ausprobieren.

Loewe1991, 2023-09-09:
Grüße,
hat leider auch nichts gebracht.
Ich werde langsam das Gefühl nicht los dass womöglich das Funkmodul im Inverter defekt ist.
Gibt es irgendwie eine Möglichkeit zu prüfen ob Modul funktioniert ?

beegee3, 2023-09-11:
🤔 Funkmodul im Inverter defekt? Glaube ich eigentlich nicht. Das lässt sich ggfs. wohl nur mit Original DTU und App checken.
Prüf doch nochmal die ganze Ahoy Verkabelung. Dazu ist hier im Forum schon einiges geschrieben worden (was funktioniert hat, was nicht; z.B Kabel gelötet ist sicher besser als gesteckt). Und nochmals die Pin Belegungen in der Konfiguration gegenchecken. Vielleicht solltest du auch mal andere Ahoy Versionen probieren (ältere oder auch die aktuelle dev.)
Sorry, wenn ich da auch nur in der Luft stochere, aber etwas anderes fällt mir aus der Ferne nicht ein. Im Prinzip bräuchtest du jemanden, der ein laufendes Ahoy hat, wo du deine ESP mit seinem Inverter auf Funktionsfähigkeit prüfen könntest und umgekehrt deinen Inverter mit seinem Ahoy.

rmayergfx, 2023-09-13:
Auch wenn es banal klingt, der TSOL M800 ist wirklich auch ein M800 und kein MS800? Wie alt ist das Gerät?
Wenn der Aufbau mit einem Hoymiles einwandfrei funktioniert müsstest du nur die Seriennummer ändern und dann sollte die Kommunikation spätestens nach einem Reboot funktionieren.
Seriennummer hat nicht auch noch ein "R" vorangestellt?
Welche Farbe hat die LED am WR?

Loewe1991, 2023-09-14:
Guten Morgen,
auch wenn die Frage banal ist, hat sie ihre Berechtigung.
Ich habe ein Bild vor dem Einbau des WRs gemacht von der Rückseite .
Der WR ist so ziemlich genau 1 Jahr in Betrieb (mal bei uns). Ich glaube aber zu wissen das mein alter Herr den WR damals auf Kleinanzeigen geschossen hat. Dazu müsste ich ihn aber heute Abend nochmal befragen. 
https://ibb.co/j5xJPL9

Die LED Farbe kann ich leider nicht mehr sehen, da der WR verbaut wurde. 

Wenn ich in den Einstellung der Ahoy-DTU meine Seriennummer, von meinem Hoymiles eintrage funktioniert es auf anhieb. Dazu ist nicht mal ein Reboot nötig. Aber auch reboots haben nichts gebracht.

Ich würde daher eine falsche Einstellungen / Verkabelung / Verdrahtung whatever eigentlich zu 100% ausschließen.
Daher kam mir ja schon der Verdacht auf ein defektes Funkmodul um WR selbst.

Habe mal meinen NetAlly G2 AirCheck ausgepackt und geschaut ob das Ding evtl. irgendwas ausstrahlt was ich damit messen kann (Bluetooth und WLAN ggf. "Störer"). Aber auch da wäre mir nichts aufgefallen.

rejoe2, 2023-09-21:
Wenn das Ding doch schon älter ist, ist es _vielleicht_ doch ein 2nd gen. Gerät. Du könntest testen, ob der WR antwortet, wenn du die Seriennummer "rückwärts" änderst, also auf 1041anpaßt.

Falls es doch 3rd gen ist, wäre das nicht weiter dramatisch - der MI-Code berücksichtig das und sollte dann "das ist doch ein neueres Gerät!" melden und ab da 3rd gen Kommandos aussenden (das klappte zumindest mit meinen 3rd gen MI-1500).

lumapu, 2023-09-30:
gibt es Neugikeiten?

Loewe1991, 2023-10-08:
Hallo, sorry für die späte Rückmeldung.
Leider brachten auch die Ratschläge von @rejoe2 nichts.

lumapu, 2023-10-09:
dann ist der Inverter scheinbar zu alt. Ist der issue auch für andere relevant?

stefan123t, 2024-10-29:
@Loewe1991 funktioniert es mit Deiner AhoyDTU in der Zwischenzeit bzw. hast Du es mit der OpenDTU Software mal versucht ?

Laut einer kurzen Suche wird die `1141` bei OpenDTU unterstützt, mein HM-600 hat die selbe Modellnummer.
https://github.com/search?q=repo%3Atbnobody%2FOpenDTU%201141&type=code