lumapu/ahoy #1146: [Bug] Homeassistant Discovery entries not marked as RETAIN leds to occasional "Enitity not provided"

bentolor, 2023-09-09:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.36

### Github Hash

ba218ed

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Added an MQTT target (mosqitto)
Initially created/triggered HASS discovery using v0.6.9

### Error description

Thanks for this awesome piece of software!

1. Yesterday I installed ahoyDTU and successfully integrated everything also into MQTT/HASS using v0.6.9. Afterwards I also upgraded to the latest version (which dropped my RF config, btw.)
2. Today, suddenly HASS reported "Entity no longer provided by integration MQTT" for all ahoyDTU entities
3. I inspected the situation using `mqttui`: Obviously all data updates at `inverter/…` still get published, but the autodiscovery entries (`homeassistant/sensor/topicname/ch0_…`) have gone missing. 
4. Interestingly some autodiscovery leftover entries made by v0.6.9 still seemed to there. I deleted them manually using `mqttui`
5. After manually retriggering `[Discovery Config (homeassistant) > Send]` everything seems to be working again.

I can remember some (to me obscure) option to _"delete homeassistant on shutdown"_ entries which where present in the 0.6.9 ui, but no longer. As of today, I suspect, unchecking them send RETAIN mqtt messages for the autoconfig entries, while checking thi soption just sent "normal" mqtt messages.

My interpretation of the situation: looking at the (re-)created autodiscovery entries created by 0.7.36 I see that none of the values are marked as RETAIN, while they are for all other MQTT HASS autodiscovery entries (i.e. ESPHome, Valetudo, Xiaomi DaFang Hacks).

Obviously, after some time the non-RETAIN homeassistant autodiscovery mqtt messages ahoyDTU created got lost this morning, causing home assistant mqtt integration to assume the entity to be no longer available.

I see two options to avoid the erratic breakage of the HASS integration:

1. Mark the mqtt config entries as RETAIN. Of course, this requires manual deleting them in case you i.e. renamed your devices OR
2. ~~Republish them automatically. But honestly I don't know how often this needs to be done, as you have no guarantee at all that they are visible when home assistant needs them.~~

I just [looked at the home assistant documentation](https://www.home-assistant.io/integrations/mqtt/#mqtt-discovery), and It's quite obvious that the RETAIN flag is missing for correct operation:

> Retain: The -r switch is added to retain the configuration topic in the broker. Without this, the sensor will not be available after Home Assistant restarts.

Am I right that the behaviour changed here recently?

lumapu, 2023-09-16:
will fix it

lumapu, 2023-09-16:
checked the code. As you can see in the code-snippets below the retained flag is on for homeassistant. These snippets are from version `0.7.36` (last release)

autodiscovery publish:
https://github.com/lumapu/ahoy/blob/ba218edbdb1b0a168e0c721bc2259fcc97c57f8a/src/publisher/pubMqtt.h#L417

publish-function-prototype:
https://github.com/lumapu/ahoy/blob/ba218edbdb1b0a168e0c721bc2259fcc97c57f8a/src/publisher/pubMqtt.h#L178

bentolor, 2023-09-18:
@lumapu Thanks for the reference. I just retested the situation with `0.7.36` by deleting the entries and recreating them. Not they correctly appear to be RETAIN values. 

So my best guess is, that this was some effect of pre-`0.7.36` in combination with update.

I'm not sure on @darkcheater situation, but from my side this ticket can be closed.

darkcheater, 2023-09-20:
I just redid the HA discovery and now I have more values and the names are better. Forgot to redo the discovery after changing the config.
And I think my problem comes from somewhere else because my controller seems to crash after some time. I think this can be closed. Thanks!

closed 2023-09-20 by bentolor