lumapu/ahoy #1147: [Bug]: Energy overflow at 166755 Wh - Ahoy or Hoymiles bug?

solarpirats, 2023-09-09:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.36

### Github Hash

ba218ed

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

{"wifi":{"ssid":"","pwd":"","ap_pwd":"esp_8266","hidd":false,"dev":"AHOY-DTU","adm":"","prot_mask":61,"dark":false,"reb":true,"ip":"","mask":"","dns1":"","dns2":"","gtwy":""},"nrf":{"intvl":1,"maxRetry":5,"cs":15,"ce":0,"irq":2,"sclk":0,"mosi":0,"miso":0,"pwr":1,"en":true},"ntp":{"addr":"pool.ntp.org","port":123,"intvl":720},"sun":{"lat":0,"lon":0,"dis":false,"offs":0},"serial":{"intvl":5,"show":false,"debug":false},"mqtt":{"broker":"","port":1883,"user":"","pwd":"","topic":"inverter","intvl":0},"led":{"0":255,"1":255,"act_high":false},"plugin":{"disp":{"type":0,"pwrSafe":false,"pxShift":true,"rotation":0,"contrast":60,"data":255,"clock":255,"cs":255,"reset":255,"busy":255,"dc":255}},"inst":{"en":false,"rstMidNight":true,"rstNotAvail":false,"rstComStop":false,"strtWthtTime":false,"yldEff":1,"iv":[{"en":true,"name":"Batterie","sn":000,"yield":[0,0,0,0,0,0],"pwr":[300,0,0,0,0,0],"chName":["Batterie","","","","",""]}]}}

### Debug Serial Log output

```shell
Timestamp	AC voltage	AC current	Power	Total Energy	Daily yield	DC voltage	DC current	DC Power	Temperature
1694189061	223.5	1.36	304.8	166755	3182.0	49.6	6.46	319.1	40.9
1694189074	224.1	1.36	305.4	101220	-62353.0	49.5	6.48	319.8	40.9
```

### Error description

The total yield jumps from 166755 to 101220. Ahoy read bug or Hoymiles overflow? 
Log see above.

Solarteur, 2023-09-09:
Is more than one DTU (Hoymiles DTU, ahoyDTU, OpenDTU) operating at the same time? 

If this is the case, this is “normal” behavior and has been described as such many times.

solarpirats, 2023-09-09:
No, only one DTU and one inverter. --Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

Solarteur, 2023-09-09:
Ok. 

Was the display correct again after timestamp 1694189074?
DC voltage seems pretty high to me at > 49 V. Do you use a battery?

solarpirats, 2023-09-09:
Yes, I use a battery. The display is still the same, Screenshot attached. The daily yield was something about 62000 Wh, maybe this caused the problem. I wonder, why the yield is not read correctly, so I think this can also be an error from hoymiles. --Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

scharrrfi, 2023-10-05:
I have the same issue by using on a battery. Daily yield is reset on 65500 Wh to 0 and the Total yield is also reduce over 65500 Wh. For statistics this behavior is very bad.

Is this a hoymiles issue or an ahoy issue?

solarpirats, 2023-10-05:
I guess this is a hoymiles issue. Total yield is calculated as the sum of eprom value and RAM daily value. Yield is only saved permanent if DC input drops below certain value. If there is a yield overflow (16bit), total yield is reset to eprom value. Only war around is switch off dc before 65767 is achievrd. --Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

scharrrfi, 2023-10-05:
Is there any option for save the Ram and clear it without disconnect DC?

solarpirats, 2023-10-05:
You can try to trigger permanent storage with AC disconnect. --Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

scharrrfi, 2023-10-05:
It seems that there are no permanent storage on AC disconnect. Any other solution? Is it possible to trigger the storage with some DTU command?
Alternatively I have to make the yield statistics with an smart socket instead of the internal inverter data.

lumapu, 2023-10-05:
The overflow of yield day at 65535 is related to the storage register depth which Hoymiles spent (16bits).
The only choice we have is to trigger an inverter reboot. There is a command for this in our documentation as well as a button in the webUI.

scharrrfi, 2023-10-06:
Is it possible that the UI button for restart does not work? When I press it the the yield day does not reset (or the inverter does not a reboot). 
Turn off and turn on works well and are daily used while charging the battery. I am in the latest stable version.

stefan123t, 2024-10-29:
@solarpirats, @scharrrfi the same issue has been logged on OpenDTU / OpenDTU-OnBattery (ODOB) issue tracker too.
See https://github.com/tbnobody/OpenDTU/issues/1552 and https://github.com/tbnobody/OpenDTU/issues/2103 but also e.g. https://github.com/hoylabs/OpenDTU-OnBattery/issues/181

It was estimated that the counter will rollover / overflow after about 32 days with e.g. a HMS-2000 (65kWh/2kWh/d = 33d).

@lumapu I recall that you simply have to subtract the `_previous` uint16 value of the inverter from the `_current` uint16 value of the inverter to get a uint16 `_difference` with the rollover ignored by uint16 specific maths / bit-operations.
You then add the uint16 `_difference` to the actual uint32 `daily_yield` value and have a proper value in your chosen precision, without any rollover issues.

```c++
uint16 _previous, _current, _difference;
unit32 daily_yield;

_difference = _previous - _current;
daily_yield += _current
_current = _previous;
```