lumapu/ahoy #1149: AhoyDTU Hoymiles HM800 - Wrong data rate?

lunanigra, 2023-09-11:
The communication between AhoyDTU and Hoymiles inverter is not stable. Sometimes everything ok for days, then not connection at all (RX no answer).

Status of NRF24L01+ is shown as connected. But data rate seems being not correct - I assume data rate should be 250 kbps.

Is there a way to set the data rate hard-coded?

## Hardware
* [x] ESP32

### nRF24L01+ Module
* [x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode

### Antenna:
* [x] circuit board
* [x] external antenna (SMA)

### Power Stabilization:
* [x]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module

[image]

kiu77, 2023-09-11:
In such strange cases a somehow notsological procedure helped me:
Just reflash the ESP with erasing the flash and then update to a newer version of the firmware.
Then make the settings new.

I had behaviour in some ESP32 which could not fixed otherwise.

lunanigra, 2023-09-11:
@kiu77, thanks for your quick response. I will try reset and/or erasing the flash.

Just for understanding... Is the data rate an indicator for possible errors? Or is it not such easy?

kiu77, 2023-09-11:
At least the displayed data range sounds very strange to me.
BTW: just resetting the settings didn't work in my cases. I had to flash with erasing.

lumapu, 2023-09-12:
so you could fix it by reflashing with erase?

lunanigra, 2023-09-12:
Sorry, not so far... Inverter is located at my parent's house. Will check on weekend (at the latest).

lunanigra, 2023-09-15:
@kiu77, @lumapu, I erased ESP32 and re-flashed AhoyDTU. Data rate is shown as 256 kbps now - unluckily this is the only change.

Still only "RX no answer". I'm wondering what could be wrong?
- nRF24L01+ is connected (tried with three modules / with and without ext. antenna)
- I verified wiring multiple times (breadboard)
- I also verified serial number of inverter
- Inverter works (green LED)
- distance between inverter and ESP32/nRF24L01+ approx. 2 m

Any idea what else to check?

lumapu, 2023-09-15:
the pinout, we changed the default pinout recently

lunanigra, 2023-09-16:
Ok, will check wiring again. But must be at least nearly correct as nRF24L01+ is shown as connected.

Following the [v0.7.36](https://github.com/lumapu/ahoy/releases/tag/ahoy_v0.7.36) release notes pin assignment was changed for ESP8266 only.

fsck-block, 2023-09-16:
Hi @lunanigra 
had the same problem some month ago. 
After changing the NRF24l01+ Modul all works well again.

kiu77, 2023-09-16:
Good aspect, @fsck-block ,
I had one nrf24 module from eByte (about of 20) with defect and an ESP32 module (out of 20) which give little smoke rings...
A defect quote of 5% is not small.

kiu77, 2023-09-16:
@lunanigra 
Did you check if in setup -> system config CE is set to GPIO4?

lunanigra, 2023-09-16:
After changing the pin assignment of CE/CS the data rate is shown with 2 mbps again.
I tried three different nRF24L01+ modules. Of course no idea if they are working; but at least one was newly ordered.
Overall no change...

[image]
[image]

lumapu, 2023-09-16:
I don't understand why the datarate goes up to 2Mbps

fsck-block, 2023-09-17:
> I don't understand why the datarate goes up to 2Mbps
After facing this issue I did some investigation.
It happen's in method `HMRadio::sendPacket`. 
After calling `mNrf24.stopListening()` the datarate was changed from 250kbps to 2Mbps. 
Did also a check of the library function but no obvious bug.  
Tried to set the datarate back to 250kbp at this point but without success.

lumapu, 2023-09-17:
@fsck-block thank you for pointing out that setting the data rate back does not fix the issue. This was my next attempt but now I don't have to test this.

lumapu, 2023-09-30:
any updates here? @lunanigra

lunanigra, 2023-10-01:
@lumapu, same as before - data rate goes to 2 mbps and "RX no answer".

Besides this I don't see anything wrong. Maybe it is a hardware issue. But quite strange as there are no clear other errors.

Any experiences how important the elko is? I tried with and without... Just wondering if needed at all?

kiu77, 2023-10-01:
@lunanigra 
I think your theory is right: a hardware defect.
The Elko can't be responsible for changing the data rate.

lunanigra, 2023-10-02:
As I tried three nRF24L01+ modules so far wondering what kind of hardware defect it might be?

Something on ESP32 itself or the wiring? But as the nRF24L01+ is shown as connected it must be small or quite strange defect...

lumapu, 2023-10-02:
can't imagine that 3 NRF modules should be defect.
Do you provide 3.3V to the modules (not 5V)?
Have you tried another ESP as well?
Can you check that the chip select works properly?

kiu77, 2023-10-02:
Can you show a photo of your setup?

lunanigra, 2023-10-07:
@kiu77, the nRF24L01+ is connected to 3.3V. Also don't see anything wrong on CS.

But will set up everything from scratch the next days. Perhaps there really something with the breadboard wiring.

Btw., is the elko really useful? Or in which cases it is needed... When using bad power supply?

kiu77, 2023-10-07:
Some say so. I have it on my PCB.
All I can say is that my system works. And that I hat 2 dead or problematic ESP32 and 2 nRF24+ until now (out of over 20 pieces of each).

lunanigra, 2023-10-15:
After putting the used components on PCB it looks stable. Surprisingly, as I never had issues on breadboard before.
However, thanks for support :)

closed 2023-10-15 by lunanigra

kiu77, 2023-10-15:
So good luck!